tbnobody/OpenDTU #616: Prometheus metric `opendtu_YieldTotal` (and `opendtu_YieldDay`) should be counter, not gauge

easimon, 2023-02-22:
### What happened?

Among the prometheus metrics exposed by OpenDTU, there are also yield metrics (total and per day). 

```
# HELP opendtu_uptime Uptime in seconds
# TYPE opendtu_uptime counter
opendtu_uptime 8462
...
# HELP opendtu_YieldDay in Wh
# TYPE opendtu_YieldDay gauge
opendtu_YieldDay{serial="123",unit="0",name="BKW",type="AC",channel="0"} 70.000000
# HELP opendtu_YieldTotal in kWh
# TYPE opendtu_YieldTotal gauge
opendtu_YieldTotal{serial="123",unit="0",name="BKW",type="AC",channel="0"} 4.666000
```

While it is arguable, whether the daily yield is more of a `counter` or a `gauge` (or if it is a useful metric for prometheus at all, since it has per-day-semantics), the total Yield should be a `counter`, since it is meant as a eternal sum, monotonously rising (maybe resetting to 0 after certain events, e.g. inverter firmware updates).

### To Reproduce Bug

Open http://<hostname-of-opendtu>/api/prometheus/metrics 
Search for `# TYPE opendtu_YieldTotal`
It is declared as type `gauge`

### Expected Behavior

`opendtu_YieldTotal` should be declared as a `counter`, like `opendtu_uptime`.
Probably the same is valid for `opendtu_YieldDay`.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

97bc964

### Anything else?

https://prometheus.io/docs/concepts/metric_types/

tbnobody, 2023-02-22:
To be honest, I don't use the prometheus endpoint. And have no clue about the syntax which is rendered. Are the lines beginning with # not just comments? Do they really belong to the syntax of the output?
I am passionless regarding this topic.

easimon, 2023-02-24:
I think I found a solution to correct this, please have a look at #624

tbnobody, 2023-02-26:
Should be fixed with c67b3e3cbdf944786b53001f98e82e217dc8cc89

closed 2023-02-26 by tbnobody

easimon, 2023-02-28:
Just for the record: this, plus 4d18ccf01db17958786d59040ccb10e3254819e3, which suppresses (incorrect) metrics until the first successful read after a DTU reboot.

Thanks for fixing this :-)