tbnobody/OpenDTU #617: [Request] Hardware nciht gefunden

schwitzer78, 2023-02-22:
### Is your feature request related to a problem? Please describe.

Hallo ich habe einen 
HM-600 WR
600.HRS0.EU.HM

und OpenDTU erkennt ihn nur als unbekanntes Modell - und ich soll hier ein Ticket aufmachen.
Braucht ihr sonst noch 
[image]

### Describe the solution you'd like

HM-600 WR
600.HRS0.EU.HM

tbnobody, 2023-02-22:
Hallo, Danke für die Info. Hardware Teilenummer 0 bzw. Hardware version 00.00 hört sich sehr seltsam an. Könntest du nach einer nacht (wenn der Inverter Stromlos war) ggf. nochmal schauen?

tbnobody, 2023-03-30:
Da hier keine Rückmeldung mehr kam würde ich den "Issue" erstmal schließen.

closed 2023-03-30 by tbnobody