tbnobody/OpenDTU #618: OpenDTU does not connect to local Wifi anymore

dersch81, 2023-02-22:
### What happened?

During an FW upload the OpenDTU did not respond anymore. Then after a powercycle it just opened up the OpenDTU AP but did not connect to the local anymore. 

Then performed the FW update again but it still does not connect to the Wifi anymore. Then went back to the previous FW but same issue. Did a config restore, factory default and config restore. Changing the wifi settings and back to the local one. 

But nothing helps. The unit is not connecting to the wifi anymore. I don't even try to connect.
[image]

But the local OpenDTU AP is working every time. So i'm really out of idea what coud be the reason. 

The OpenDTU HW is at the same place since several months and is working great there. Powered up via 5v 2A Adapter. An outdoor AP is in front of it just a few meter away. The wifi coverage is very good. 

### To Reproduce Bug

no idea

### Expected Behavior

Just connect to the fu**ing wifi. 

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

97bc964

ntfrnd, 2023-02-22:
Kann ich bestätigen. Habe vorhin auch auf die 97bc964 geupdated. Davor lief alles prima, nach dem Update und einem Spannung aus/Spannung an - selbes Verhalten. Schließlich habe ich die vorherige Version wieder aufgespielt und alles neu eingerichtet.

dersch81, 2023-02-22:
Ich habe es nun nur durch einen vollständigen Tausch des ESP32 hinbekommen. Läuft soweit mit 97bc964. Ich denke ein erasen und neues flashen des ESP hätte es auch getan. Aber ich hatte noch einen und es war einfacher die HW direkt zu tauschen. 

Bei dem betroffenen ESP32 habe ich nun ein erasing gemacht und neu geflasht. Denke damit läuft der wieder. Der kommt dann beim Schwiegervater zum Einsatz :)

ntfrnd, 2023-02-22:
Hatte ich zwischenzeitlich auch probiert und direkt die 97bc964 geflasht. Ging aber auch nicht stabil. Nach ein paar Neustarts war es wieder nicht erreichbar.

dersch81, 2023-02-22:
Eine Vermutung meinerseits ist dieser change hier: https://github.com/tbnobody/OpenDTU/commit/07f4473a4efcb0e8bbef71971ff35d367e17341a

Allerdings läuft auf einem neuen ESP32 die e97bc964 inkl dieses changes bisher einwandfrei.

tbnobody, 2023-02-22:
Als ich es gelesen habe hatte ich auch zuerst diesen commit im verdacht. das ist der einzige der was am Wifi krams geändert hat. Aber andererseits hab ich hier 2 ESPs ohne Probleme damit am laufen (und ich flashe das ja auch regelmäßig und bei jeder Änderung) und bei dir läufts jetzt auch... Ich verstehe es ehrlich gesagt noch nicht.

dersch81, 2023-02-22:
Ja mal beobachten. Ist wirklich eigenartig. Ich hatte aber schon seit ein paar Tagen das Problem, das ich bei WebGUI Zugriff keinen respond mehr bekam und power cyceln musste. Allerdings wurden stets die MQTT topics geschrieben und auch via REST bekam ich meine Werte. Daher ist das immer nur dann aufgefallen wenn ich mal auf die GUI bin.

Eventuell hat es bei mir doch auch was mit dem ESP zu tun gehabt. Evtl ist das Issue bei @ntfrnd zufällig das Gleiche aber nicht das Selbe. Mal  schauen ob sich noch mehr melden.

b-a-9-0, 2023-02-22:
Ich hatte heute auch das Problem, dass keine Verbindung mehr zu Stande kam (fixe IP). Konsolenoutput war: Lost connection, trying to connect... oder so ähnlich alle paar Sekunden. Nach ein paar Mal rebooten ging es wieder.

helgeerbe, 2023-02-23:
Der commit https://github.com/tbnobody/OpenDTU/commit/07f4473a4efcb0e8bbef71971ff35d367e17341a kommt von mir. 

Im default Modus macht der esp einen fast scan und verbindet sich mit dem ersten gültigen  Access Point. Bei meinem Mesh (Fritz Box) führte es dazu, dass er nach einem Reboot zuerst der Fritz Box hing, statt am Repeater im Gartenhaus. Irgendwann schubst ihn dann die Fritz!Box auf den Repeater, aber das dauert seine Zeit. Dabei war der ESP nicht immer ansprechbar.

Mit der Änderung führt der esp erst einen vollen Netzwerk-Scan durch  und verbindet sich mit dem stärksten, geeigneten Access-Point.  Bei mir läuft das mit 2 ESPs einwandfrei.

Man kann das auch wieder rausnehmen, wenn es Probleme macht. Dauert dann eben nur eine Weile, bis man die volle Bandbreite, bzw. eine stabile WiFi Verbindung hat.

dersch81, 2023-02-23:
Ich bin mir auch nicht sicher ob es was damit zu tun hat. Jedenfalls war nach über einer Stunde der ESP immer noch nicht verbunden. Ich habe eine Unifi Umgebung. Der ESP sieht an seiner Position potentiell 4-5 AP's wobei aber einer klar der Stärkste ist da nur 5 Meter Freifläche gegenüber. Beides ist Außen im Garten positioniert. Natürlich sind da aber auch sämtliche Nachbar AP's sichtbar die da in der Gegen rumfunken. 

Könnte man den Commit nicht so abändern, dass diese Funktion abschaltbar ist? Dann könnte man prüfen ob in der eigenen Umgebungssituation der Scan evtl ein Problem ist.

helgeerbe, 2023-02-23:
@dersch81 kompilierst du selber? Dann brauchst du in `src/NetworkSettings.cpp` nur die beiden Zeilen (124-15) rauszunehmen.

````c++

            WiFi.setScanMethod(WIFI_ALL_CHANNEL_SCAN);
            WiFi.setSortMethod(WIFI_CONNECT_AP_BY_SIGNAL);
````

dersch81, 2023-02-23:
Jo ich kompiliere selber. Nachdem ich den ESP32 getauscht habe läuft es einwandfrei. Daher bezweifele ich in deinem Commit die Ursache. Aber die Funktion im Webif abschaltbar zu machen könnte evtl anderen helfen.

tbnobody, 2023-02-23:
> Könnte man den Commit nicht so abändern, dass diese Funktion abschaltbar ist?
> Aber die Funktion im Webif abschaltbar zu machen könnte evtl anderen helfen.

Das würde ich so nicht machen. Sonst kann man bald für alles und jeden Separate Settings einführen. Speicher ist limitiert. Wenn das wirklich Probleme macht, dann kommts komplett raus und es dauert bei manchen Leute ggf. etwas länger bis nach einem Restart die WiFi Verbindung da ist. Aber aktuell ist ja noch nichtmal sicher ob es daran liegt.

LorenzFi, 2023-02-25:
Ich hatte dasselbe Problem bei mir mir zweiten "Gästewlan". Bin dann auf das Hauptwlan umgestiegen dort verbindet sich das Device ohne Probleme nach einem Neustart wieder mit dem WLAN. Vielleicht hilft die Information zur Fehlersuche.

klamasi, 2023-02-26:
Das Problem ist bei mir heute auch aufgetreten (Version 8b158a0). Die DTU lief 2 Monate ohne Probleme. Sie hat heute keinen Zugang mehr zum WLAN. Nachdem ich den Comment von LorenzFi oben gelesen habe, habe ich es mit dem GästeWLAN probiert - und siehe da die Verbindung klappte. 

Leider klappt es weiterhin nicht mit dem normalen WLAN. Liegt der Fehler gar nicht an der DTU? Macht die Fritzbox aus irgend einem Grund dicht und liefert keine IP-Adresse?
_________2 Stunden später_______________________
Jetzt habe ich den WPA Modus der FritzBox von WPA2 + WPA3 auf den gleichen Modus wie beim Gastzugang, also auf WPA (CCMP) gesetzt und siehe da die DT verbindet sich!

klamasi, 2023-02-26:
Schade, heute morgen war die DTU wieder offline - es liegt wohl doch nicht wie vermutet am WPA Modus. 
Ich habe dann ein Update auf Version 97bc964 eingespielt - erfolglos!

__Erfolg:__ 
 Es hatten alle WLAN Accesspoints (hinter einer FritzBox 7490) die gleiche SSID. Seit ich dem Repeater in der Nähe der  DTU eine individuelle SSID gegeben und die DTU damit verbunden habe, steht die Verbindung der DTU jetzt schon seit 10 Stunden.

ntfrnd, 2023-02-26:
> **Erfolg:** Es hatten alle WLAN Accesspoints (hinter einer FritzBox 7490) die gleiche SSID. Seit ich dem Repeater in der Nähe der DTU eine individuelle SSID gegeben und die DTU damit verbunden habe, steht die Verbindung der DTU jetzt schon seit 10 Stunden.

Was weiß bzw. stört einen Teilnehmer im WLAN, ob die SSID einmal oder mehrmals vorhanden ist? Solange SSID und Passwort passen, sollte es doch funktionieren, oder?
Außerdem empfiehlt AVM in einem Mesh-System hinter einer FritzBox immer dieselbe SSID zu nutzen. Von daher ist es bei mir auch so und ich hatte ja auch so meine Probleme mit der Version  [https://github.com/tbnobody/OpenDTU/commit/97bc964b6c761401cbd72120e1db564a78cdb91b](https://github.com/tbnobody/OpenDTU/commit/97bc964b6c761401cbd72120e1db564a78cdb91b) aber an meinem WLAN-Aufbau oder dem Gastnetz habe ich nicht rumgespielt, das bleibt so wie es ist.

LorenzFi, 2023-02-26:
Ich muss auch leider berichten, dass die vermeidliche Lösung mit dem WLAN-Netzwechsel keinen Erfolg gebracht hat. Gesten mehrmals neu gestartet ohne Probleme, heute eine Katastrophe. Also auch bei mir noch kein Erfolg in Sicht.

tbnobody, 2023-02-26:
Dann  nehmt doch einfach mal https://github.com/tbnobody/OpenDTU/commit/07f4473a4efcb0e8bbef71971ff35d367e17341a raus und  versucht es erneut.

dare83, 2023-02-27:
Hallo, habe auch Probleme mit einem meiner WLANS. Habe das Gefühl, dass die DTU nicht mit WLAN Passwörtern zurecht kommt, wenn diese nur aus Zahlen bestehen. Also 12345678 geht nicht abcdefgh geht, bei gleichem AP und SSID.

grisu642, 2023-02-28:
ich klink mich mal eben hier ein, weiß nur nicht obs passt:
Bin Neustarter, also heute 1. mal ESP und 1. mal Odtu, hat direkt funktioniert, nur seit ein paar Neustarts, weil andere Steckdose genutzt etc, findet er meinen WR nicht mehr, online müsste er sein weil über handy z.b. aufrufbar, passt das hier rein? Verbindet sich der WR direkt mit der Antenne des ESP bzw nrf? Weil meine PV aufm Balkon kein Wlan signal abbekommt

klamasi, 2023-02-28:
Schau dir mal unter NTP die Uhrzeit an und synchronisier sie gegebenenfalls von Hand. 
...dann ->Einstellungen -> Neustart. 

So klappt's bei mir.
klamasi

grisu642, 2023-02-28:
Danke für den schnellen Hinweis, werd ich später mal ausprobieren.

cthiele-mogic, 2023-03-06:
Ich klinke mich mal ein, ich habe eine neue ESP32 mit aktueller OpenDTU Version und es kann keine WLAN Client Verbindung hergestellt werden.  Der AP bleibt bestehen, er kann sich aber halt nicht in mein WLAN verbinden. 

In der Konsole 

21:50:31.631 > Try reconnecting
21:50:31.639 > Network lost connection
21:50:32.656 > WiFi disconnected
21:50:32.656 > Try reconnecting
21:50:32.662 > Network lost connection
21:50:33.688 > WiFi disconnected
21:50:33.688 > Try reconnecting
....

Irgendwann steht die Verbindung dann auf inaktiv. Umgebung ist ebenfalls Fritzbox, Mesh und ein paar Repeater.  Zeit habe ich manuell gesetzt, hat aber keine Änderung gebracht. 

Eine Idee wäre noch ein "!" im Passwort !?

tbnobody, 2023-03-06:
Hast du den ESP mal komplett Stromlos gemacht? (Nur so als Idee, hat  schon das ein oder andere Mal geholfen)

cthiele-mogic, 2023-03-06:
Ja, mehrfach neu gestartet, Strom gezogen. 

Interessanterweise habe ich gerade via Android einen Hotspot aufgemacht und die Verbindung hat innerhalb von 1-2s funktioniert.

tbnobody, 2023-03-06:
Haben $Leute bei denen es die Probleme macht ggf. alle Repeater im Einsatz? Oder haben $Leute Repeater im Einsatz und es geht?

b-a-9-0, 2023-03-06:
Bei mir treten die Probleme mit normalem FritzBox WLAN ohne Repeater auf.

grisu642, 2023-03-06:
Ich hab n Repeater im Einsatz und funktioniert, bin mir aktuell nicht sicher ob mit fritzbox oder fritzrepeater verbunden. Müsste aber eig der repeater aufgrund entfernung sein --Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

dersch81, 2023-03-06:
Ich habe keine Repeater im Einsatz. Aber seit dem Tausch des ESP32 auch
keine Probleme mehr gehabt.

ntfrnd, 2023-03-07:
Ich hab ne Fritzbox 7490 mit FritzRepeater 2400 im Einsatz und mit oben genannter Version meine Probleme. Es war auch ein neuer ESP32.

cthiele-mogic, 2023-03-08:
Ich konnte das Problem umgehen, meine OpenDTU liegt jetzt neben der Fritzbox (Mesh Master) und kann sich problemlos einloggen - offenbar hat er ein Problem wenn der Repeater ein stärkeres Signal sendet.

ntfrnd, 2023-03-08:
Mir ist in dem Zusammenhang FritzBox - Mesh aufgefallen, dass seit geraumer Zeit sehr oft zwischen Repeater und Fritzbox gewechselt wird. Mein Notebook steht am Schreibtisch, Repeater ist ca. 6m entfernt und liefert volles Signal. Notebook ist damit verbunden. Plötzlich bricht das WLAN-Signal kurz ab und danach sehe ich nur einen Balken. Wenn ich dann nachsehen, ist das Notebook plötzlich mit der FritzBox verbunden, die aber ein Stockwerk tiefer steht und deshalb auch der Empfang deutlich schlechter ist. Deshalb ja auch der Repeater (meiner arbeitet aber als Access-Point, ist also per LAN angebunden).

Im FritzBox-Systemlog (WLAN) sieht man deutlich, das auf einmal der Repeater ein abmelden und wieder anmelden an der FritzBox initiiert. Der Grund hierfür ist offenbar 

[image]

Also immer wenn der Repeater auf seinem Kanal einen anderen Router/Repeater aus der Nachbarschaft entdeckt (auch wenn das Signal noch so schwach ist, denn am Notebook sehe ich das WLAN nicht), fängt er an, meldet alles ab, bei der Fritzbox wieder an und nach ein paar Minuten werden alle Geräte wieder zum Repeater umgemeldet.

Ist für mich irgendwie völlig unnötig, da es keinerlei Beeinträchtigung für mich gibt. Nur das Umgemelde nervt enorm.
Vielleicht hat das auch was mit dem Problem hier zu tun?

Jedenfalls kam das Verhalten erst vor kurzem mit irgendwelchen Fritz-Updateas dazu, denn diese Kombination läuft schon seit mehreren Jahren problemlos.

Ich habe nun ein paar Haken in der Fritzbox anders gesetzt und teste mal wie sich das hier verhält.

andi-he, 2023-03-18:
Derzeitiger andauernder Kanalwechsel der Fritz!Box mit Ab- und Anmeldung macht Opendtu nicht  mit.   2,4 G fest auf Kanal 1 und alles klappt wieder. Habe Heltec_wifi_lora_32_V3 im Einsatz.

CptWonderful, 2023-03-19:
Ich möchte mich hier auch mal 'einklinken':
Habe einen Synology Router und draussen einen Outdoor EAP von TP-Link.
Die openDTU Box wurde im Büro mit meinem IoT Netzwerk (2,4 & 5GHz) verbunden (Zur Ersteinrichtung + Test), muss aufgrund der schlechten Verbindung zum WR aber in Garage wo der Outdoor AP steht. Hier verbindet sich openDTU nicht mehr mit dem gleichen Netzwerk (Gleiche SSID, gleiches PW, nur der Outdoor AP). Hole ich die Box wieder ein -> Funktioniert es. Behelfsweise habe ich auf dem Outdoor AP ein dediziertes Netzwerk im gleichen VLAN erstellt -> Funktioniert. 

Nutzt openDTU ESSID zur Verbindungherstellung (Kann ich mir eigentlich nicht vorstellen)?`

klamasi, 2023-03-20:
> Schade, heute morgen war die DTU wieder offline - es liegt wohl doch nicht wie vermutet am WPA Modus. Ich habe dann ein Update auf Version [97bc964](https://github.com/tbnobody/OpenDTU/commit/97bc964b6c761401cbd72120e1db564a78cdb91b) eingespielt - erfolglos!
> 
> **Erfolg:** Es hatten alle WLAN Accesspoints (hinter einer FritzBox 7490) die gleiche SSID. Seit ich dem Repeater in der Nähe der DTU eine individuelle SSID gegeben und die damit verbunden habe, steht die Verbindung der DTU jetzt schon seit 10 Stunden
----------------------
Die DTU läuft jetzt an der FB mit individueller SSID auch nach 3 Wochen immer noch fehlerfrei.

...aber ich habe auch andere (alte) Geräte die seit kurzem das gleiche Problem haben: z.B. 1 Nexus TAB 2013.  Das Nexus wechselt ständig den AccessPoint und gibt irgenwann die Verbindungsversuche auf. Da muss sich ohne mein Zutun etwas am Fritz Mesh (Version 7.29) geändert haben. Wenn ich die Repeater im Mesh eingebunden lasse und nur den Haken in der folgenden Option entferne funktionieren die Geräte problemlos wie seit Jahren. **Ich denke das Problem liegt bei AVM!!!!**

> Einstellungen aus dem Mesh automatisch übernehmen
> 
> Dieser FRITZ!Repeater ist Bestandteil des WLAN Mesh und übernimmt automatisch Einstellungen der FRITZ!Box.
> 
> Dazu gehören:
> 
>     Alle relevanten WLAN-Einstellungen, darunter Name des Funknetzes, Gastzugang, Funkkanal
>     Einstellungen für Auto-Updates, Push Service und AVM-Dienste
>     Der im Bereich Heimnetz des Mesh Masters für dieses Gerät vergebene Geräte-Name
> 
> [ ] Einstellungsübernahme aktiv

klaus-weindel, 2023-03-29:
Die Fritzboxen und Mesh spielen sicher auch eine Rolle, die verwendete Software aber auch. Nachdem ich nach einem Update keine Verbindung mehr bekam bin ich auf Ahoy DTU umgestiegen und hier funktioniert die gleiche Hardware tadellos. Auch meine zahlreichen Tasmota Geräte funktionieren ohne Probleme.

Xenon-s, 2023-04-21:
gleiches Problem, mehrfach neu geflasht und auch andere Versionen ausprobiert, er bekommt keine Verbindung zum WLAN. Schade, bleibe ich halt bei AhoyDTU, da funktioniert es auf Anhieb

indie89, 2023-04-24:
Ich habe aktuell auch ein vermutlich Wifi-bezogenes Problem. Ich versuche es mal mit dem festen Kanal für Wifi und falls das nicht klappt, eröffne ich ein eigenes Issue.
Kurzbeschreibung meines aktuellen Problems: Ich nutze den openDTU Adapter von ioBroker und kommuniziere per Web API mit der openDTU. Nach 1/2-x Tagen fängt der Adapter an, fast sekündlich neue Websocket-Connections zur openDTU aufzumachen. Gleichzeitig  verändern sich die WR-Gerätenamen in openDTU (hab 2 connected) in komische (Sonder-)Zeichen. Und dann stürzt der Adapter im ioBroker ab (wegen mehrfachen Neustarts). Was Ursache und was Wirkung ist, weiß ich nicht, ich hab aber schon einiges probiert. Evtl. verliert die openDTU die Verbindung zum Wifi und der Adapter versucht dann immer wieder zu connecten...oder es ist doch etwas anderes (wie gesagt, ich eröffne ein eigenes Issue, falls das mit den festen Kanälen nicht hilft).

TheOtherDoctor, 2023-04-29:
Seit heute hier dasselbe Problem. Frisch eingerichtetes OpenDTU lief seit einigen Tagen problemlos, dann musste ich heute mal mein WLAN runterfahren und seit das danach (unverändert) wieder hochkam, will OpenDTU nicht mehr connecten. Keine Verbindungsversuche sichtbar, stattdessen wird der 192.168.4.1 wieder aktiviert. Dort in den Einstellungen steht mein WLAN noch drin, aber OpenDTU verbindet sich einfach nicht. In den WLAN-Einstellungen steht  Status "nicht aktiv", obwohl meine WLAN Infos hinterlegt sind, genau wie im Initial Post hier. Sieht mir also so aus, als ob das Problem seit mehr als 2 Monaten unverändert existiert. :-(

Hier ist eine aus dem bin package installierte v23.4.25 am Start - Netzwerk ist langweilig: Fritzbox als äusseres Gateway und ein lokaler AP, der dahin durchroutet, für zig Clients seit jeher bestens funktionierend. Kein Mesh oder so.
Sehr frustrierend, sehe keinen Ansatzpunkt mehr.

TheOtherDoctor, 2023-04-29:
> gleiches Problem, mehrfach neu geflasht und auch andere Versionen ausprobiert, er bekommt keine Verbindung zum WLAN. Schade, bleibe ich halt bei AhoyDTU, da funktioniert es auf Anhieb

Könntest du verraten, bis zu welcher Version du zurückgegangen bist? Dann würde ich von da ab weiter zurückgehen... Ich mag nicht gleich auf AhoyDTU wechseln, hatte mich grad erst auf OpenDTU eingerichtet.

klamasi, 2023-04-30:
Hast du dir mal die Zeit auf der DTU angeschaut? Ist die aktuell?
Erst wenn ich die manuell (per direkt Access) setze verbindert die DTU sich danach mit dem WLAN. 
Warum das so ist weiß ich nicht. Ich habe das Anfang Jahr schon mal beschrieben.

Troubadix, 2023-05-06:
Hallo,

ich bin neu im Solarbereich und bin dabei auf dieses tolle Projekt gestoßen.

Leider habe uach ich mit der aktuellen Version 23.5.3 das Problem, das sich der ESP32 nicht mit meinem Heimnetz verbinden möchte.
Ich habe dann mal testweise den ESP komplett gelöscht und die 23.4.28 aufgespielt, mit dieser Version verbindet sich der ESP sofort mit meinem Netz (Unifi U6-Enterprise).
Wenn ich dann übers Webinterface ein Update auf die 23.5.3 mache, ist der ESP wieder draußen aus dem WLAN.

Gibt es denn besondere Einstellungen, die fürs Unifi-WLAN empfohlen werden für erhöhte Kompatibilität)

tbnobody, 2023-05-06:
Das ist auch interessant... weil eigentlich hat  sich zwischen 23.4.28 und 23.5.3 absolut garnichts am code geändert. Nur ein klein wenig Doku und die Webapp die aber nur am Client ausgeführt wird.

https://github.com/tbnobody/OpenDTU/compare/v23.4.28...v23.5.3

stefankubis, 2023-05-08:
Habe (hatte) in meinem Setup auch Probleme.

Ich habe mehrere Ubiquity Access Points im Einsatz - mit dem Ergebniss, das die Verbindung zu den ESPs sehr Instabil läuft. 

Testweise habe ich mal eine SSID so konfiguriert, das nur einer der Access-Points sie ausstrahlt und OpenDTU darauf konfiguriert - seitdem läuft das stabil. Dies ist jedoch eher ein Workaround als eine Lösung.

isostar1101, 2023-05-08:
Hallo Zusammen, Du hast die Möglichkeit die OPENDTU an einen AP zubinden das hat bei mir sehr gut funktioniert! Den Punkt Lock to Access Point findest Du unter Client Devices.

Troubadix, 2023-05-14:
Ich muss hier nochmal nachhaken. ..

Die 23.4.28 tuts nach erneutem Flashen auch nicht mehr. Übrigens auch bei der aktuellen 23.5.9 das gleiche Problem.

Wo bekomme ich denn ältere Versionen her? 
Unter "Releases" sind nur 10 Stück zu finden, die älteste von Anfang April. Den Beiträgen hier zu folgern scheint das Problem ab Februar zu starten. Ich würde daher gerne mal die "last known good" Version testen - wo kann ich diese downloaden?

PS: Ich habe nur einen AP, das "pinnen" an einen bestimmten ist also nicht zutreffend für mich.

Danke!

SpiGAndromeda, 2023-05-17:
OpenDTU hat sich bei mir einmal kurz mit der Fritzbox verbunden. Sie hat auch eine IP Adresse bekommen. Aber seitdem geht gar nichts mehr. Mesh komplett deaktiviert und Router stromlos genommen. Verschiedene Kanäle und Passwörter probiert.

Eigentlich wollte ich von AhoyDTU umsteigen wegen besserer MQTT Funktionalität. Aber OpenDTU ist so leider (für mich) nicht nutzbar.

tbnobody, 2023-05-17:
Es bräuchte mal jemanden der das Problem hat und den Code selbst compiliert. Sonst wird das mit testen recht mühsam...

SpiGAndromeda, 2023-05-17:
> Es bräuchte mal jemanden der das Problem hat und den Code selbst compiliert. Sonst wird das mit testen recht mühsam...

Kann ich gerne machen. Soll ich was spezielles compilieren?

kreidler, 2023-05-21:
Es verwässert zwar schon hier, aber auch mein Senf:
Für den Vater ein System gebaut und mit 23.5.9 geflasht. Wifi Verbindung bei mir im Haus auf AVM7590 Mesh mit kurzer SSID ohne Probleme. Einrichtung vor Ort beim alten Herren - kein Erfolg bis ich die Standard-SSID seiner Fritzbox auf etwas ohne Leerzeichen geändert hatte. Gegentest danach natürlich nicht mehr gemacht. Seither keine Probleme

ntfrnd, 2023-05-21:
Ich hatte oben auch schon mal geschrieben, dass ich Probleme habe. Bei meiner Fritzbox 7490 inkl Repeater 2400 im Mesh ist die SSID lediglich 8 Zeichen,  keine Umlaute, Sonderzeichen oder Leerzeichen.

matbol, 2023-05-26:
Hallo zusammen,
ich habe heute erstinbetriebnahme gemacht, tatsächlich sehe ich auch Wechselrichterwerte - solange ich über den lokalen AP des esp drauf gehe. Möchte ich über meinen Fritz Repeater (WPA3?) auf die IP zugreifen, erhalte ich timeouts. Fritzbox weißt aber der openDTU eine IP zu, steht aktiv und zeigt mir auch an, dass es über den repeater ins netz geht......

`$ git log --oneline`

`889e191 (HEAD -> master, tag: v23.5.23`

001oc, 2023-06-02:
Hallo zusammen,

ich habe auch die "in Verdacht stehende" Konfiguration Fritzbox 7490, mehrere APs im Mesh sowqie die erfolglosen Verbindungsversuche ins WLAN. Compiliert habe dich die aktuelle Latest Version v23.6.1.

Ich habe dann als nächstes mal die beiden Zeilen in der NetworkSettings.cpp ausgeklammert und neu kompiliert, vorsichtshalber nochmal vor dem Upload das Flash erased - kein Effekt, immer noch twork lost connection
15:26:42.516 > WiFi disconnected
15:26:42.518 > Try reconnecting
15:26:42.520 > Network lost connection
15:26:44.935 > WiFi disconnected
15:26:44.938 > Try reconnecting
15:26:44.938 > Network lost connection
...

001oc, 2023-06-02:
... dann habe ich das ganze nochmal gemacht. Jetzt habe ich dann nochmal, nur so zum Spaß, statt der Default Config die ESP32 C3 Config gewählt (im Bewußtsein, dass das für meine Hardware falsch sein müßte). Kompilierung hat ohne probleme geklappt, Upload hat dann gemeckert wegen falscher Hardware. Als nächstes habe ich dann die Generic Variante (in der Annahme, dass das auch gleichzeitig die Default Variante ist) genommen. Wieder vorsichtshalber den BuildCache gecleaned und komplett neu kompiliert. Upload hat dann funktioniert. Anschließend über den AP vom ESP32 (192.168.4.1) zu allererst unter NTP die Zeit manuell angepasst. Als nächstes  unter Network das lokale WLAN (FritzBox 7490, Mesh) mit DHCP eingerichtet. Das serielle Logging in VSCode lief mit. Nach dem Abspeichern hat der ESP32 anstandslos eine Verbindung zum FritzBox Mesh-AP aufgemacht und hat eine IP bekommen. Als nächstes habe ich erstmal nur einen Reset gemacht - hat auch geklappt. Als nächstes habe ich mal Stromversorgung getrennt und wieder angesteckt - hat auch geklappt. Langzeitprognosen sind natürlich jetzt noch nicht möglich und schlecht bleibt, dass das der Fehler bis jetzt erstmal nicht gut reproduzierbar scheint. ich beobachte mal weiter. Achso, ich hatte in beiden Fällen die beiden Zeilen in der NetworkConfig.cpp auskommentiert. Es ist nicht vollständig auszuschließen, dass noch alte Teile objectfiles vom ersten Versuch inkl. eingebunden Zeilen in der NetworkCConfig.cpp mit gelinkt wurden. Ich werde, wenn ich etwas mehr Zeit habe, die ganze Geschichte nochmal in einer garantiert cleanen VM kompilieren und schauen ob sich da was repoduzierbares ergibt...

tbnobody, 2023-06-02:
Es könnte auch sein, dass diese Scan Methoden bzw. Sort Methoden in der WiFi Config im ESP persistiert werden und durch auskommentieren einfach nicht mehr auf den Standardwert gesetzt werden. Es würde hier helfen vor einem Neuflashen den ESP zu löschen oder die Scan Methoden in der NetworkSettings.cpp auf die Standardparameter zu setzen.

Standard wären folgende Parameter:
```cpp
WiFi.setScanMethod(WIFI_FAST_SCAN);
WiFi.setSortMethod(WIFI_CONNECT_AP_BY_SIGNAL);
```

001oc, 2023-06-02:
@tbnobody ich habe das m.E. beim ersten Versuch auch so gemacht, jedenfalls habe ich vor dem Upload den Flash gelöscht gehabt mit platformio run -e generic -t erase

Ich habe jetzt nicht nochmal nachgeschaut, aber ich meine das diese Sequenz (zumindest bei Tasmota) zuverlässig den Flash löscht, inkl. Wifi-Config und sonstigem. Mein Verdacht geht tendenziell eher in Richtung Build-System, das da nach dem löschen noch was liegen geblieben ist und dann wieder mitgelinkt wurde.

Ich kann das im Moment in meiner Konfiguration noch nicht gut genug reproduzieren.

Was das persistieren von Daten über die APs angeht, kann ich das grundsätzlich aber bestätigen, aber eigentlich sollten die nach dem Löschen des Flashspeichers auch weg sein.

001oc, 2023-06-02:
das könnte m.E. schon daran liegen. Mein Ablauf war wie folgt.

1.) Standard-Source (latest) in VScode inkl. PIO kompiliert, hochgeladen ... funktioniert nicht mit lokalem WLAN
2.) build clean (ggf. nicht ganz clean), flash erase (ggf. nicht ganz erase) Modifikation in NetworkConfig.cpp, kompiliert, hochgeladen ... funktioniert nicht mit lokalem WLAN.
3.) ein bißchen Verwirrung stiften im Bild-System (div. Varianten installiert - hat vielleicht dazu geführt, dass tatsächliche alle caches und die .pio tatsächlich mal komplett gelöscht wurden). Die Variante "generic" installiert, Modifikation in NetworkConfig.cpp, kompiliert, hochgeladen, NTP Einstellungen gemacht, Networkeinstellungen gemacht ... seit ein paar Stunden funktioniert es jetzt auch trotz diverser Resets bzw. (provozierten) Brownouts.

001oc, 2023-06-02:
auffällig ist ja auch, dass es sich möglicherweise ausschließlich um Fritzboxen mit  gemeshten APs handelt die das Problem haben/machen. Das könnte schon darauf hinweisen, dass es sich um ein Problem bei der Zuordnung von der SSID mit wechslenden MAC-Adressen handeln könnte, die die ESP WIFI Lib nicht korrekt verarbeitet bekommt. Ich versuche morgen erstmal ein bißchen Roaming mit meinem ESP32 und trage den mal zu den verschiedenen APs bei mir und schaue mal ob sich das Problem dann beim AP-Wechsel wieder einstellt.

001oc, 2023-06-03:
Leider keine weiteren Hinweise auf die Fehlerursache. Ich habe eben mal ein paar Roaming-Vorfälle provoziert, die haben aber bei mir ohne Probleme funktioniert. Ich habe sowohl durch Standortwechsel in unmittelbare Nähe von APs als auch durch Abschaltung einzelner APs und anschließenden Reset des ESP32 jeweils Roaming-Wechsel erzwungen. Ausserdem habe ich auch völlig neue APs, die vorher im Mesh unbekannt waren (neue MAC-Adresse) und anschließendem Neuverbinden des ESP32 versucht den Fehler zu reproduzieren. Alles hat anstandslos funktioniert.

Ich werde mal die statistische Basis etwas verbessern und Morgen mal zusätzlich 9 openDTUs aufbauen (im Moment habe ich nur 10 NRF24L01+ hier rumliegen) und laufe dann mit denen mal rum.

tbnobody, 2023-06-03:
Es wäre jetzt der Gegentest interessant. Also wenn du die beiden Zeilen wieder einfügst ob es dann wieder nicht mehr funkioniert.

[48 more replies]

closed 2024-01-19 by tbnobody