tbnobody/OpenDTU #625: [Request] Prebuild firmware files via GitHub Actions

foorschtbar, 2023-02-24:
### Describe the solution you'd like

Hi, as i saw you already use GitHub Actions, but why you don't build the Firmware files directly for OTA-Update? I have done this many time, but maybe there is a reason? If not, i can provide a PR for the pipeline with releases if you wish.

tbnobody, 2023-02-24:
The files are generated. Just download them within the action.

foorschtbar, 2023-02-25:
Only the generic version, not as binary and you need to login to download - or have I overlooked something?

mathieucarbou, 2023-02-25:
> The files are generated. Just download them within the action.

Just a node: It would be nice when a build is successful to create a tag and attach the artifact to the release page. Downloading from the Actions tab require to be logged into GH and consequently have a GH account... This is OK for devs but not everybody using OpenDTU is a dev ;-)

Coxeroni, 2023-03-02:
I am a newbie, I do have a github account but still cannot find any binary under the actions tab...
Anybody could guide me? Filtering by "master" doesn't seem to work and I stuck with 297 items from which I have no clue which one to take.

tbnobody, 2023-03-02:
Use the latest one: https://github.com/tbnobody/OpenDTU/actions/runs/4283851260

Quote from the Readme: 

> The pre-compiled files can be found on the [github page](https://github.com/tbnobody/OpenDTU) in the tab "Actions" and the sub menu "OpenDTU Build". Just choose the latest build from the master branch (search for "master" in the blue font text but click on the white header text!). You need to be logged in with your github account to download the files.

Coxeroni, 2023-03-05:
Thanks for helping me. In fact I missed the part that the binary is visible under Artifacts. Maybe you could mention that in the readme.

closed 2023-04-11 by tbnobody