tbnobody/OpenDTU #626: OpenDTU reboots every evening, probably short time after "nautical sunset"

madmartin, 2023-02-24:
### What happened?

Since my last upgrade, OpenDTU makes a restart, I assume around the timepoint of "nautical sunset".
Current Time: 21:23 - current uptime: 2:26h -> 18:57
Current nautial sunset: 18:49:00

This came up because after the restart, all counters in the LiveView are zeroed, because the inverters dont answer after sunset.

This happened today and yesterday (at least).
Before the last upgrade, the "git pull" was    6a6165e..97bc964  master     -> upstream/master
This was at 2023-02-22 19:30:20 (thanks to bash history).
My hardware works very reliable and is unchanged since months.
So I assume 6a6165e..97bc964 introduces some new instability.

### To Reproduce Bug

Wait 24h....

### Expected Behavior

No  restart for a long time.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

97bc964

### Anything else?

martin@bln9716cm ~/swbuild/OpenDTU $ git log --pretty=format:"%h %G? - %an / %s" 6a6165e..97bc964
97bc964 N - Thomas Basler / Fix lint errors
dc309e9 N - Thomas Basler / Update sunset from 1.1.3 to 1.1.7
8b49f0b N - Thomas Basler / webapp: add app.js.gz
39c5550 N - Thomas Basler / Merge branch 'pr594' into dev
c61064c N - Thomas Basler / Merge branch 'pr605' into dev
bf01646 N - Thomas Basler / webapp: use terser for further size reduction
4feb37e N - Thomas Basler / webapp: update dependencies
98dfbfb N - Thomas Basler / webapp: use unplugin-vue-i18n to reduce bundle size
07f4473 N - helgeerbe / Connect Wifi to strongest AP   * do full channel scan   * connect to strongest AP
dc66a82 N - Thomas Basler / Fix typo in configuration. Relates to #603
20a83fc N - Thomas Basler / webapp: Update dependencies
fa3d037 N - Thomas Basler / Fix isReachable and isProducing based on PollEnabled
42b6006 N - Thomas Basler / webapp: Change badge style and encapsulate it into component
f6dedf7 N - Thomas Basler / webapp: Add missing translations
e5e2365 N - Thomas Basler / webapp: Add additional hint that no data is received if no sun is shining
4c14417 N - Thomas Basler / webapp: Add column inverter settings to show inverter send/receive status
cd99ab8 N - Thomas Basler / Added settings to pause polling/sending commands in general and at night
b319c78 N - Thomas Basler / Added methods to calculate and show sunrise and sunset times
19a1c0a N - Thomas Basler / Added config parameters to set longitude and latitude
7da782c N - Thomas Basler / Hoymiles Lib: Added flags to inverter to enable/disable polling and commands to the inverter
fc86ff7 E - Torben Woltjen / Fix typos in German localisation and change some words to more common one

tbnobody, 2023-02-24:
If its predictable every 24h, please provide the output of the serial console (not the web one). There should be the stack trace if a crash occours. I cannot reproduce this on my side. Runs perfectly without  problems.

madmartin, 2023-02-26:
I attached a notebook to my OpenDTU and started a monitor task. The logs since then are several MB large, so I trimmed the logs with this commands:

```
cat $FILE | sed \
-e '/SystemConfigPara/d' \
-e '/Fetch inverter/d' \
-e '/Interrupt received/d' \
-e '/> RX /d' \
-e '/> TX /d' \
-e '/> .* missing/d' \
-e '/Request retransmit/d' \
-e '/Nothing received/d' \
-e '/> Success/d'
```

First, the reboots are not every 24h, the occur more often. Here is the log from 230225-114514 until next day 12:30:

```
martin@bln9716n ~/swbuild/OpenDTU $ ./log-trim.sh platformio-device-monitor-230225-114514.log
11:45:53.857 > Websocket: [/livedata][1] connect
11:46:08.647 > Retransmit timeout
11:46:38.033 > Request device info
11:47:39.085 > Request device info
11:47:57.316 > Admin mode disabled
12:00:20.216 > Websocket: [/livedata][2] connect
12:00:29.241 > Websocket: [/livedata][2] disconnect
12:41:26.041 > Websocket: [/livedata][3] connect
12:47:40.659 > Websocket: [/livedata][3] disconnect
12:50:45.300 > Retransmit timeout
13:00:21.598 > Retransmit timeout
13:01:52.456 > Retransmit timeout
13:02:37.779 > Retransmit timeout
13:34:16.992 > Websocket: [/livedata][1] disconnect
13:36:47.465 > Retransmit timeout
13:42:27.815 > Websocket: [/livedata][4] connect
13:46:59.895 > Websocket: [/livedata][4] disconnect
13:56:49.261 > Retransmit timeout
14:48:26.402 > Retransmit timeout
14:50:27.658 > Retransmit timeout
14:51:13.094 > Retransmit timeout
15:44:36.248 > Retransmit timeout
16:06:21.550 > Retransmit timeout
16:31:38.468 > Retransmit timeout
16:43:01.744 > Retransmit timeout
16:45:17.860 > Retransmit timeout
16:52:09.683 > Retransmit timeout
17:03:16.974 > Retransmit timeout
17:09:37.255 > Retransmit timeout
17:27:37.916 > Packet handling error
20:04:59.592 > Websocket: [/livedata][5] connect
20:05:13.537 > Websocket: [/livedata][6] connect
20:05:13.650 > Websocket: [/livedata][5] disconnect
20:05:29.720 > Websocket: [/livedata][6] disconnect
20:06:42.528 > Websocket: [/livedata][7] connect
20:08:11.168 > Websocket: [/livedata][8] connect
20:11:31.939 > Websocket: [/livedata][7] disconnect
20:15:13.415 > Websocket: [/livedata][9] connect
20:23:50.418 > Websocket: [/livedata][8] disconnect
20:24:38.426 > Websocket: [/livedata][10] connect
20:26:40.954 > AsyncWebSocketClient::_queueMessage: Too many messages queued, closing connection
20:26:40.962 > Websocket: [/livedata][9] disconnect
20:27:35.042 > Websocket: [/livedata][11] connect
20:31:21.890 > Websocket: [/livedata][10] disconnect
20:32:10.024 > Websocket: [/livedata][12] connect
20:34:11.468 > AsyncWebSocketClient::_queueMessage: Too many messages queued, closing connection
20:34:11.476 > Websocket: [/livedata][11] disconnect
20:35:07.238 > Websocket: [/livedata][13] connect
20:36:52.991 > Websocket: [/livedata][12] disconnect
20:36:52.995 > Websocket: [/livedata][13] disconnect
20:37:41.552 > Websocket: [/livedata][14] connect
[... 613 lines cut ...]
12:29:56.357 > done
12:29:56.358 > Configuring WiFi STA DHCP IP... done
12:30:11.945 > Disable search for AP... WiFi disconnected
12:30:11.952 > Try reconnecting
12:30:11.954 > Network lost connection
12:30:11.956 > done
```

After 10:41 the ESP lost the WiFi connection and did not manage to re-establish it. 
Looking into the DHCP server's log, I can see DHCPDISCOVER and DHCPOFFER messages, but no more (DHCPREQUEST should be next...)
Some time later, at 12:28 I went to the ESP an pressed the reset button two times.
The WiFi connection did not re-establish (see log). 
I had to powercycle the ESP. 
Until 20:00, there occurred  5 more reboots of the ESP.

I am a little clueless what has caused this instability. No changes in my house network the last weeks, no new firmware in the router....

I see two suspicious commits:
- 07f4473 N - helgeerbe / Connect Wifi to strongest AP   * do full channel scan   * connect to strongest AP
- f5fb628 N - Thomas Basler / Update espressif32 from 6.0.0 to 6.0.1

I will revert these changes one by one and lets see the results....

Other ideas? 
Can I enable any futher debugging to get a better stack trace from the crashes or similar?

madmartin, 2023-02-26:
I see similarities to #618

madmartin, 2023-03-11:
In the last two week I did a lot of testing and I could not find a reason for the crashed and reboots in less than 24 hours. 
I tried to revert those suspicious commits above, no change. After a week, I did not wanted to continue to search and updated my ESP back to the current release - and then it suddenly was stable for 6 days.
Everything as written above without any changes to the rest of the infrastructure.
So I close this issue as sporadic error with unknown reason.

closed 2023-03-11 by madmartin