tbnobody/OpenDTU #643: Reset via URL (with or maybe without PW if it is easier)

GitHag42, 2023-02-27:
### Is your feature request related to a problem? Please describe.

I discovered that the MQTT data is not updated anymore once in a while, although the webinterface opens normally.
It can be noticed with a watchdog timer in NodeRed and I would like to software reset the ESP then, because this solved the problem.

### Describe the solution you'd like

A software reset via URL that can be triggered with automation (via NodeRed for instance). Or is this already implemented?

### Additional context

I somehow thought of this issue that happens to be related to MQTT libraries:
"MQTT client does not connect to MQTT broker after a TCP reconnection" https://github.com/mruettgers/SMLReader/issues/47
But I could not yet reproduce this with the openDTU ESP for sure.
After disconneting and connecting my access point again the ESP with openDTU just rebooted and was working correctly.
With the SMLreader I could reproduce the problem by restarting my Fritzbox and the SMLreader is not sending MQTT data although connected to the WiFi. 
But the ESP with openDTU is connected to my Unifi access points in a different WiFi. So it is not the absolute same situation.
For the SMLreader the watchdog soultion is working fine so far.

LustigePerson, 2023-03-01:
I just noticed this today with FW `6a6165e`. In the morning, no MQTT data was sent. But this only occured today for the first time.

ehofm3, 2023-03-02:
idk if that's a sufficient solution for you, but you can indeed trigger a reboot via web request (your browser does nothing different when you hit the button in the web interface).
I managed it with curl:
```
curl 'http://opendtu/api/maintenance/reboot' \
  -H 'Content-Type: multipart/form-data; boundary=----WebKitFormBoundaryEKkgmvYBqjoZr06U' \
  -H 'authorization: Basic YW..VT' \
  --data-raw $'------WebKitFormBoundaryEKkgmvYBqjoZr06U\r\nContent-Disposition: form-data; name="data"\r\n\r\n{"reboot":true}\r\n------WebKitFormBoundaryEKkgmvYBqjoZr06U--\r\n'
```

I just recorded the web requests with Chrome and then copied the reboot one as a curl command, and then cut everything unnecessary. Maybe it can be shortened even further.

I don't know if there is another way to retrieve the auth key (YW..VT), didn't look into the code. You can also supply your credentials in the url (as you would for the api) like `http://admin:<pw>@opendtu/api/maintenance/reboot`, and then just get rid of the authorization line.

closed 2023-03-06 by tbnobody

GitHag42, 2023-03-07:
I tried it yesterday but http://admin:<pw>@opendtu/api/maintenance/reboot with pw and opendtu replaces by the password and the IP did not work. I need something like this link to use it as a http request in Nodered. I don't know what to do with this curl stuff...

ehofm3, 2023-03-07:
If you press the reboot button while you record the requests via Chrome (F12) you should be able to see the request:
[image]

All required data for you to build your http request in NodeRed should be in this recording.
As described above, you can even copy the request as a curl command or whatever you like. Just google "chrome inspect network activity" or so.