tbnobody/OpenDTU #644: Unbekanntes Modell! Hardware-Teilenummer 269496322 HM-400

dare83, 2023-02-28:
### What happened?

Unbekanntes Modell! Hardware-Teilenummer 269496322 HM-400

### To Reproduce Bug

Unbekanntes Modell! Hardware-Teilenummer 269496322 HM-400

### Expected Behavior

Unbekanntes Modell! Hardware-Teilenummer 269496322 HM-400

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

21aa3c5

Solarteur, 2023-02-28:
Die oben angegebene Teilenummer passt nicht in das Schema der HM-Seriennummern. Diese müsste für einen HM-400 mit 1121 beginnen  und acht weitere Ziffern haben, also **HM-400: 1121 xxxx xxxx**

Vgl. z, B. https://www.mikrocontroller.net/topic/525778?page=single#7020587

tbnobody, 2023-02-28:
Die Nummer von oben scheint wirklich eine Hardware-Teilenummer zu sein und keine Seriennummer.

[image]

Rot markiert wäre die entsprechende Nummer. Passt also schon irgendwie ins Schema.

@dare83 was sind denn die linken 7 Stellen der Seriennummer?

dare83, 2023-02-28:
Also die Seriennummer ist 112163908926

[image]

Die Hardwareteilenummer habe ich aus der Anzeige vom OpenDTU

[image]

Solarteur, 2023-02-28:
Gibt es eine Fehlermeldung bei Hinzufügen eines neuen Wechselrichters unter Eingabe der Seriennummer (1121...)? Die Hardware-Teilenummer ist da ohne Bedeutung . Ist der HM-400 in Betrieb,  d. h. ein PV-Modul angeschlossen und es ist draußen hell? Nur dann kann OpenDTU den Wechselrichter erreichen.

tbnobody, 2023-02-28:
Nein.... Er wird ja auch ausgelesen.... Soweit alles gut. Es kann nur nicht mit der Hardware-Teilenummer auf die Leistung usw. geschlossen werden.

dare83, 2023-02-28:
> Gibt es eine Fehlermeldung bei Hinzufügen eines neuen Wechselrichters unter Eingabe der Seriennummer (1121...)? Die Hardware-Teilenummer ist da ohne Bedeutung . Ist der HM-400 in Betrieb, d. h. ein PV-Modul angeschlossen und es ist draußen hell? Nur dann kann OpenDTU den Wechselrichter erreichen.

ließ sich hinzufügen, läuft und zeig alles richtig an (soweit ich das verstehe)

dare83, 2023-02-28:
> Nein.... Er wird ja auch ausgelesen.... Soweit alles gut. Es kann nur nicht mit der Hardware-Teilenummer auf die Leistung usw. geschlossen werden.

ich habe nur das gemacht, was in der meldung steht (siehe screenshot) ;) "Bitte melden sie die Hardwareteilenummer und das Modell..."

tbnobody, 2023-02-28:
> ich habe nur das gemacht, was in der meldung steht (siehe screenshot) ;) "Bitte melden sie die Hardwareteilenummer und das Modell..."

Alles gut, danke dafür! :)

Werde den Teilenummer in die Mappings mit aufnehmen.

commit f423ce0a (2023-03-02): Fix #644: Added additional HM-400 to mapping table
closed 2023-03-02 by tbnobody