tbnobody/OpenDTU #646: Web API Auth required needs to be updated

polarbearing, 2023-03-01:
### What happened?

Upgraded OtA yesterday. 

Tried the web API, but it seems some calls that are listed in the [doc](https://github.com/tbnobody/OpenDTU/blob/master/docs/Web-API.md) as no auth actually need one. 

If that's intended, the doc should be updated accordingly. 

### To Reproduce Bug

curl -v

### Expected Behavior

doc matches system

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

a341c88

### Relevant log/trace output

```Shell
curl -v  http://opendtu/api/livedata/status
HTTP/1.1 401 Unauthorized

curl -v http://admin:****@opendtu/api/livedata/status
HTTP/1.1 200 OK

same e.g. for /api/system/status
```

tbnobody, 2023-03-01:
Have you enabled this?
[image]

If not, you need a password

tbnobody, 2023-03-07:
Added some remarks here: b91da0f681ef96bb611b723ffebe718fb6c08ca5

closed 2023-03-07 by tbnobody