tbnobody/OpenDTU #649: prometheus endpoint: string name not used in metrics

mpw96, 2023-03-02:
### What happened?

I have assigned names to the strings in the UI, bt the metrics for the panels show only the string numbers, not the names that I have assigned ("Westen" and "Osten"). Here's an example:

```
opendtu_Power{serial="114180126058",unit="0",name="Dach",type="DC",channel="0"} 0.500000
opendtu_Power{serial="114180126058",unit="0",name="Dach",type="DC",channel="1"} 0.500000
```

### To Reproduce Bug

Go to endpoint `/api/prometheus/metrics`.

### Expected Behavior

```
opendtu_Power{serial="114180126058",unit="0",name="Dach",type="DC",channel="Osten"} 0.500000
opendtu_Power{serial="114180126058",unit="0",name="Dach",type="DC",channel="Westen"} 0.500000
                                                                            ^^^^^^
```

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

a341c88

tbnobody, 2023-03-02:
it's not a bug its a feature to have the string numbers in there i would guess. because. Maybe easier to parse? Don't know. As mentioned before, i don't use the prometheus endpoint and don't know anything of the syntax etc. What do other users think?

easimon, 2023-03-05:
IMHO the tags of real metrics should not contain moving parts, since this would start new time series every time you rename the string. 

You could introduce a new pseudo metric similar to opendtu_build, that translates channel number to channel name (by having both tags). And then do a label join between those if you want to display them in a graph.

mpw96, 2023-03-06:
> IMHO the tags of real metrics should not contain moving parts, since this would start new time series every time you rename the string.

That's exactly what I would like to have. The names of the strings in the metrics' labels. I think renaming happens almost never after the initial setup of the panels. Could as well be an additional label like e.g. `channelName` or even better `stringName`.

easimon, 2023-03-07:
"almost never" is something that always happens when you least expect it :)

Anyway, `metric name + tags` is the ID of a time series. Any change to these labels ends a time series and starts a new one, making it hard to have consistent graphs across intervals when the change happened. You can do some tricks with `sum()` or `label_replace()` to reconnect them, but its ugly. Btw, this problem already exists for the "name" label. If I rename the Inverter, the series breaks apart.

The same is true for *adding* labels to existing metrics, but this would at least be ignorable with corresponding relabel rules.

Anyway, I would encourage you to at least keep the numeric (fixed) label, so it does not break existing series totally.
The preferred way would be to add a new constant-1 pseudo-metric like e.g. 

```
opendtu_panel_info{serial=\"%s\",unit=\"%d\",name=\"%s\",channel=\"%d\",channelname=\"%s\"} 1
```

and then do a join like described here 

https://ypereirareis.github.io/blog/2020/02/21/how-to-join-prometheus-metrics-by-label-with-promql/

easimon, 2023-03-08:
I have created a PR that exports panel name (and maxpower, offset) in #671

mpw96, 2023-03-11:
@tbnobody In the end you can do something like this with Prometheus (and Grafana), nice visualizations:

[image]

This is the edit mode of a grafana panel that displays the power over time of my two individual solar panels. I have marked what is possible now with @easimon's PR. I can now use the panelname as a label. Works perfectly for me (latest master + #671). I guess you could merge it... ;-)

tbnobody, 2023-03-17:
Going to close this issue as the information are available as a separate measurement as mentioned in the PR above

closed 2023-03-17 by tbnobody