tbnobody/OpenDTU #652: [Request]passwort vergessen

fabischr, 2023-03-04:
### Describe the solution you'd like

hallo zusammen
ab folgendes problem
wollte das standard passwort ändern, mein Ipad hat ein starkes `eigenes` passwort vergeben. Allerdings hat sich die seite geschlossen aber das passwort wurde nicht gespeichert. Kann man dass Passwort zurücksetzen ?

Solarteur, 2023-03-05:
Ein Passwort, das man - ohne es zu kennen - selbst zurücksetzen kann, macht aus meiner Sicht keinen Sinn. 
Also: Von Grund auf neu installieren, ggf. einschließlich "ERASE".

helgeerbe, 2023-03-05:
Du solltest nach dem Aufsetzen, die Config sichern. Dann ist das neue Aufsetzen auch zukünftig schnell gemacht. Übrings steht das Passwort in der gesicherten Config im Klartext (falls du es mal vergessen solltest). 

@tbnobody sicherer wäre es daher, dass Passwort zu hashen.

helgeerbe, 2023-03-05:
Stimmt, da habe ich gerade nicht dran gedacht. Das würde nur gehen, wenn man das Admin Passwort davon trennt und die Wifi Passwörter verschlüsselt.  Aber das macht es etwas aufwändiger.

tbnobody, 2023-03-06:
Wie bereits oben erwähnt kann das Passwort durch Löschen und erneutes flashen zurückgesetzt werden. Die Config kann ja ebenfalls entsprechend gesichert werden. (Muss geschehen bevor das Passwort vergessen wird, aber dann stünde das Passwort ja auch in der Config). Ein feature zu Implementieren macht hier meiner Meinung nach keinen Sinn.

closed 2023-03-06 by tbnobody

Kalle0709, 2024-03-13:
Moin, bin neu und checke leider überhaupt nichts. 

Habe mein pw automatisch ausfüllen lassen an meiner DTU. Leider hat der Laptop es nicht richtig im Schlüsselbund abgelegt. Komme also nicht mehr ins Menü um weitere Wechselrichter anzutragen. 

Bin komplett neu in dieser Welt. Habe es jetzt unter Windows schon einmal geschafft das er den Controller unter COM3 erkennt aber mit dem neu flaschen klappt einfach nicht. 

Flash downloader habe ich runter geladen und installiert aber ich finde nicht die richtigen Dateien zum aufspielen. 

Und ob es dann klappt bin ich mir auch nicht so sicher. Kann mir einer von euch helfen?