tbnobody/OpenDTU #654: [Request] Option to rotate display output by 180 degrees

ikitzmann, 2023-03-05:
### Is your feature request related to a problem? Please describe.

Sometimes you are restricted on the orientation you incorporate the display into your OpenDTU case/board. Or you need to flip the case/board (because your usb power comes from the other side etc.)

### Describe the solution you'd like

It would be great to have an option to rotate the display content by 180 degrees. Ahoi DTU has such a setting.

### Describe alternatives you've considered

The only other way is to physically rotate the display - which might not always possible.

apf-doit, 2023-03-06:
i confirm the display rotation of my SSD1306 which now is upside down without an option to reverse that which was correct in at least version 97bc964.

ikitzmann, 2023-03-06:
I forgot to mention that I am using a SSD1306 display.

I did a look at the display code and the library has an easy way to rotate the screen image:
In the constructor https://github.com/tbnobody/OpenDTU/blob/master/src/Display_Graphic.cpp#L35 the first argument defines the display orientation.
See https://github.com/olikraus/u8g2/wiki/u8g2reference
U8G2_R0 = No rotation, landscape
U8G2_R2 = 180 degree clockwise rotation

Additionally there is a method setflipmode (https://github.com/olikraus/u8g2/wiki/u8g2reference#setflipmode) to rotate the image later.

This should be quite doable to add.

madmartin, 2023-03-11:
@tbnobody Should I try to implement this and file a PR?

tbnobody, 2023-03-17:
Implemented in 2e33f5cd51ca201abea1753d7aa1322371186fdd

closed 2023-03-17 by tbnobody