tbnobody/OpenDTU #655: OpenDTU hängt sich beim Limit setzen über NodeRed/MQQT auf

TheGrumpyBadger, 2023-03-05:
### What happened?

Statt eines Phyton Scriptes möchte ich die Nulleinspeisung über NodeRed realisieren da das auf dem VenusOS schon mit drauf ist.
Den Flow lege ich bei.
Das läuft auch an, aber nach  ein paar Sekunden friert der HM-350 mit 0 W ein und ich muss den WR und den ESP rebooten damit es wieder läuft.
ESP und HM-350 sind etwa 15cm voneinander entfernt.

### To Reproduce Bug

Flow liegt bei.

### Expected Behavior

WR sollte entsprechend dem Power des Shelly 3EM geregelt werden.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

b42fff8

### Anything else?

[flows.json.txt](https://github.com/tbnobody/OpenDTU/files/10892420/flows.json.txt)

tbnobody, 2023-03-05:
Wie oft setzt du denn das Limit? Was siehst du auf der Seriellen Console?

TheGrumpyBadger, 2023-03-05:
Alle 15 Sekunden. Wie es aussieht schreibe ich aber ab und an negative Werte über MQQT, das kann natürlich nicht gut gehen.
Fehler sitzt wohl vor dem Bildschirm, die Konsole hatte ich nicht beachtet.

tbnobody, 2023-03-06:
Feel free to open it again if the problem persists with valid values

closed 2023-03-06 by tbnobody

jonnycastaway, 2023-04-13:
> Ich versuche auch das Limit über Node Red zu setzen, leider habe ich es nicht geschafft, da ich nur einen curl Befehl dazu gefunden habe. Bisher habe ich folgendes versucht in einem Node Red Function Node:
> 
> msg.method = "post"; msg.url = "http://192.168.178.70/api/limit/config"; msg.data = {"serial": "114182xxxxxx", "limit_type": 1, "limit_value": 50}; msg.headers = {"Content-Type":"application/x-www-form-urlencoded"}; msg.auth = {"admin":"xxxxxx"};
> 
> return msg;

Falls Du/ihr noch Hilfe brauchst, hier läuft es schon seit etlichen Wochen mit Node-Red ;-)

ThomasCr, 2023-10-09:
Hi, ich habe scheinbar das selbe Problem mit NodeRED. Gibt es da ein Zeit-Limit bzgl. setzten des Tempären Limits?
Im Inverter-Log steht mehrfach:

14:18:11 | 14:18:11 | 2 | DTU command failed

Ich habe jetzt schonmal die anzahl der Nachrichten auf 1msg/30sec beschränkt. Aber das scheint das Problem nicht zu beheben.

Andre7tyone, 2023-10-09:
Die Werte müssen im Bereich 5 bis 100 Prozent liegen und müssen ganzzahlig sein.
Damit nicht zu viele Werte an den Hoymiles gesendet werden, frage ich ab, ob der neue zu übertragene Wert gleich dem alten Werte ist bzw. sich in einem Fenster von +/- 3 Prozent befindet. Trifft das zu, wird der Wert nicht erneut übertragen. Dadurch hat sich die Anzahl der übertragenen Werte deutlich verringert.

jonnycastaway, 2023-10-16:
> Hi, ich habe scheinbar das selbe Problem mit NodeRED. Gibt es da ein Zeit-Limit bzgl. setzten des Temporären Limits? Im Inverter-Log steht mehrfach:
> 
> 14:18:11 | 14:18:11 | 2 | DTU command failed
> 
> Ich habe jetzt schonmal die Rate der Nachrichten auf 1msg/30sec beschränkt. Aber das scheint das Problem nicht zu beheben.
> 
> EDIT: Evtl. habe ich das Problem gefunden - ich habe mit NodeRED float-Werte gesendet - ich wandle gerade vorher um mit der JS Funktion `parseInt()` - ich werde beobachten und nochmal Rückmeldung geben.

Ich habe festgestellt das ein setzen des Limits per API mit % deutlich schneller geht als mit Watt.
Und die %-Angabe sollte natürlich kein Decimal sein.

killertomate94, 2024-01-21:
@jonnycastaway kannst du dazu mal den flow zur verfügung stellen?