tbnobody/OpenDTU #661: [Request] Basic nRF24L01+ transmission statistics

mawa85, 2023-03-06:
### Is your feature request related to a problem? Please describe.

Just fine-tuning of openDTU Hardware placement and transmission power configuration

### Describe the solution you'd like

Display the percentage of successful or failed transmit in System - Radio Information

glucosum, 2023-04-21:
Maybe something like RSSI and Quality but for NRF24...

Vujagig, 2023-04-29:
Just like in AhoyDTU, so then you can see problems with the radio connection

stefan123t, 2024-07-10:
There is a packet Receive / Sent counter on the Inverter side.

#### [GetLossRate | `0x15`](https://github.com/lumapu/ahoy/wiki/Protocol#getlossrate--0x15)

The GetLossRate `0x15` command should return information which could be useful for a RX/TX statistic of the inverter.

TX 15 72615582 78563411 80 1500 62E830E7 0000 00000000 0000 3B54 7D    GetLossRate 0x15
RX 95 72615582 72615582 81 9B26 0930 8FC8 D7                           UsartNrf3_Process_DevInform_GetLossRate()
                           ^^^^--------------------------------------- MI_CF_Num_Start/End Micro-inverse receive count
                                ^^^^---------------------------------- MI_RF_Num_Start/End Micro-inverse send count

You would need to remember / initialize the Values on the OpenDTU side, in case these are (still) 0.

After these Start Values can be subtracted from the current (End) Values returned by the GetLossRate command and this would yield the RX/TX Difference for the past measurement period.

tbnobody, 2024-10-03:
Radio statistics are implemented in a54b19bf5bb5b3a4771f29e961d87cd2119f4e74

closed 2024-10-03 by tbnobody