tbnobody/OpenDTU #663: Error when compiling with ESP32-C3

ss89, 2023-03-06:
### What happened?

I tried to compile OpenDTU for ESP32-C3

### To Reproduce Bug

Add a configuration to platformio.ini like:

```
[env:seeed_xiao_esp32c3]
; https://wiki.seeedstudio.com/XIAO_ESP32C3_Getting_Started
board = seeed_xiao_esp32c3
build_flags = ${env.build_flags}
    -DHOYMILES_PIN_MISO=9
    -DHOYMILES_PIN_MOSI=10
    -DHOYMILES_PIN_SCLK=8
    -DHOYMILES_PIN_IRQ=1
    -DHOYMILES_PIN_CE=2
    -DHOYMILES_PIN_CS=3
```

### Expected Behavior

Compile should succeed

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

b42fff8

### Relevant log/trace output

```Shell
.pio/libdeps/esp32-c3-devkitm-1/ESP Async WebServer/src/AsyncWebSocket.cpp: In member function 'IPAddress AsyncWebSocketClient::remoteIP() const':
.pio/libdeps/esp32-c3-devkitm-1/ESP Async WebServer/src/AsyncWebSocket.cpp:740:28: error: call of overloaded 'IPAddress(unsigned int)' is ambiguous
         return IPAddress(0U);
                            ^
In file included from /Users/sebastian/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:170,
                 from .pio/libdeps/esp32-c3-devkitm-1/ESP Async WebServer/src/AsyncWebSocket.cpp:21:
/Users/sebastian/.platformio/packages/framework-arduinoespressif32/cores/esp32/IPAddress.h:51:5: note: candidate: 'IPAddress::IPAddress(const uint8_t*)'
     IPAddress(const uint8_t *address);
     ^~~~~~~~~
/Users/sebastian/.platformio/packages/framework-arduinoespressif32/cores/esp32/IPAddress.h:50:5: note: candidate: 'IPAddress::IPAddress(uint32_t)'
     IPAddress(uint32_t address);
     ^~~~~~~~~
/Users/sebastian/.platformio/packages/framework-arduinoespressif32/cores/esp32/IPAddress.h:29:7: note: candidate: 'constexpr IPAddress::IPAddress(const IPAddress&)'
 class IPAddress: public Printable
       ^~~~~~~~~
```
```

### Anything else?

Project owner already knows this issue and it is related to a dependency of this project, See: https://github.com/yubox-node-org/ESPAsyncWebServer/issues/9

This issue is purely for awareness of people interested in this project and wanting to use the C3 as well as documenting when it's build process is fixed.

ss89, 2023-03-06:
@tbnobody Seeing that the issue on the dependency was not getting any responses from the repo owner and the last commit being from almost 1 year ago: Would it make sense to fork the repo and apply the needed fixes?
Or would it make sense to try out other forks?
I am not into ESP32 yet at all and am just starting to tinker a tiny bit, so i have no experience with the common used libraries and their pitfalls.
I do however know from the arduino libraries that things can regularly break and see if you'd rather not want to switch the dependency origin.

ss89, 2023-04-09:
I do see issues when not having the esp32-c3 connected to my laptop but instead using a usb power supply. It seems that the firmware(?) requires the log messages to be read. At least the firmware does only seem to "start" when i use the monitor command of platformio in vscode. Is there a way around?

tbnobody, 2023-05-03:
Just add the following to your profile:
`custom_patches = esp32c3`

This applies all patches in the `patches/esp32c3` directory to the libraries and fixes this issue.

closed 2023-05-03 by tbnobody