tbnobody/OpenDTU #667: Ohne NRF24L01+, nur mit Olimex ESP32-POE-ISO oder anderem PoE-Board

HalliHalloSchatz, 2023-03-07:
### Is your feature request related to a problem? Please describe.

Ist es möglich, nur mit dem ESP32-POE Board eine Verbindung zum Wechselrichter auf zu bauen?

Das Board wird ja bereits mit Ethernet verbunden und hat die freie WLAN Verbindung. Kann diese genutzt werden, sodass man sich das NRF24L01+ sparen kann?

Danke 

### Describe the solution you'd like

Eine Variante, die mit einem PoE-Board funktioniert, sodass das zweite WLAN Modul entfallen kann.

tbnobody, 2023-03-07:
Nein, nur weil sowohl das NRF24 Modul als auch WLAN 2,4GHz verwenden bedeutet das nicht dass das gleiche Protokoll darüber gesprochen wird. Das NRF Modul ist auf jeden Fall nötig.

closed 2023-03-07 by tbnobody

HalliHalloSchatz, 2023-03-08:
Danke für die schnelle Antwort.

Gibt es da weitere Informationen darüber, wieso das interne WLAN nicht die NRF24L01+ ersetzen kann?
Interessiert mich gerade, was die NRF24L01+ mehr kann, was der internen WLAN Karte fehlt.

x90cr, 2023-03-12:
Übrigens läuft das auch mit dem Olimex MOD-NRF24LR problemlos, der wird als NRF24L01+ erkannt.