tbnobody/OpenDTU #670: Incorrect hiding of unused strings in Live View

PATROMO, 2023-03-08:
### What happened?

As the title says: Incorrect hiding of unused strings in Live View

Look at screenshots

[image]

[image]

Bug is from 220884f72ee331cf4fc0ecb30c33feccd7eca560. 

### To Reproduce Bug

Open Live View

### Expected Behavior

Look at screenshot

[image]

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

cbbbc21

tbnobody, 2023-03-08:
What shows the setting of your inverter? (especially the max power for each string)

PATROMO, 2023-03-08:
@tbnobody this is just an HTML markup error. Look at my Screenshort in "Expected Behavior". 
My first two strings are unused :)

tbnobody, 2023-03-08:
Ahhh I see... problem seems to be this line:
https://github.com/tbnobody/OpenDTU/blob/cbbbc211343cc01dab89f684e3bc9df2ad8e16cb/webapp/src/views/HomeView.vue#L97

the column div is always output even if the innerhtml is later on hidden... will fix it this soon.

closed 2023-03-08 by tbnobody

PATROMO, 2023-03-09:
Thank you, it works.

commit 61795172 (2023-03-08): Fix #670: Incorrect hiding of unused strings in Live View