tbnobody/OpenDTU #674: PCD8544 display not working and wrong implementation of display contrast

nobodyman1, 2023-03-08:
### What happened?

I tried to use a PCD8544 display but it remains blank.
After testing https://github.com/tbnobody/OpenDTU/pull/650 and a lot of investigation the display shows some data.
But I had to modify the code: `Display.contrast = 255`

### To Reproduce Bug

Connections:

| ESP32 | PCD8544   |
| ----- | --------- |
| IO13  | DN (MOSI) |
| IO14  | CLK       |
| IO15  | CS        |
| IO22  | DC        |
| IO25  | RESET     |

### Expected Behavior

The range of display contrast in the webinterface is 0 ... 100%.
This value is used without any modification to set display contrast in the u8g2 library. 
But I found in the u8g2 specification:
` value` : Contrast or brightness from 0 to 255

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

055eb3a

dAjaY85, 2023-03-17:
Nokia don't support contrast change, if contrast is setted, display stays off. In this case, I implemented in my fork, if contrast is 255, the contrast will not setted.

tbnobody, 2023-03-18:
According to the data sheet it does support setting the contrast. Even the library supports it: https://github.com/olikraus/u8g2/blob/4d9f63394f070a9f6e079ae8bccb97bad5967481/csrc/u8x8_d_pcd8544_84x48.c#L124

With my patch above the contrast value from 0...100 is spread from 0...255... But if it's still not working there is maybe a mistake in the display library. will doublecheck this later.

dAjaY85, 2023-03-18:
I have testet it, without backlights, maybe with contrast you can adjust the blacklights?

nobodyman1, 2023-03-21:
I tried firmware version [d508b41](https://github.com/tbnobody/OpenDTU/commits/d508b41).

```
build_flags = ${env.build_flags}
    -DHOYMILES_PIN_MISO=19
    -DHOYMILES_PIN_MOSI=23
    -DHOYMILES_PIN_SCLK=18
    -DHOYMILES_PIN_IRQ=26
    -DHOYMILES_PIN_CE=21
    -DHOYMILES_PIN_CS=5
    -DDISPLAY_TYPE=1
    -DDISPLAY_DATA=22 ; D/C
    -DDISPLAY_CS=15
    -DDISPLAY_RESET=25
```

Display connections
| ESP32 | PCD8544   |
| ----- | --------- |
| IO23  | DN (MOSI) |
| IO18  | CLK       |
| IO15  | CS        |
| IO22  | DC        |
| IO25  | RESET     |

The display shows some text (OpenDTU) but the contrast setting is very bad.
If I try to change the default contrast (which is 60%) to another value and press the save button an error message is shown:
`Profil muss zwischen 1 und 63 Zeichen lang sein!`

mischerr, 2023-03-26:
Today I'm back home and checked your settings - yes it's working, but I only see "OpenDTU!" msg with good contrast at 50%.
But how to see the live data?
On the web interface I can see that my HM600 is producing fine.

nobodyman1, 2023-03-26:
> Today I'm back home and checked your settings - yes it's working, but I only see "OpenDTU!" msg with good contrast at 50%. But how to see the live data? On the web interface I can see that my HM600 is producing fine.

I´m sorry, I only prepared my OpenDTU but I don´t have a Hoymiles device to test.

nobodyman1, 2023-04-07:
[v24.4.6](https://github.com/tbnobody/OpenDTU/releases/tag/v23.4.6) seems to work without any modification.
Thanks a lot!!

The date output in the last line doesn´t look very nice. It would be better to have hours and minutes instead of the day (`%a`).

dAjaY85, 2023-04-07:
Please look at the implementation of the displays at the ahoy Projekt. This steps are already done there.

Would be nice, if OpenDTU will get the same state with the displays like Ahoy. Code is there, need smaller adjustments.

If the implementation is needed, I can send a new PR.

ms-doneck, 2023-04-07:
Here it's already running fine. I'm not very familar using GIT, but I try to commit my changes under my private account. Use it as you like.

I did some renaming in the displays constructor to clearfy the pin names and I also did a pin-mapping file and some sketches.
I changed the date time format to "%d.%m.%y %T" if its not a "large" display.
The reset pin is not needed - so I used ths for the DC pin.

platformio_override.ini:

`[env:esp32_pcd8455]
board = esp32dev
build_flags = ${env.build_flags}
    -DHOYMILES_PIN_MISO=19
    -DHOYMILES_PIN_MOSI=23
    -DHOYMILES_PIN_SCLK=18
    -DHOYMILES_PIN_IRQ=16
    -DHOYMILES_PIN_CE=4
    -DHOYMILES_PIN_CS=5
    -DDISPLAY_TYPE=1
    -DDISPLAY_DATA=12  ; DIN 
    -DDISPLAY_CLK=14   ; CLK
    -DDISPLAY_CS=26    ; CE
    -DDISPLAY_RESET=13 ; DC`

dAjaY85, 2023-04-07:
look here:
https://github.com/dAjaY85/OpenDTU

closed 2025-04-18 by github-actions[bot]