tbnobody/OpenDTU #676: Keine Anzeige der LiveDaten aber MQTT Befehle werden übermittelt

DPO99, 2023-03-09:
### What happened?

Die Live-Anzeige wartet seit 1500 Sekunden auf eine Verbindung, währenddessen werden aber MQTT Befehle über IoBroker an den Hoymiles übermittelt und ausgeführt(dies kann ich an der entsprechenden Leistung am Shelly messen, welcher am Ausgang des WR hängt.

### To Reproduce Bug

Fehler ist nicht lösbar, von daher auch nicht produzierbar. Meine Vermutung liegt darin, dass die MQTT Befehle ggf. den WR oder OpenDTU überfordern.

### Expected Behavior

Trotz Sendung mehrere MQTT Befehle alle 10-15 Sekunden sollte sich die Live-Anzeige aktualiseieren.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

055eb3a

DPO99, 2023-03-09:
Update. Nach 1800 Sekunden hat sich die Anzeige nun doch aktualisiert, hängt aber nun wieder etwas fest (aktuell bei 500 Sekunden).

Ein Zweiter Wechselrichter läuft nun auch immer in Richtung 300+ Sekunden. In der Regel hatte dies sonst immer nur ca. 120 Sekunden gedauert.

mroenne2022, 2023-03-09:
@DPO99 : Ich hab seit heute mittag auch die Version auf meinem ESP32 WROOM mit 38Pin plus NRF mit interner Antenne am laufen. Ich nutze ihn ebenfalls für 2 HM WR, allerdings nur die WebGUI, kein MQTT! Hab keinerlei Probleme, alles lief prima, wurde auch laufend aktualisiert.

tbnobody, 2023-03-10:
> Meine Vermutung liegt darin, dass die MQTT Befehle ggf. den WR oder OpenDTU überfordern.

Da liegst du korrekt. Die MQTT Steuerbefehle werden so schnell abgearbeitet wie möglich. Wenn jedoch neue MQTT Steuerbefehle empfangen werden bevor das RF Modul im Idle Modus ist werden keine Status Requests in die TX Queue aufgenommen. Du kannst das Problem also lösen indem du die Steuerbefehle in längeren Zeitabständen sendest.

DPO99, 2023-03-10:
Danke!

DPO99, 2023-03-10:
Danke

closed 2023-03-10 by DPO99

markusruebe, 2023-07-06:
Hi, may I also raise the issue that - since I started to implement Home Assistant my openDTU does not show LIVE Data and also MQTT Settings cannot be accessed. Network Settings, Info etc. are working (with a litte delay) but LIVE page ist not shown and also MQTT cannto be accessed. I am using the up tp date bin file and installed an OLED 3 days ago - after that all was fine. since yesterday I try Home Assistant and have issues. 
Any idea? dependency? proposal?

Thanks a lot.

MQTT data is still transmitted.