tbnobody/OpenDTU #686: Code size to large

EckmarSchmitz, 2023-03-12:
### What happened?

Getting the Git code and compile with "d1 mini esp32" I get the following error message:

Checking size .pio\build\d1 mini esp32\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  17.4% (used 57160 bytes from 327680 bytes)
Error: The program size (1706889 bytes) is greater than maximum allowed (1572864 bytes)
Flash: [=====*** [checkprogsize] Explicit exit, status 1
=====]  108.5% (used 1706889 bytes from 1572864 bytes)
============================================ [FAILED] Took 38.87 seconds ============================================

Environment    Status    Duration
-------------  --------  ------------
d1 mini esp32  FAILED    00:00:38.873
======================================= 1 failed, 0 succeeded in 

With another ESP8266 board I get the same problem.
How to change the file size?

### To Reproduce Bug

Compile code from git hub with visual studio code

### Expected Behavior

binary should be generated.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

?

### Relevant log/trace output

```Shell
Checking size .pio\build\d1 mini esp32\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  17.4% (used 57160 bytes from 327680 bytes)
Error: The program size (1706889 bytes) is greater than maximum allowed (1572864 bytes)
Flash: [=====*** [checkprogsize] Explicit exit, status 1
=====]  108.5% (used 1706889 bytes from 1572864 bytes)
============================================ [FAILED] Took 38.87 seconds ============================================

Environment    Status    Duration
-------------  --------  ------------
d1 mini esp32  FAILED    00:00:38.873
======================================= 1 failed, 0 succeeded in
```

tbnobody, 2023-03-12:
Maybe you have done some own modifications to the souce code. The original code compiles without any problems and also fits into the flash.

EckmarSchmitz, 2023-03-12:
Hallo tbnobody,
ich habe nochmals über VisualStudioCode das komplette Projekt aus Github neu geholt. Keine Änderungen gemacht nur neu über Ausführen/Debug starten das Projekt kompiliert.

Das ist die Ausgabe komplett:
*  Task wird ausgeführt: C:\Users\mail\.platformio\penv\Scripts\platformio.exe debug

Processing generic (board: esp32dev; framework: arduino; platform: espressif32@>=6.0.1)
---------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Firmware Revision: g055eb3a
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20007.0 (2.0.7)
- tool-esptoolpy @ 1.40500.0 (4.5.0)
- tool-openocd-esp32 @ 2.1100.20220706 (11.0)
- toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 45 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP Async WebServer @ 1.2.3+sha.1dde9cf
|   |-- AsyncTCP @ 1.1.1
|   |-- ArduinoJson @ 6.20.1
|   |-- FS @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- ArduinoJson @ 6.20.1
|-- espMqttClient @ 1.3.3+sha.6a3f3a8
|   |-- AsyncTCP @ 1.1.1
|   |-- WiFiClientSecure @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- RF24 @ 1.4.5
|   |-- SPI @ 2.0.0
|-- U8g2 @ 2.34.15
|   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|-- sunset @ 1.1.7
|-- LittleFS @ 2.0.0
|   |-- FS @ 2.0.0
|-- Hoymiles @ 0.0.1
|   |-- RF24 @ 1.4.5
|   |   |-- SPI @ 2.0.0
|   |-- Every
|   |-- SPI @ 2.0.0
|   |-- TimeoutHelper
|-- Ethernet @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- Update @ 2.0.0
|-- ResetReason
|-- DNSServer @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- WiFi @ 2.0.0
|-- MqttSubscribeParser
|   |-- espMqttClient @ 1.3.3+sha.6a3f3a8
|   |   |-- AsyncTCP @ 1.1.1
|   |   |-- WiFiClientSecure @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- Ticker @ 2.0.0
Building in debug mode
Converting .pio\build\generic\index.html.gz.txt.o
Converting .pio\build\generic\zones.json.gz.txt.o
Converting .pio\build\generic\favicon.ico.txt.o
Converting .pio\build\generic\app.js.gz.txt.o
Compiling .pio\build\generic\src\Configuration.cpp.o
Compiling .pio\build\generic\src\Display_Graphic.cpp.o
[... 297 lines cut ...]
Archiving .pio\build\generic\libFrameworkArduino.a
Linking .pio\build\generic\firmware.elf
Retrieving maximum program size .pio\build\generic\firmware.elf
Checking size .pio\build\generic\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  17.4% (used 57160 bytes from 327680 bytes)
Error: The program size (1706737 bytes) is greater than maximum allowed (1572864 bytes)
Flash: [=====*** [checkprogsize] Explicit exit, status 1
=====]  108.5% (used 1706737 bytes from 1572864 bytes)
============================================ [FAILED] Took 39.90 seconds ============================================

Environment    Status    Duration
-------------  --------  ------------
generic        FAILED    00:00:39.903
======================================= 1 failed, 0 succeeded in 00:00:39.903 =======================================

*  Der Terminalprozess "C:\Users\mail\.platformio\penv\Scripts\platformio.exe 'debug'" wurde mit folgendem Exitcode beendet: 1.
 *  Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.

Das Ergebnis ist das Gleiche wie mit den Änderungen für das D1 Mini ESP32. Was ist da falsch?
Visual Studio Code ist nicht meine normale Entwicklungsumgebung. Ich benutze für meine Projekte das normale Visual Studio 2022.
Ich hoffe du kannst mir helfen warum der Code nicht in den Microcontroller passen soll. Eigentlich wäre ja genügend ROM verfügbar.

Gruß
Eckmar

tbnobody, 2023-03-13:
> *  Task wird ausgeführt: C:\Users\mail\.platformio\penv\Scripts\platformio.exe debug

You are building with debug symbols. This leads to a larger code size (also the timings are different). I don't know which button you are pressing to compile the source, but its the wrong one :-)

helgeerbe, 2023-03-13:
@tbnobody I know this might be a breaking change. But is it possible to change `partitions_customs`and take 200kB from the file system? So can gain additional 100kB for the firmware? 

While writing I see, that debug build needs additional 130 kB. So this will not work. But to extend the firmware partition would help for further extensions.

tbnobody, 2023-03-13:
> @tbnobody I know this might be a breaking change. But is it possible to change partitions_customsand take 200kB from the file system? So we can gain additional 100kB for the firmware?

Yes that will be needed in a nearby future. But as you said, it will be a very breaking change. All users have to re-flash everything using the USB port. This is why I would postpone this as long as possible. Also the further integration of HMS/HMT inverters will take some memory. But currently it still fits.

helgeerbe, 2023-03-13:
Well my OpenDTU-onBattery has 22kB left for the sketch. I would vote for as soon as possible. ;)

tbnobody, 2023-03-14:
I am going to close this issue as it referes only to the debug build

closed 2023-03-14 by tbnobody