tbnobody/OpenDTU #696: No continuous polling

m-froehlich, 2023-03-14:
### Is your feature request related to a problem? Please describe.

If I am not using MQTT, but polling fresh values from a supervising data collector, the OpenDTU should not poll every x seconds itself, and that the whole night as well.

### Describe the solution you'd like

For one there should be a sunset/sunrise detection for my location built in and if enabled, no polling during night time.

And second there should be an option to be set, where the DTU does not poll fresh values every x seconds, but only if it is explicitly asked for fresh values via the WebAPI or REST or what ever channel and only ask the Hoymiles instantly then.

I could even imagine an API call just to trigger reading new values, but not to return them. This way control over polling behavior could be externalized.

Of course when the LiveView is open, fresh values still have to be fetched every x seconds.

### Describe alternatives you've considered

Having a feature like this would render the OpenDTU perfect for me. With this feature missing I would prefer to use the original DTU, where I can poll fresh live instant values whenever I query them, which makes the data a bit more reliable and live.

LucaDev, 2023-03-15:
Regarding the manual polling: Is a 2.5s (mean with a polling rate of 5s) old data point really that bad for you? I don't see an immediate benefit over the current system.

m-froehlich, 2023-03-16:
Polling times differ a bit as it seems. Some refreshes are very quick and some aren't. Sometimes the last available value set ist 45 seconds old even though I configured a delay of 5 seconds. That's one point.

The other point is, that the OpenDTU should be considered as the provider for PV values. And if MQTT is not my way to go, a fixed and internal refresh rate doesn't make much sense. The values should be fetched when and only when they're actually requested, which the OpenDTU does not decide here.

While data is only recorded a much lower refresh rate might be good enough. While showing live view a high rate is desiarable.

m-froehlich, 2023-03-16:
Of course currently the OpenDTU does not currently shut down during night time and keeps on polling. While this is the case, it is even more beneficial to leave refresh decisions up to an external entity.

tbnobody, 2023-03-16:
Just turn off polling during the night. If you set the poll interval to 5 seconds and get updates only every 50 seconds (with one inverter), then your RF quality is just too bad. Look at the console. The TX packets are sent every n seconds. I no response is coming from the inverter, there is of course no update via mqtt or the web interface.

tbnobody, 2023-03-16:
Within the inverter settings. I cannot create a screenshot right now but click on Settings --> Inverter --> the pen to edit the inverter --> There are 4 switches which define when the inverter is polled and when commands are sent.

m-froehlich, 2023-03-16:
Got it. Thanks. I searched for it on DTU level.

stefan123t, 2024-07-10:
@m-froehlich I assume this issue can be closed as you have been able to set it up according to the instructions given ?

### Settings > Inverter > Edit inverter:

#### General
**Receive / Send**
Poll inverter data: [x]
Poll inverter data at night: [ ]
Send commands: [x]
Send commands at night: [ ]

**Hint**: The inverter is powered by its DC input. If there is no sun, the inverter is off. Requests can still be sent.

closed 2024-10-03 by tbnobody