tbnobody/OpenDTU #697: [Sunrise/sunset] API extension point

m-froehlich, 2023-03-15:
### Is your feature request related to a problem? Please describe.

OpenDTU could improve with sunrisd/sunset calculation.

### Describe the solution you'd like

Once the OpenDTU calculates sunrise/sunset it should communicate the two times for the following evening and morning via WebAPI and an extra extension point. This would help not to calculate it multiple times on multiple devices in the same eco system.

LucaDev, 2023-03-15:
Calculating sunset times isn't that hard. It barely requires computing power.
I don't see an immediate benefit here. What's your use case?

m-froehlich, 2023-03-15:
The use case is for one exactly saving computing power and energy. Second I don't see a need for redundant computation of the same thing more than once.

And you wouldn't need to look up the code for this computation, if you need that information somewhere else in the same network while the information is there ready to be used.

ehofm3, 2023-03-15:
I'm pretty sure that a request of this data via WiFi requires more power than the quick, redundant computation.
But since that doesn't seem to bother you, you can also request the calculated sunrise/sunset values via public web APIs, e.g. [https://api.sunrise-sunset.org/json?lat=36.7201600&lng=-4.4203400](url)

In general, that's a good thought, but it doesn't make much sense since we have the Internet (and the focus of the project is completely different).

LucaDev, 2023-03-15:
I completely agree with @ehofm3

While it isn't a huge deal to add it, openDTU isn't a sun calculator service. It should focus on providing the current energy data. Computing the sunrise won't even use a Wattsecond of power. It could be convenient for some people to have it included though. I'd keep it in the backlog but with a very low priority. There are much more relevant tasks right now.

m-froehlich, 2023-03-16:
Actually my first approach was to find a free and reliable webservice and couldn't find one. Now that there is one, it is less important.

Though yes, it is not a big deal to add it. So why not provide the available (local)  information? I agree about the priority.

tbnobody, 2023-03-16:
Due to the fact that the frontend is completly decoupled from the backend, all values shown in the web interface are also available via WebAPI endpoints.... In this case: http://<ip>/api/ntp/status
You have to parse the values yourself as in terms of the web api, the main focus is the frontend and the easy handling there.

closed 2023-03-16 by tbnobody