tbnobody/OpenDTU #707: [Request] Set grid over-voltage limit

hbarthel, 2023-03-17:
### Is your feature request related to a problem? Please describe.

As the grid AC voltage may be 230V +/- 23V my HM-600 detects over voltage sometimes and goes to 0 Watt. For Germany the default seems to be 253V. To get back to 600 Watts it takes about 12 minutes.

### Describe the solution you'd like

I know the grid type can be changed with the official DTU to increase the cut-off voltage to 255V. May be you can find out how this works and can be implemented in OpenDTU.

### Describe alternatives you've considered

1. Maybe a surge protection plug could help
2. Complain about high voltage at grid carrier :)
3. Use a different phase

But I think the little too high voltage at the Hoymiles can also be caused by my house-installation.

Solarteur, 2023-03-17:
OpenDTU is a tool to **read** data from Hoymiles HM-xxx inverters.
If you already suspect that your electrical installation is the reason for the (actually very) high voltages in the grid at home, check the installation or have the installation checked.

hbarthel, 2023-03-17:
Well... guess it can also write, e.g. set inverter power. I'll see what the experts say...

homeautomation2022, 2023-03-17:
Unsere Hoymiles kommen mit der Voreinstellung: https://github.com/lumapu/ahoy/wiki/Protocol#file-name-de_vde4105_2018
Ändern kann das derzeit nur die originale DTU und wird auch zeitnah kein Feature dafür kommen. Ist auch unnötig, denn die Landesvorgaben sollten auch eingehalten werden. Eventuell ist es auch ein HM aus einem anderen Land, dann gelten andere Regeln, die allgemeinen EU Stromnetzbestimmungen sind natürlich die gleichen.

hbarthel, 2023-03-29:
Nach dem Dokument dürfte er erst bei 287.5V abschalten bzw. bei 253V Durchschnitt in 10 Minuten. Das ist bei meinem leider nicht so und daher vermute ich, hat er ein falsches GridProfile. Hab ihn von www.gartenkraftwerke.de

mmz06, 2023-04-20:
Hi there, 
Please accept my apologies as I interfere here and therefore, I'm just a user, not a developper, so it's a pure coincidence I met this topic of interest. 
According to [this online document from HM](https://www.hoymiles.com/wp-content/uploads/2022/11/Technical-Note-How-to-set-Hoymiles-3Gen-Grid-Profile-V1.2.pdf), it seems the default profile configuration limits the maximum Voltage to 253V instead of 287.5V as mentioned by @homeautomation2022.
This may explain the behaviour seen by @hbarthel.
Now according to the same doc, it seems you need to access "Hoymiles Monitoring Platform", through an official HM DTU to be able to do this change by yourself...

bromic1, 2023-07-13:
Hi @robbe-23 , can you pls reconfirm that you used the DTU WLite for that? When I want to order at Shinetech they state me that the WLite is only for monitoring not for setting anything. I am also struggeling with grid overvoltage.
Thank you.

robbe-23, 2023-07-13:
yes, it worked just fine (see screenshot above). The only difficulty was that the WLite can only be used to configure one inverter, and I have five.

stefan123t, 2024-07-10:
@hbarthel the Feature Request has been described in #900 specifically [detailled here](https://github.com/tbnobody/OpenDTU/issues/900#issuecomment-2209546838).

There the Feature Request is to set a known Grid Profile and send it to the inverter.
A second Step would be, to allow changing the values in a Grid Profile within the allowed Ranges, in order to modify the Grid Profile individually.

robbe-23, 2024-07-10:
> Hi @robbe-23 , can you pls reconfirm that you used the DTU WLite for that? [...]
Yes, I also got mine from Shinetech. The difficulty was that I could only connect one HM at a time. Other than that, no problem setting the grid profile. Actually, I have never touched the WLite again after that, so it is has been resting for a year. If there have been any changes since I am not aware of them.

abuesen, 2024-07-21:
@homeautomation2022 
> Ich glaube nicht, dass dein Stromversorger wirklich für längere Zeit 240V+ auf der Leitung hat!

Seit gestern habe ich einen HMT-2250 zusätzlich in Betrieb. Profil ist EN 50549-1:2019. Hab dadurch erst auf die Spannung geschaut und tatsächlich zeigt die Messung am EHZ bereits Werte von 250V+, selbst wenn ich beide PV-Systeme trenne. Die historischen Werte zeigen regelmäßig 250V an sonnigen Tagen.

Laut meinem Elektriker muss das auf VDE umgestelt werden, eben genau wegen der Spannung und weil laut TAB ein VDE-Profil vorgeschrieben sei.