tbnobody/OpenDTU #711: Model HM-800

tabacha, 2023-03-17:
### What happened?

The Software tells me to fill a bug and tell you that I use a HM-800

[image]

### To Reproduce Bug

I Click on the Button: "Zeige Wechselrichter Informationen"

### Expected Behavior

No Message that i should fill a bug.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

4c3172f

homeautomation2022, 2023-03-17:
Bitte nochmal DTU neustarten(bei Tag natürlich), dann klappt es vermutlich, anscheiennd wurden Daten falsch oder nicht geparst, da Teilenummer und Version eine 0 drin steht. Der Rest passt komischerweise, also das Packet mit den Daten kam an vom Wechselrichter.

tabacha, 2023-03-19:
Okay, jetzt taucht die Meldung nicht mehr auf

closed 2023-03-19 by tbnobody