tbnobody/OpenDTU #714: [Request] Seriennummern der Hoymiles DTUs in OpenDTU verwenden

FrodoVDR, 2023-03-18:
### Is your feature request related to a problem? Please describe.

Die Seriennummern meiner DTUs von Hoymiles fangen mit 10FD7... oder 10D9... an, 
um z.B. im aktielle Testfall der HMS/HMT-Inverter zwischen den DTUs wechseln zu können wollte ich nun die Original Serien Nummer meiner DTU-Pro-S eintragen, leider ist da nicht möglich weil das Eingabe Feld nur Zahlen akzeptiert.

### Describe the solution you'd like

Das Feld für die Wechselrichter Seriennummer als String verwalten.

homeautomation2022, 2023-03-19:
Das Feld sollte eher so abgeändert werden, das man da nur die letzten 8 Ziffern der DTU Serial einträgt, denn die 4 Zeichen(Ziffern/Buchstaben) davor sind nur der DTU Typ und werden nirgends verwendet in OpenDTU.

Das andere Problem ist, bei HMS/HMT mit original DTU und OpenDTU gleichzeitig zu arbeiten habe ich bereits im Discord erklärt, das wird schwierig und zeitnah sicher nicht möglich/umgesetzt werden.

Offizielle Empfehlung für OpenDTU ist immer nur eine DTU gleichzeitig mit einem Wechselrichter zu verwenden.

ccoenen, 2023-09-18:
Ich habe hier eine Seriennummer in der Form

**R**12**E**1234567**A**1**C**12

Das ist ein TSUN TSOL-MS600 - nein, der ist nicht in der Kompatibilitätsliste explizit drin, aber angeblich sei der quasi der TSUN TSOL-M800, deswegen wollte ich mal probieren, ob das geht.

Hat jemand eine Idee, ob/wie/welchen Teil man da eintragen muss?

stefan123t, 2023-09-22:
@homeautomation2022 man könnte die Seriennummer der DTU wie Du und @FrodoVDR vorgeschlagen habt in zwei Teilen eingeben:
DTU Serial: [`<DTU model prefix>`] v [`<Serialnumber>`], also ein Pull Down + Input field.

Intern kann man die Seriennummer weiterhin als Integer speichern nur die Eingabe sollte wie vorgeschlagen und für die Inverter Seriennummern auch üblich im Hoymiles Format erfolgen.

In der S-Miles Installer App sind folgende Prefixe zusammengefaßt:

| prefix | type | models |
| --- | --- | --- |
| `10F7` | `CW_WIRED_2G` | DTU-Pro |
| `10FB`, `4101`, `10FC`, `4120` | `CW_WIRED_4G` | DTU-Pro |
| `10F8`, `4100`, `10FD`, `4121` | `CW_WIRED_WIFI` | DTU-Pro |
| `10D3`, `4110`, `10D8`, `4130`, `4132`, `4133`, `10D9`, `4111` | `CW_WIFI` | DTU-W100, DTU-Lite-S |
| `10D2` | `CW_2G` | DTU-G100 |
| `10D6`, `10D7`, `4131` | `CW_4G` | DTU-Lite |

Hier die anderweitig bereits bekannten Modellprefixe aus https://github.com/lumapu/ahoy/blob/main/doc/hoymiles-format-description.md

| model | prefix |  
| -- | -- |
| DTU-G100 | 10D2 |  
| DTU-W100 | 10D3 |  
| DTU-Lite-S | 10D3 |  
| DTU-Lite | 10D6 |  
| DTU-Pro | 10F7 |  
| DTU-Pro | 10F8 |  
| DTU-Pro | 10FA |  
| DTU-Pro | 10FB |

tbnobody, 2023-09-22:
Warum sollte man diesen Aufwand machen? Die ersten 4 Stellen werden sowieso nicht verwendet.

stefan123t, 2023-09-24:
@tbnobody es hilft Probleme bzw. die Kommunikationsart mit dem WR (NRF24 bzw. CMT2300A) bei den neueren HMS/HMT & bei der DTU+WR Kombination der ganz neuen HMS-xxxW WiFi Modellen zu unterscheiden :wink: 

Auch wenn wir mit den WiFi Modellen (noch) nichts anfangen können:
Man könnte evtl. in Zukunft auch einen WiFi AP/STA aufmachen und mit dem DTUBI (Built In / Inner DTU) per WLAN kommunizieren.

stefan123t, 2023-09-24:
> Das ist ein TSUN TSOL-M**S**600 
> Hat jemand eine Idee, ob/wie/welchen Teil man da eintragen muss?

Laut Datasheet ist auch das ein HMS/HMT mit WiFi Modell. 
> Monitor: Integrated WiFi (Optional)

Die HMS/HMT-WiFi Modelle werden Stand heute noch nicht unterstützt, da hier ein komplett anderes Protokoll (S-Miles Cloud o.a.) zum Einsatz kommt.

ccoenen, 2023-09-25:
Vielen Dank für die Rückmeldung :-)

stefan123t, 2023-09-28:
Siehe auch hier https://github.com/lumapu/ahoy/issues/1189

stefan123t, 2024-07-10:
> Ich habe hier eine Seriennummer in der Form
>
> **R**12**E**1234567**A**1**C**12
>
> Das ist ein TSUN TSOL-M**S**600 - nein, der ist nicht in der Kompatibilitätsliste explizit drin, aber angeblich sei der [...]

Hallo Claudius, @ccoenen kannst Du mal Deine Seriennummer in #519 posten, da haben wir noch einen Kandidaten mit einem TSOL-M1600. Dein TSOL-MS600 entspricht eventuell auch einem HMS-600, also mit CMT2300A Funkverbindung ?

stefan123t, 2024-07-10:
@FrodoVDR der Grund warum Du gerne die Hoymiles DTU Serial ID eintragen möchtest ist vermutlich weil Du die Verbindung zwischen WR und DTU gemeinsam nutzen möchtest. Das ist aber prinzipiell nicht unterstützt, wir haben z.B. keinen Read-Only mode, bei dem wir sowohl die Commands von der Hoymiles DTU als auch die Antworten von den WR empfangen und entsprechend verarbeiten.

Du könntest Dir selbst Deine DTU Seriennummer analog zu dem von @Birkenstab hier [beschriebenen Verfahren](https://github.com/tbnobody/OpenDTU/issues/1751#issue-2137109935) in Deine Config eintragen.

@tbnobody wäre so ein Promiscuous Modus evtl. sinnvoll und wünschenswert um ggf. auch unbekannte Commands und die Antworten der Inverter aufzuzeichnen ?

Die OpenDTU müsste im **Promiscuous Modus** eben alle Commands und Antworten sammeln und wenn ein vollständiges Command + alle Antwortpakete vollständig vorliegen die Informationen verarbeiten oder zumindest in der USB / Serial Console mit loggen.

stefan123t, 2024-07-18:
> Der Vollständigkeit halber hier auch die laut App zusammengehördenden Prefixe der neuen HMS-800W-2T und anderer HMS-WiFi Modelle:
> prefix 	type 	models
> `1124`, `1125`, `1403` 	`1-in-1` 	HMS/HMT
> `1144`, `1143`, `1145`, **`1412`** 	`2-in-1` 	HMS/HMT
> `1164`, `1165`, `1166`, `1167`, `1222`, `1422`, `1423` 	`4-in-1` 	HMS/HMT
> `1361`, `1362` 	`4-in-1` 	HMS/HMT
> `1381`, `1382` 	`6-in-1` 	HMS/HMT

@tbnobody hier sind evtl. schon die DTUBI / HMS-800**W** Modelle enthalten.
Zumindest die `1412` von @PerfectAnswer ist bereits dabei gewesen.