tbnobody/OpenDTU #718: OLIMEX ESP32 POE   Ethernet is not working

PhiDiWoe, 2023-03-19:
### What happened?

The Network (Ethernet) Option of the OLIMEX ESP32 Module doesn´t work.

### To Reproduce Bug

Every kind of firmware produces the same, its getting no IP-Adress by Ethernet. Retry every 10 seconds

### Expected Behavior

wouldt like to connect by Ethernet it is more stable than Wifi, and i could power the module by POE.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

97bc964

### Relevant log/trace output

```Shell
09:41:21.049 > Network connected
09:41:28.838 > ETH connected
09:41:28.844 > Switch to Ethernet mode
09:41:28.849 > WiFi disconnected
09:41:28.882 > Configuring Ethernet DHCP IP... done
09:41:28.889 > Setting Hostname... done
09:41:30.838 > ETH disconnected
09:41:30.838 > Network lost connection
09:41:30.841 > Switch to WiFi mode
09:41:30.873 > E (362953) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffb26e0 (config or if_key is NULL or duplicate key)
09:41:30.884 > E (362956) esp.emac: esp_eth_mac_new_esp32(595): alloc emac interrupt failed
09:41:30.890 > [360323][E][ETH.cpp:267] begin(): esp_eth_mac_new_esp32 failed
09:41:30.895 > Setting Hostname... done
09:41:30.912 > E (362994) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffb2670 (config or if_key is NULL or duplicate key)
09:41:30.924 > E (362996) esp.emac: esp_eth_mac_new_esp32(595): alloc emac interrupt failed
09:41:30.929 > [360363][E][ETH.cpp:267] begin(): esp_eth_mac_new_esp32 failed
09:41:30.936 > Configuring WiFi STA using new credentials... done
09:41:30.942 > Configuring WiFi STA DHCP IP... done
09:41:31.017 > WiFi connected
09:41:31.050 > WiFi got ip: 192.168.178.XXXXX
09:41:31.050 > Network connected
```

PhiDiWoe, 2023-03-19:
Also tested with git-hash bf05dc8

PhiDiWoe, 2023-03-20:
I´ve got a OLIMEX ESP32-POE-ISO Version of this board, maybe its a problem with powering by USB without using POE. The Injector will be delivered tomorrow.

i´ve got not very much experience by using Visual Code. First time i changed all in the platformio.ini, later in the override.ini

PhiDiWoe, 2023-03-20:
Pah, da können wir ja deutsch schreiben...

[image]

so hab ich den ganzen Spaß compiliert und hochgeladen. Also "env:olimex_esp32_poe". Ich hab festgestellt, das an meinen ZYXEL Switches garnix geht, an der Firtzbox kommt alle 10 Sek der connect Versuch. 

Ich hab die Version mit Isoliertem POE Interface um hier keine Potentialprobleme zu bekommen, Denke ich muss hier noch abwarten bis morgen der POE Injektor kommt.

PhiDiWoe, 2023-03-21:
So bei Stromversorgung per POE bekomme ich an zwei Olimex ESP32 POE ISO das gleiche Ergebnis.

18:59:31.802 > ETH connected
18:59:31.802 > Switch to Ethernet mode
18:59:31.811 > WiFi disconnected
18:59:31.873 > Configuring Ethernet DHCP IP... done
18:59:31.879 > Setting Hostname... done
18:59:33.803 > ETH disconnected
18:59:33.803 > Network lost connection
18:59:33.806 > Switch to WiFi mode
18:59:33.835 > E (1251308) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffb26e0 (config or if_key is NULL or duplicate key)
18:59:33.846 > E (1251310) esp.emac: esp_eth_mac_new_esp32(595): alloc emac interrupt failed
18:59:33.851 > [1246393][E][ETH.cpp:267] begin(): esp_eth_mac_new_esp32 failed
18:59:33.858 > Setting Hostname... done
18:59:33.876 > E (1251349) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffb2670 (config or if_key is NULL or duplicate key)
18:59:33.886 > E (1251350) esp.emac: esp_eth_mac_new_esp32(595): alloc emac interrupt failed
18:59:33.892 > [1246433][E][ETH.cpp:267] begin(): esp_eth_mac_new_esp32 failed
18:59:33.898 > Configuring WiFi STA using new credentials... done
18:59:33.904 > Configuring WiFi STA DHCP IP... done
18:59:33.982 > WiFi connected
18:59:34.017 > WiFi got ip: 192.168.178.78
18:59:34.019 > Network connected
18:59:41.805 > ETH connected

tbnobody, 2023-03-21:
Hab es gerade selbst getestet... Generic Profil geflashed, xxxx als pin_mapping.json hochgeladen und unter Settings --> Hardware das entsprechende Profil ausgewählt:
[image]

Der Startprozess sieht dann so aus:
```
20:33:36.792 > Starting OpenDTU
20:33:36.792 > Initialize FS... done
20:33:36.799 > Reading configuration... done
20:33:36.915 > Reading PinMapping... found valid mapping done
20:33:36.947 > Initialize Network... ETH start
20:33:41.203 > done
20:33:41.203 > Setting Hostname... Configuring WiFi STA using new credentials... done
20:33:41.217 > Initialize NTP... done
20:33:41.217 > Initialize SunPosition... done
20:33:41.219 > Initialize MqTT... done
20:33:41.223 > Initialize WebApi... done
20:33:41.236 > Initialize Display... done
20:33:41.239 > Check for default DTU serial... done
20:33:41.242 > Initialize Hoymiles interface... E (9368) wifi:Set status to INIT
20:33:41.253 > Connection successfull
20:33:41.253 > WiFi disconnected
20:33:41.256 >   Setting radio PA level...
20:33:41.258 >   Setting DTU serial...
20:33:41.261 >   Setting poll interval...
20:33:41.264 > done
20:33:41.264 > Switch to WiFi mode
20:33:41.264 > E (9388) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffb26e0 (config or if_key is NULL or duplicate key)
20:33:41.278 > E (9395) esp.emac: esp_eth_mac_new_esp32(595): alloc emac interrupt failed
20:33:41.283 > [  4519][E][ETH.cpp:267] begin(): esp_eth_mac_new_esp32 failed
20:33:41.289 > Setting Hostname... done
20:33:41.311 > E (9437) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffb2670 (config or if_key is NULL or duplicate key)
20:33:41.322 > E (9439) esp.emac: esp_eth_mac_new_esp32(595): alloc emac interrupt failed
20:33:41.328 > [  4563][E][ETH.cpp:267] begin(): esp_eth_mac_new_esp32 failed
20:33:41.333 > Configuring WiFi STA using existing credentials... E (9472) wifi:sta is connecting, return error
20:33:41.349 > [  4591][E][WiFiSTA.cpp:317] begin(): connect failed! 0x3007
20:33:41.354 > done
20:33:41.354 > Configuring WiFi STA DHCP IP... done
20:33:41.376 > E (9503) wifi:Set status to INIT
20:33:41.383 > WiFi disconnected
20:33:41.383 > Try reconnecting
20:33:41.387 > Network lost connection
20:33:41.409 > E (9535) wifi:Set status to INIT
20:33:41.416 > WiFi disconnected
20:33:41.416 > Try reconnecting
20:33:41.420 > Network lost connection
20:33:41.487 > WiFi connected
20:33:42.603 > WiFi got ip: 192.168.20.139
20:33:42.603 > Network connected
20:34:17.154 > ETH connected
20:34:17.154 > Switch to Ethernet mode
20:34:17.162 > WiFi disconnected
20:34:17.183 > ETH got IP: 192.168.1.119
20:34:17.183 > Network connected
20:34:17.285 > Configuring Ethernet DHCP IP... done
20:34:17.294 > Setting Hostname... done
```

--> Klappt Problemlos.

Wie schauen jetzt deine konfigurierten Pins aus?
Hier aktuell so:
[image]

PhiDiWoe, 2023-03-21:
OK, jetzt steh ich am Bahnhof. :-(

Generic bekomme ich noch hin.

pin_mapping.json ? wo finde ich die?
Das Profil kann ich aktuell nicht anwählen und das Pinmapping passt somit auch nicht.

tbnobody, 2023-03-21:
Die generelle Funktionsweise ist hier beschrieben: https://github.com/tbnobody/OpenDTU/blob/master/docs/DeviceProfiles.md

Vordefinierte Profile befinden sich hier: https://github.com/tbnobody/OpenDTU/tree/master/docs/DeviceProfiles
In deinem Fall ist das: [olimex_esp32_poe.json](https://github.com/tbnobody/OpenDTU/blob/master/docs/DeviceProfiles/olimex_esp32_poe.json)

Hochladen kannst du es unter Settings --> Konfigurationsverwaltung:
[image]

PhiDiWoe, 2023-03-21:
So nach deiner Anleitung alles gemacht alles gestartet.

Modul1 immernoch der gleiche Fehler, alle 10 sekunden ist der Ethernet weg.

Das gleiche mit dem zweiten Olimex Modul gemacht und schon läuft alles, noch Ohne NRF

Danke für den Support

PhiDiWoe, 2023-03-21:
Hab am 2ten Modul noch keinen NRF angeschlossen und an Modul 1 ist es auch ohne den NRF das gleiche.

closed 2023-03-22 by PhiDiWoe