tbnobody/OpenDTU #719: Compile Error: "[.pio/build/generic/src/WebApi_prometheus.cpp.o] Error 1"

dl7ata, 2023-03-19:
### What happened?

Using Platform.IO / Visual Studio Code V.1.76.2 and try to compile bf05dc8. 
Failed

### To Reproduce Bug

Trying to compile any time.

### Expected Behavior

Should run without any error on 

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

bf05dc8

### Relevant log/trace output

```Shell
src/WebApi_prometheus.cpp: In member function 'void WebApiPrometheusClass::onPrometheusMetricsGet(AsyncWebServerRequest*)':
src/WebApi_prometheus.cpp:98:30: error: exception handling disabled, use -fexceptions to enable
     } catch (std::bad_alloc& bad_alloc) {
                              ^~~~~~~~~
src/WebApi_prometheus.cpp:99:115: error: 'bad_alloc' was not declared in this scope
         MessageOutput.printf("Call to /api/prometheus/metrics temporarely out of resources. Reason: \"%s\".\r\n", bad_alloc.what());
                                                                                                                   ^~~~~~~~~
src/WebApi_prometheus.cpp:99:115: note: suggested alternative:
In file included from /home/fko/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_construct.h:59,
                 from /home/fko/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_tempbuf.h:60,
                 from /home/fko/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_algo.h:62,
                 from /home/fko/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/algorithm:62,
                 from /home/fko/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:162,
                 from .pio/libdeps/generic/ESP Async WebServer/src/ESPAsyncWebServer.h:24,
                 from include/WebApi_prometheus.h:4,
                 from src/WebApi_prometheus.cpp:6:
/home/fko/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/new:54:9: note:   'std::bad_alloc'
   class bad_alloc : public exception
         ^~~~~~~~~
*** [.pio/build/generic/src/WebApi_prometheus.cpp.o] Error 1
```

tbnobody, 2023-03-19:
Should be related to your installation as it's compiling without any problems here and also the action from 2 days ago compiled without problems: https://github.com/tbnobody/OpenDTU/actions/runs/4450074421/jobs/7815085096

dl7ata, 2023-03-19:
I switched back to 3421c9a and it went without a failure.

tbnobody, 2023-03-19:
Yes thats clear. Because this specific code does not exist in this version. But I don't understand why it's not compiling at your setup

dl7ata, 2023-03-19:
What else of information regarding the compiler do you need?

tbnobody, 2023-03-19:
I don't think its related to a compiler issue as I use the same version (toolchain-xtensa-esp32@8.4.0+2021r2-patch5)
But maybe a different framework version. Have you tried a clean and rebuild?

Can you post the version of the different packages?
Should look like this:

```
Firmware Revision: g5e37019
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.0.1) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6) 
 - tool-esptoolpy @ 1.40400.0 (4.4.0) 
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
```

dl7ata, 2023-03-19:
Yes - I try a clean new build in a new directory.

<Wo finde die Angaben zu den Versionen?>

tbnobody, 2023-03-19:
Just scroll up in the window with the error message. should be one of the first output in the build window.

dl7ata, 2023-03-19:
Complete output from update with failure:
#--------------------------------------------------------------------------------------------------------------------------------------------#

_Processing generic (board: esp32dev; framework: arduino; platform: espressif32@>=6.0.1)
----------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Firmware Revision: gbf05dc8
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2023.1.1+sha.02c67db) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 2.0.6 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 43 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP Async WebServer @ 1.2.3+sha.1dde9cf
|   |-- AsyncTCP @ 1.1.1
|   |-- ArduinoJson @ 6.21.0
|   |-- FS @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- ArduinoJson @ 6.21.0
|-- espMqttClient @ 1.4.1+sha.15041c4
|   |-- AsyncTCP @ 1.1.1
|   |-- WiFiClientSecure @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- RF24 @ 1.4.5
|   |-- SPI @ 2.0.0
|-- U8g2 @ 2.34.15
|   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|-- sunset @ 1.1.7
|-- LittleFS @ 2.0.0
|   |-- FS @ 2.0.0
|-- Hoymiles @ 0.0.1
|   |-- RF24 @ 1.4.5
|   |   |-- SPI @ 2.0.0
|   |-- Every
|   |-- SPI @ 2.0.0
|   |-- TimeoutHelper
|-- Ethernet @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- Update @ 2.0.0
|-- ResetReason
|-- DNSServer @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- WiFi @ 2.0.0
|-- MqttSubscribeParser
|   |-- espMqttClient @ 1.4.1+sha.15041c4
|   |   |-- AsyncTCP @ 1.1.1
|   |   |-- WiFiClientSecure @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- Ticker @ 2.0.0
Building in release mode
Compiling .pio/build/generic/src/WebApi_prometheus.cpp.o
Compiling .pio/build/generic/src/WebApi_webapp.cpp.o
Compiling .pio/build/generic/src/WebApi_ws_console.cpp.o
Compiling .pio/build/generic/src/WebApi_ws_live.cpp.o
src/WebApi_prometheus.cpp: In member function 'void WebApiPrometheusClass::onPrometheusMetricsGet(AsyncWebServerRequest*)':
src/WebApi_prometheus.cpp:98:30: error: exception handling disabled, use -fexceptions to enable
     } catch (std::bad_alloc& bad_alloc) {
[... 102 lines cut ...]
|-- Ethernet @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- Update @ 2.0.0
|-- ResetReason
|-- MqttSubscribeParser
|   |-- espMqttClient @ 1.3.3+sha.6a3f3a8
|   |   |-- AsyncTCP @ 1.1.1
|   |   |-- WiFiClientSecure @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- Ticker @ 2.0.0
|-- DNSServer @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- WiFi @ 2.0.0
Building in release mode
Retrieving maximum program size .pio/build/generic/firmware.elf
Checking size .pio/build/generic/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  13.6% (used 44572 bytes from 327680 bytes)
Flash: [=======   ]  66.0% (used 1038137 bytes from 1572864 bytes)
============================================================ [SUCCESS] Took 6.97 seconds ============================================================

Environment    Status    Duration
-------------  --------  ------------
generic        SUCCESS   00:00:06.971_

lribic, 2023-03-20:
I had the same issue. The build worked again after cleaning the git repo (removing .pio/?) and removing ~/.platformio/.

closed 2023-03-20 by dl7ata