tbnobody/OpenDTU #727: Prometheus - Werte sind manchmal 0

Thoro, 2023-03-21:
### What happened?

Falls openDTU ein unvollständiges Paket erhält kann es passieren das alle Prometheus Werte 0 since.

Das führt zu unschönen Zacken und Problemen in den Daten, opendtu_last_update wird fleißig hochgezählt obwohl die Daten nicht aktualisiert wurden.

[image]
[image]

### To Reproduce Bug

System laufen lassen mit und die /api/prometheus/metrics page öfters refreshen, irgendwann sind alle Werte 0, falls RX Fragments fehlen. (>20m Distanz zum WR)

### Expected Behavior

Prometheus Werte sollten entweder nicht aktualisiert werden, oder gar nicht ausgegeben werden.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

bf05dc8

tbnobody, 2023-03-21:
Wie hast du geprüft ob in diesem Moment wirklich RX Fragments fehlen? Diese werden nämlich nur in den Parser übertragen wenn alle vorhanden sind. Was sagt denn in diesem Moment die Console? Ich würde eher vermuten, dass nicht genug zusammenhängender Speicher vorhanden ist um die Ausgabe zu rendern. Das würde aber in der Console angezeigt.

Thoro, 2023-03-21:
hab mir getStatisticsLength ausgegeben, wenns gepasst hat wars 64 und wenn nicht 4 - nachdem das nur durch das addFragment erhöht wird, würde ich schließen das nicht alle da waren.

Inverter ist ein HM-1500

tbnobody, 2023-03-21:
Verwendest du eine Original DTU parallel? oder einen ESP mit NRF?

Thoro, 2023-03-21:
Sorry, getStatisticsLength hab ich reinprogrammier - und wieder gelöscht - gibt nur die private _statistics_length vom Parser aus!

ja, ESP8266 mit NRF UND OriginalDTU - dann wird es hier zu Überschneidungen kommen, sollte aber trotzdem keinen Fehlerzustand auslösen

tbnobody, 2023-03-21:
> dann wird es hier zu Überschneidungen kommen, sollte aber trotzdem keinen Fehlerzustand auslösen

Doch! Das ist genau das Thema. Der Fehler findet aber nicht auf der DTU statt sondern im Inverter. Dieser liefert Fragmente mit gültigen CRC Prüfziffern aber ungültigen Payloads. Was sonst intern schiefgeht weiß kein Mensch. Daher absolut unsupported!

Thoro, 2023-03-21:
Ok, interessant - gut - oder schlecht für den Inverter

Aber was spricht gegen einen sanity check?

tbnobody, 2023-03-21:
Es gibt sogar 3 Sanity Checks:
* Eine CRC Prüfsumme die das NRF Modul am empfangenen Paket prüft
* Eine CRC Prüfsumme die das einzelne Fragment nach der Übertragung über den SPI Bus prüft
* Eine CRC Prüfsumme die alle Payloads in Summe verifiziert

Etwas weiteres sehe ich hier erstmal nicht ein. Du schreibst oben "oder schlecht für den Inverter"... Vielleicht auch schlecht für den Besitzer? Wenn niemand weiß was in so einem Fall intern im Inverter passiert, wie kann man sicherstellen das nicht z.B. der NA Schutz in so einem Fall ebenfalls betroffen ist?

Daher, so leids mir tut, kein Support für solche Multi-DTU-Setups und kein weiterer Check.

closed 2023-03-21 by tbnobody