tbnobody/OpenDTU #737: Mqtt not working

thisisfl0, 2023-03-23:
### What happened?

Mqtt feature not working

### To Reproduce Bug

Just turn on the mqtt feature

### Expected Behavior

..

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

4301b84

### Relevant log/trace output

```Shell
21:38:28.333 > Connecting to
MOTT..
21:38:28.347 > Disconnected from
MQTT.
21:38:28.347 > Disconnect
reason: TCP DISCONNECTED
```

tbnobody, 2023-03-23:
Works perfectly here and on other instances. 
What broker are you using? DDo you use authentication? Do you use ACLs? Is your broker reachable with e.g. mqtt explorer?

thisisfl0, 2023-03-23:
I tried to connect a wemos d1 mini with tasmota using the mqtt function. As it did not work I checked the terminal and saw the error message. I don’t know what ACLs are but I also tried changing the mqtt settings (with and without password authentification.

tbnobody, 2023-03-23:
I dont know what you are meaning. You need a broker (like mosquitto) where tasmota and opendtu is connecting to. What are you using as a broker?

thisisfl0, 2023-03-23:
Sorry. Habe noch nie etwas über mqtt gemacht. Habe eben mal kurz gegoogelt. Da gibt es auch ein Plugin für homebridge (homebridge-mqtt) ich versuche nur die Reichweite des Opendtu zu erweitern indem ich die Daten auf einen wemos d1 mini mit display weiter übertragen möchte um das Display in der Wohnung und nicht auf dem Balkon platzieren zu können. Ich dachte die Verbindung wäre direkt möglich. Aber bin nun davon ausgegangen dass das Problem vom opendtu ausgeht da dieser meldet dass mqtt nicht funktioniert.

tbnobody, 2023-03-23:
Eine direkte Verbindung ist nicht möglich. MQTT benötigt immer einen Server auf den sich alle Geräte verbinden.

thisisfl0, 2023-03-23:
Gut zu wissen! Ich werde das mal ausprobieren. Gibt es eventuell noch eine andere einfachere Möglichkeit die relevanten Daten an einen weiteren esp weiterzugeben der diese dann auf einem Display anzeigt?

tbnobody, 2023-03-23:
Meines wissens nichts was aktuell schon implementiert ist.

closed 2023-03-23 by tbnobody

thisisfl0, 2023-03-24:
Moin,

habe "Aedes embedded MQTT broker" für Homebridge installiert und auch hier wieder mit TLS und ohne sowie mit und ohne authentifizierung probiert eine Verbindung aufzubauen. Klappt beides nicht. Bekomme nun folgenden Output aus der OpenDTU Konsole:

09:20:37.305 > Connect09:20:37.320 > ing t09:20:37.334 > o MQ09:20:37.347 > TT.09:20:37.396 > ..
09:20:37.407 > 09:20:37.418 > Disconnected from MQTT.
09:20:37.419 > Disconnect reason:TCP_DISCONNECTED

lässt sich TCP irgendwie aktivieren?

tbnobody, 2023-03-24:
Versuche doch einfach mal ob du dich mit einem Tool wie Mqtt explorer auf deinen Broker verbinden kannst. Wenn das nicht klappt geht es mit dem ESP auch nicht.

thisisfl0, 2023-03-24:
Das funktioniert wohl auch nicht.. in der Konsole vom DTU steht ja schon dass MQTT nicht läuft. Hast du eine Idee woran das liegen könnte?

tbnobody, 2023-03-24:
Nein. Ich kenne kein "Aedes embedded MQTT broker". Keine Ahnung was das machen sollte o.ä.

thisisfl0, 2023-03-24:
Um den geht es auch nicht... Die Konsole meldet dass MQTT nicht funktioniert.. habe es trotzdem im MQTT Explorer versucht ohne Erfolg. Die bereits eingebundene Testverbindung funktioniert. Der Fehler liegt beim DTU.

Argafal, 2023-03-24:
Wenn du dich mit MQTT Explorer nicht mit deinem MQTT-Server verbinden kannst, dann liegt das Problem vermutlich an deinem MQTT-Server oder dessen Einstellungen. An deiner Stelle würde ich versuchen, das Problem einzugrenzen: Schau dir die Server logs an. Gibt es Fehlermeldungen? Geht eine manuelle Verbindung mit mosquitto_sub direkt auf dem Server, also zu localhost? Geht eine manuelle Verbindung von einem anderen Rechner? Wenn nicht, warum nicht? 

Hier haben relativ viele die OpenDTU über MQTT erfolgreich mit ihren MQTT-Servern am laufen... Ich vermute also dass es schlau sein könnte, wenn du dir deine MQTT-Server-Einstellungen näher ansiehst und unabhängig von OpenDTU testest. Wenn das dann geht, zurück zur OpenDTU. Viel Erfolg!

tbnobody, 2023-03-24:
Es kann nicht an der DTU liegen! Du musst dich mit MQTT Explorer auf den Broker verbinden, nicht auf die DTU! Nur so kannst du prüfen ob der Broker funktioniert!

thisisfl0, 2023-03-24:
Na gut, das hat jetzt funktioniert. Die Daten kommen jetzt auch im Explorer an und eine Spalte für Tasmota taucht auch auf. Trotzdem fraglich ob ich das hinbekomme :D Danke, falls es nochmal Probleme gibt melde ich mich.

Argafal, 2023-03-24:
Erzähl doch mal, für den Nächsten der das Problem auch hat: Was war die Problemursache, und was war die Lösung?

Argafal, 2023-03-24:
> ich habe wohl versehentlich die IP des DTU verwendet und auch im Broker eingetragen bis ich verstanden hab dass dort die IP des Homebridge Instanz rein gehört..

Dann war OpenDTU also doch unschuldig. Juhuu :)

thisisfl0, 2023-03-24:
Ist das überhaupt möglich dass Tasmota sich per MQTT die Daten vom OpenDTU holen kann um diese dann anzuzeigen? oder geht das "nur in eine Richtung"?