tbnobody/OpenDTU #743: Inverters are not consistently updated

cody82, 2023-03-25:
### What happened?

I have set the DTU update interval to 5s, but the inverters seem to be updated only every 10s or even every 20s. The interval is inconsistent. Also sometimes I see that the power display in the web interface changes, but the "last update" does not change to 0s.

### To Reproduce Bug

Set the update interval to 5s(with two inverters). Most of the times it will only fetch one inveter every 10s. Also I am setting the limit_nonpersistent_absolute every >10s.

### Expected Behavior

Every inverter should be updated every 5s.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

b42fff8

### Relevant log/trace output

```Shell
https://gist.github.com/cody82/ac45de46e13eba913bd5518e3be231a5
```

### Anything else?

The log output is sometimes garbled:
12:33:19.282 > Received MQTT message on topic: opendtu/116183073567/cmd/limit_nonpersistent_ab**T**solute
12:34:09.490 > Received MQTT message on topic: opendtu/116183073567/cmd/limit_nonpersist**12:34:09.554 > ent_absolute**

tbnobody, 2023-03-25:
The log looks good but you have to keep in mind that sending control commands has highest priority. If the software is busy with sending limit commands (what's definetly the case if you send udpates every 10 seconds) no statistic requests are placed into the command queue.
Seems to be the same behavior like here: https://github.com/tbnobody/OpenDTU/issues/676

cody82, 2023-03-25:
So to improve the situation, what would be possible?
1. In the log it looks like I do have a lot of retransmits. Should I buy a proper antenna?
2. The retransmit is done 2s after the command transmission, which is kind of late because on success the response is received after ~0.2s. So for me it looks like that a timeout of ~0.5s would improve this?
3. Could the statistic request be queued if a control command is still in progress?

tbnobody, 2023-03-25:
> In the log it looks like I do have a lot of retransmits. Should I buy a proper antenna? (One inverter is 2m from the DTU but with a wall in between. The other is 10m away but with only a glass between)

Do you use a capacitor to stabilize the current of the NRF module?

> The retransmit is done 2s after the command transmission, which is kind of late because on success the response is received after ~0.2s. So for me it looks like that a timeout of ~0.5s would improve this?

You mean for the ActivePowerControl? Thats true. Timeout is 2sec. Should be the same value as on the original DTU. If the command is confirmed earlier the timeout will never be reached and the time till the next command is much less. 

> Could the statistic request be queued if a control command is still in progress?

No because this could leed to a queue overflow and to some out of memory exceptions.

> The frequent power limit commands are for limiting the export to the grid.

That may be the case, on the other hand, a original DTU sends the statistics data only every 15min.

cody82, 2023-03-25:
I attached a capacitor today before recording the logs. It did not improve anything.

When looking at the logs, it looks like that even without a timeout/retransmission, setting the power limit takes 2s:
```
12:34:49.329 > Received MQTT message on topic: opendtu/114181013675/cmd/limit_nonpersistent_absolute
12:34:49.329 > Limit Non-Persistent: 216 W
12:34:49.390 > TX ActivePowerControl Channel: 3 --> 51 81 01 36 75 80 15 72 12 81 0B 00 08 70 00 00 DB 02 4C 
12:34:49.390 > Received MQTT message on topic: opendtu/116183073567/cmd/limit_nonpersistent_absolute
12:34:49.390 > Limit Non-Persistent: 339 W
12:34:49.460 > Interrupt received
12:34:49.531 > RX Channel: 75 --> D1 81 01 36 75 81 01 36 75 81 00 00 0B 00 14 07 48 
12:34:51.375 > RX Period End
12:34:51.375 > Success
```
Maybe I interpret this wrong. The response is received <0.2s after the TX, but the "RX Perdiod End" is logged 2s after the TX. Why does the RX Period not end immediately after the reception of the response?

Larger intervals will probably improve this, but that will decrease power control accuracy. As the communcation channel is not nearly saturated, I think there is room for improvement by lowering timeouts or better scheduling of commands. There may be limitations on the inverter side though. It is strange that the responses for the power limits seem to be missing more often than for the other requests.

DPO99, 2023-03-26:
Basically it is a display problem. The commands are also processed, even if the display is not permanently up-to-date. I don't know the size of the system but I believe that an interval of 15-20 seconds is acceptable and the "losses" are not really relevant. You will never manage to generate a 100% zero feed. In addition, it is unclear, at least to me, whether the Hoymiles can even react at this speed.

cody82, 2023-03-26:
Well, there is no special value for "acceptable", but faster updates are generally better. 100% zero feed is not possible even with most commercial zero feed setups, but they update in <1s. I switched to 15s now and the tracking is less precise. 20s will probably get rid of the problems.
It just should not look like as if there is an inverter that has more than 600W. To me it looks like the hoymiles reacts quite fast(few seconds max) for limiting.

DPO99, 2023-03-27:
The question that arises is whether you can do without this delayed display of the OpenDTU or whether you can live with it. For me personally, sending MQTT settings is more important. Under certain circumstances, this behavior must simply be understood as "works as designed".

cody82, 2023-03-30:
I still dont think it is good that updates are skipped instead of delayed until the command is ready. But yes, I can live with it.