tbnobody/OpenDTU #744: OpenDTU stops sending MQTT Messages

fellpower, 2023-03-25:
### What happened?

Every Day, at different Times, OpenDTU stops sending MQTT Messages. I can proof that, by checking my MQTT Server (Mosquitto) and my influxdb (1.8). There is no data at this point. All my other stuff, thats sending MQTT Messages - work.

WLAN is very strong - and i have that NRF Module with antenna. If im checking the OpenDTU Webinterface, i see that Data from the WR - and its changing, so the communication between OpenDTU and the WR is ok.

FW Version: 6ddb34e

Because, there is no log function, i dont know, whats happening.

The HM-800 is online (reachable) and producing to that times.

If im restarting OpenDTU - all works fine again ^^

### To Reproduce Bug

Wait for the next day ^^

### Expected Behavior

Sending MQTT Messages the whole time? ;)

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

6ddb34e

### Anything else?

Maybe u can build a version, that writes a log (Serial) - so i can connect to that console and check, whats happen. Or the "internal" console in the GUI ^^

tbnobody, 2023-03-25:
You are using an inofficial build with some additional display stuff. don't know what happens here.
What do you think is the meaning of "Info" --> "Console"? Doesn't it look like a serial log? But you are right it will not output any critical errors in the tcp stack or other internal components. for this you have to directly look at the serial console.

scho0ck, 2023-03-25:
I've experienced the same issue a couple of times now I'm on "Firmware Version / Git Hash | d508b41" built via VScode the webinterface was still accessible so that I can just press reboot. I'm now trying a nightly reboot via the api in the hope to solve this.

fellpower, 2023-03-25:
Some ppl have issues, with the MQTT Server from iobroker, after restarting it. So u need to restart OpenDTU too, to get it working again. 

But im using Mosquitto - and all my other MQTT things work flawlessly - so it must be a problem with OpenDTU.

Im now connected to the hardware serial port and write that log down, tomorrow we will see, whats the problem ^^

tbnobody, 2023-03-27:
Bert just fixed a reconnect issue in the MQTT library. Will integrate this fix within the next few minutes
see https://github.com/bertmelis/espMqttClient/issues/85

tbnobody, 2023-03-27:
Is implemented in 9523b99b228f24c8d9543f64be01d92758a418c5

scho0ck, 2023-03-27:
Just updated I will monitor the next few days and see if it happens again.

scho0ck, 2023-04-05:
I've not had any issues ever since so I'd consider this solved

closed 2023-04-12 by tbnobody