tbnobody/OpenDTU #753: Negative or incorrect pos. values (Energy) when offset is used (during startup of the inverters

Dom1n1c, 2023-03-27:
### What happened?

I Specified a "Yield total offset" (Ertragsversatz) for my used HM400
[image]

When the device is offline, i will get a negative value
[image]

### To Reproduce Bug

Add an offset and powercycle OpenDTU

### Expected Behavior

From my understanding the offset should only be used when the Inverter is online. (Or has produced energy !=0)

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

c47b7dd

DPO99, 2023-03-27:
Kann es sein, dass dies immer dann passiert, wenn der Wechselrichter nach dem Neustart noch keine Daten übermittelt hat und dies auch erstmal nicht tut, da keine sonne und der Offset sozusagen auf den Gesamtertragswert von 0 angewendet wird? Was passiert denn, wenn der WR Online war/ist, dann das Offset korrekt greift und dann der WR aus geht, weil keine Leistung mehr kommt?

Dom1n1c, 2023-03-28:
Ganz genau, das ist korrekt. Das passiert immer nur dann, wenn der WR noch keine Daten liefert.

commit d099bc4f (2023-03-31): Fix #753: Only apply offset if data is in the buffer to prevent negative numbers
closed 2023-03-31 by tbnobody