tbnobody/OpenDTU #758: Limit settings didn´t work correct on HM-1200

stefan-ohg, 2023-03-28:
### What happened?

After setting the temporary limit to 400 watts, the HM-1200 only produces 242.9 watts.
If the HM-1200 is immediately set back to 1200 watts, it produces the maximum possible power.
I have tested it with different watt settings, it is always produced less than is set, although the current situation allows more.

The same happens if you enter temporary values ​​in %.

Restarting the inverter and the OpenDTU brought no improvement

[image]

[image]

### To Reproduce Bug

Tried with different watt and % settings. I always used smaller values than actual possible in production.

### Expected Behavior

For example set 400 watts and the inverter produces 400 watts.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

c8a70da

tbnobody, 2023-03-28:
If you set the Limit to 400W, the inverter internally sets 200W to Ports 1&2 and 200W to port 3&4. This is fixed implemented in the inverters firmware and cannot changed in any way. Could it be that this is your problem? Do you maybe have just 2 or 3 panels and not even distributed via all ports?

stefan-ohg, 2023-03-28:
> If you set the Limit to 400W, the inverter internally sets 200W to Ports 1&2 and 200W to port 3&4. This is fixed implemented in t

That´s it. I have 4 panels, 2 on west an 2 on east.
Now I have to reconsider the calculation of my zero feed-in.

Thank you for the quick help!!!

closed 2023-03-30 by tbnobody