tbnobody/OpenDTU #759: [Request] Add additional MQTT topics

DejanBukovec, 2023-03-28:
### Describe the solution you'd like

Please add next MQTT topics:
- Firmware Version / Git Hash
- Firmware Update

That way we can get some notifications on Home Assistant that new update is availible...

DcFamas, 2023-03-30:
Hey there,
I also want to ask for a sum up of this values:

[image]
(Screenshot by AhoyDTU)

Best regards :)

tbnobody, 2023-03-30:
Please see https://github.com/tbnobody/OpenDTU/discussions/317#discussioncomment-4345649

Currently there are just other prios (like HMS/HMT)

stefan123t, 2024-07-10:
Here Eventlog Code and Count have been requested as MQTT Topics too: #424