tbnobody/OpenDTU #761: Neue Partitionstabelle ESP32 !?

mroenne2022, 2023-03-30:
### Is your feature request related to a problem? Please describe.

Durch "Zufall" hab ich heute gelesen das sich offensichtlich was in der Part.Table geändert hat und man den Upgrade nicht über OTA machen sollte. Nun hab ich kürzlich die letzte Version dennoch aus dem Menü heraus per Update installiert. Die DTU läuft seither problemlos wie immer. 

!! IMPORTANT UPGRADE NOTES !!
If you are upgrading from a version before 15.03.2023 you have to upgrade the partition table of the ESP32. Please follow the [this](https://github.com/tbnobody/OpenDTU/blob/master/docs/UpgradePartition.md) documentation!

### Describe the solution you'd like

Ist es dennoch erforderlich alles lieber erneut zu flashen wie in der Doku beschrieben ?

tbnobody, 2023-03-30:
Du kannst schon noch warten, aber irgendwann wirst du keine Updates mehr installieren können weil die neuen .bin Dateien zu groß werden.

closed 2023-03-30 by tbnobody

mroenne2022, 2023-03-30:
Ich erinnere mich das hier kürzlich darüber diskutiert wurde bei den Filesystemen was zu ändern, eines verkleinern und ein anderes vergrößern, das ist jetzt wohl erfolgt. Ich hab ja mehrere OpenDTUn , also werde ich die mal nach und nach neu flashen. 
Dank Dir.