tbnobody/OpenDTU #762: OpenDTU crash and reboot

DejanBukovec, 2023-03-30:
### What happened?

In log I see that today OpenDTU get exception and reboot...

### To Reproduce Bug

It happen first time

### Expected Behavior

Exception need to be handled to prevent reboot.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

c47b7dd

### Relevant log/trace output

```Shell
00:56:10.575 > Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
00:56:10.581 >
00:56:10.581 > Core  0 register dump:
00:56:10.585 > PC      : 0x401946cb  PS      : 0x00060b30  A0      : 0x801948bf  A1      : 0x3ffb2d40  
00:56:10.592 > A2      : 0x00000000  A3      : 0x3ffae98c  A4      : 0x00000000  A5      : 0x00000b38
00:56:10.600 > A6      : 0x00000000  A7      : 0x00000001  A8      : 0x80083d45  A9      : 0x3ffb2d40  
00:56:10.606 > A10     : 0x3ffaf1d0  A11     : 0x00001800  A12     : 0x3ffaf1d0  A13     : 0x00000007
00:56:10.614 > A14     : 0x00000005  A15     : 0x00000001  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
00:56:10.622 > EXCVADDR: 0x00000000  LBEG    : 0x40089ef0  LEND    : 0x40089f06  LCOUNT  : 0xffffffff
00:56:10.630 >
00:56:10.630 >
00:56:10.630 > Backtrace: 0x401946c8:0x3ffb2d40 0x401948bc:0x3ffb2d80 0x4019490d:0x3ffb2da0 0x4012dbf5:0x3ffb2dd0 0x40132a46:0x3ffb2e40 0x40136aba:0x3ffb2e70 0x401273f9:0x3ffb2e90
00:56:11.264 >
00:56:11.264 >   #0  0x401946c8:0x3ffb2d40 in AsyncClient::_lwip_fin(tcp_pcb*, signed char) at .pio/libdeps/generic/AsyncTCP/src/AsyncTCP.cpp:865
00:56:11.264 >   #1  0x401948bc:0x3ffb2d80 in AsyncClient::_s_lwip_fin(void*, tcp_pcb*, signed char) at .pio/libdeps/generic/AsyncTCP/src/AsyncTCP.cpp:1199
00:56:11.264 >   #2  0x4019490d:0x3ffb2da0 in _tcp_recv(void*, tcp_pcb*, pbuf*, signed char) at .pio/libdeps/generic/AsyncTCP/src/AsyncTCP.cpp:280
00:56:11.264 >   #3  0x4012dbf5:0x3ffb2dd0 in tcp_input at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/tcp_in.c:541 (discriminator 1)
00:56:11.264 >   #4  0x40132a46:0x3ffb2e40 in ip4_input at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/ipv4/ip4.c:803
00:56:11.264 >   #5  0x40136aba:0x3ffb2e70 in ethernet_input at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/netif/ethernet.c:186
00:56:11.264 >   #6  0x401273f9:0x3ffb2e90 in tcpip_thread_handle_msg at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:180
00:56:11.264 >       (inlined by) tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:154
00:56:11.264 >
00:56:11.264 >
00:56:11.264 >
00:56:11.264 >
00:56:11.264 > ELF file SHA256: 71e52406ee1f382b
00:56:11.267 >
00:56:11.267 > E (14639) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
00:56:11.267 > Rebooting...
00:56:11.267 > ets Jul 29 2019 12:21:46
00:56:11.267 >
00:56:11.267 > rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
00:56:11.267 > configsip: 0, SPIWP:0xee
00:56:11.267 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
00:56:11.267 > mode:DIO, clock div:2
00:56:11.267 > load:0x3fff0030,len:1184
00:56:11.267 > load:0x40078000,len:13192
00:56:11.267 > load:0x40080400,len:3028
00:56:11.267 > entry 0x400805e4
00:56:11.267 > E (607) esp_core_dump_flash: No core d�f���partition found!
00:56:11.270 > E (607) esp_core_dump_flash: No core dump partition found!
00:56:11.296 > 
00:56:11.296 > Starting OpenDTU
00:56:11.296 > Initialize FS... done
00:56:11.300 > Reading configuration... done
00:56:11.397 > Reading PinMapping... found valid mapping done
00:56:11.422 > Initialize Network... ETH start
00:56:13.183 > ETH connected
00:56:13.227 > done
00:56:13.227 > Setting Hostname... Configuring WiFi STA using new credentials... done
00:56:13.239 > Initialize NTP... done
00:56:13.239 > Initialize SunPosition... done
00:56:13.246 > Initialize MqTT... done
00:56:13.249 > Initialize WebApi... done
00:56:13.262 > Initialize Display... done
00:56:13.265 > Check for default DTU serial... done
00:56:13.268 > Initialize Hoymiles interface... Connection successful
00:56:13.274 >   Setting radio PA level...
00:56:13.279 >   Setting DTU serial...
00:56:13.282 >   Setting poll interval... 
00:56:13.285 > done
00:56:13.285 > Switch to Ethernet mode
00:56:13.288 > [  2021][E][WiFiGeneric.cpp:1166] mode(): Could not set mode! 12308
00:56:13.294 > [  2022][E][WiFiSTA.cpp:299] begin(): STA enable failed!
00:56:13.298 > WiFi disconnected
00:56:13.298 > Configuring Ethernet DHCP IP... done
00:56:13.306 > Setting Hostname... done
00:56:20.814 > ETH got IP: DTU-IP-ADDRESS
00:56:20.815 > Network connected
00:56:20.816 > Connecting to MQTT...
00:56:21.778 > Connected to MQTT.
00:58:04.117 > Websocket: [/livedata][1] connect
00:59:21.579 > Admin mode disabled
00:59:21.580 > E (190311) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffb2170 (config or if_key is NULL or duplicate key)
00:59:21.591 > E (190314) esp.emac: esp_eth_mac_new_esp32(595): alloc emac interrupt failed
00:59:21.599 > [190326][E][ETH.cpp:268] begin(): esp_eth_mac_new_esp32 failed
07:35:12.011 > Websocket: [/livedata][2] connect
07:35:12.095 > Websocket: [/livedata][1] disconnect
```

glucosum, 2023-04-26:
"esp_core_dump_flash: Core dump flash config is corrupted!"

I think you need to re-upload the program. It is possible that the memory has been corrupted.

DejanBukovec, 2023-04-26:
Device is new(2 months old) has been manually flashed by vscode and then repartitioned by update(which change partitions)...

tbnobody, 2024-03-31:
Closed due to inactivity

closed 2024-03-31 by tbnobody