tbnobody/OpenDTU #766: MQTT HomeAssistant Auto discovery do not work

DejanBukovec, 2023-03-31:
### What happened?

When enabling MQTT in OpenDTU it connect to broker but do not publishing a specially formatted JSON payload to Home Assistant...

In broker solar topic is visible and it refresh states:
[image]

Settings looks ok:
[image]

Other devices like shelly normally show in integration as new device...

### To Reproduce Bug

Im try enable/disable MQTT in OpenDTU but do not help.

### Expected Behavior

When MQTT is connected it publishing a specially formatted JSON payload to Home Assistant...

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

c47b7dd

tbnobody, 2023-03-31:
Are you sure you've enabled the home assistant feature in opendtu?

DejanBukovec, 2023-03-31:
Yes Im sure:
[image]

One question... I have inverters added to OpenDTU but they are not online(New installation and panels are not connected to inverters)... Can this be problem? That MQTT is send only if inverters are availible?

Live Data page also do not show anything:
[image]

tbnobody, 2023-03-31:
This might be the problem yes. The config is only published if the radio is idle: https://github.com/tbnobody/OpenDTU/blob/c47b7dd33502ee8dd9ea36d11da1c6886add7b8e/src/MqttHandleHass.cpp#L44

Problem hereby is, that if your poll interval is too low, the radio is never in idle mode because it has to wait for all the timeouts. On the other hand, the HASS config is only published when 
* Restart of DTU
* MQTT connection status changes
* MQTT settings where saved

tbnobody, 2023-03-31:
Just saw your second screenshot. You should see the inverters in the live view!
Can you make a screenshot of the inverter config?

DejanBukovec, 2023-03-31:
Im find problem... Typo in serial of test inverter... Now it show model on inverters list(Before model has been empty). Now it also show in main page:
[image]

And is send to HA MQTT...

[image]

Bug report can be closed and marked as false...

closed 2023-03-31 by tbnobody