tbnobody/OpenDTU #775: No WIFI Connection

OberstJohnJohnsen, 2023-04-06:
### What happened?

Nach dem flashen lässt sich das iPhone nicht mit dem AP des ESP32 verbinden. Andere Geräte wurden auch getestet.
Passwort "openDTU42" -> Ungültiges Passwort für OpenDTU-13171020

### To Reproduce Bug

ESP neu flashen -> neustarten -> per WIFI verbinden

### Expected Behavior

.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

f36a317 / 93c69db

tbnobody, 2023-04-06:
Bedeutet es, dass du von f36a317 auf 93c69db upgedated hast? Wenn du dort ein neues Passwort für den Admin Zugriff gesetzt hast ist dieses auch das Passwort für den AP.

tbnobody, 2023-04-06:
> Ich habe vor einem Monat eine Installation mit https://github.com/tbnobody/OpenDTU/commit/93c69dbbeb7bfea6ac552c28e9a779013455012b gemacht.

Das kann fast nicht sein. Die Version 93c69db ist von vor 2 Tagen. 

Generell kann es bei manchen ESP's helfen vorher einen kompletten Erase zu machen und alle .bin's neu zu flashen.

LagoGrande, 2023-04-06:
Im tool habe ich vorher den "Erase" Button gedrückt.

tbnobody, 2023-04-06:
Lass es mich mal so sagen, wenn es bei dem einen ESP klappt und bei dem anderen nicht, würde ich das Problem jetzt mal nicht bei der Software suchen sondern eher bei der Hardware. Hast du mal versucht, was passiert wenn du kein NRF Modul ansteckst?

Du könntest auch versuchen auf die Serielle Konsole zu schauen (mit z.B. Putty). Dort würden entsprechende Fehlermeldungen usw. ausgegeben.

LagoGrande, 2023-04-08:
Danke für die indirekten Tipps.
Anscheinend ist es beim Flashen zu einen Fehler gekommen -> Jetzt funktioniert alles bestens.

Danke!

closed 2023-04-08 by LagoGrande