tbnobody/OpenDTU #777: MQTT Hoymiles HM-300 Reachable on/off "spam" in Home Assistant log.

Neruk, 2023-04-06:
### What happened?

After updating from [7e28336](https://github.com/tbnobody/OpenDTU/commits/7e28336) to [778dc93](https://github.com/tbnobody/OpenDTU/commit/778dc93deec71a8cdddf3e8b0083cf2291f45d6e) my Home Assistant log gets filled with "Hoymiles HM-300 Reachable" on/off messages. PV production seems to be working just fine without any interruption.

**My setup:** 
[image]

HM-600
- no issues with "Hoymiles HM-XXX Reachable" messages.

HM-300
- "Hoymiles HM-300 Reachable" on/off messages. (see screenshots)
- repeating within a frequency of a few sec and a couple of mins since sunrise. 

[image]
[image]

After rolling back to [7e28336](https://github.com/tbnobody/OpenDTU/commits/7e28336) the issue stopped again.

For more information just shoot me a message in English or German.

### To Reproduce Bug

.

### Expected Behavior

.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

778dc93

tbnobody, 2023-04-06:
Thats interessting, because I didn't even change anything regarding the Hoymiles fetch logic or MQTT publish methods:
https://github.com/tbnobody/OpenDTU/compare/7e28336...778dc93

Neruk, 2023-04-06:
Since I got a few spare minutes, I did another test and tried [93c69db](https://github.com/tbnobody/OpenDTU/commits/93c69db) (I skipped this build above). Unfortunately the issue is still the same maybe a little less frequent. It took about 10min before it started to occur.

[image]

tbnobody, 2023-04-06:
Could you doublecheck whether you see this toggle also in the web UI of OpenDTU (should toggle between red and blue/yellow) or if you see this toggle on the mqtt broker?

Neruk, 2023-04-07:
I assume you mean this bar (screenshot)?
[image]

Unfortunately I don't know where to check this with the mqtt broker. From what I recall, it did not change, but I will have a look in the morning.

Do you have any preferred build you'd like me to flash, else I am going to use [93c69db](https://github.com/tbnobody/OpenDTU/commits/93c69db) since it has the least changes and includes the bug?

Recap:

- last working version for me was [7e28336](https://github.com/tbnobody/OpenDTU/commits/7e28336)
- first appearance (bug) with [93c69db](https://github.com/tbnobody/OpenDTU/commits/93c69db)

Neruk, 2023-04-07:
### It does turn RED on the OpenDTU Web UI!
[image]

Something else I've noticed is.. that the "Data Age" (Letzte Aktualisierung) timer is very inconsistent with [93c69db](https://github.com/tbnobody/OpenDTU/commits/93c69db) compared to the previous build [7e28336](https://github.com/tbnobody/OpenDTU/commits/7e28336).

[image]
_This can go up to 12sec **WITHOUT** turning the bar red and leaving **NO** message in my logs ("Hoymiles HM-300 Reachable" on/off)
With the old build [7e28336](https://github.com/tbnobody/OpenDTU/commits/7e28336), the worst is 5sec which is totally fine (margin of error) since my refresh cycle takes 4sec._

### --- NOT RELATED TO THE BUG ---
In order to get new data from both inverters, I had to reduce the polling rate from 5sec (default) to 2sec (custom) to get fresh data (from **both** inverters) before it gets published every 5sec (default) with MQTT.

[image]

**Is this intended behaviour?**
0:00:00 DTU ---> requests data from inverter **1** (HM-600)
0:02:00 DTU <--- receives data from inverter **1** (HM-600)
0:02:01 DTU ---> requests data from inverter **2** (HM-300)
0:04:00 DTU <--- receives data from inverter **2** (HM-300)
0:05:00 DTU ---> sends data to mqtt broker

I know most people don't use two (2) or more inverters, but isn't it a huge bottleneck the way OpenDTU requests data from the inverters at the moment? "I know why it is the way it is" (one job at the time / line by line) and sadly know to little about coding to understand the details of this project, but wouldn't it be possibly to rewrite the code to send out **ALL** requests first, instead of waiting until the first request has finished?

**Something like this:**
0:00:00 DTU ---> requests data from inverter **1** (HM-600)
0:00:01 DTU ---> requests data from inverter **2** (HM-300)
0:02:00 DTU <--- receives data from inverter **1** (HM-600)
0:02:01 DTU <--- receives data from inverter **2** (HM-300)
0:05:00 DTU ---> sends data to mqtt broker

_Just to clarify the custom poll rate (2sec vs 5sec) is something I had to do since day 1._

tbnobody, 2023-04-07:
> but wouldn't it be possibly to rewrite the code to send out ALL requests first, instead of waiting until the first request has finished?

Thats not how RF is working. If you are sending data to the first inverter it immediatly beginns with the answer. The answer consists of multiple packages. You have to wait for all of them. While receiving data. You cannot send data.

Main question therefor is, why is it getting red. It's getting red because https://github.com/tbnobody/OpenDTU/blob/12d7349699ba2c8e35883b7cbfcc98b34a92a294/lib/Hoymiles/src/inverters/InverterAbstract.cpp#L75

Do don't receive data for at least two requests. What shows the console?

Neruk, 2023-04-09:
Here is the console raw output (build: [93c69db](https://github.com/tbnobody/OpenDTU/commits/93c69db)).

OpenDTU console: https://drive.google.com/file/d/1B_TjJj-LGPPWtakreQ8H6a-sU8P0hedu/view

Screenshot with Home Assistant log: 
[image]

mag2bue, 2023-04-11:
I had the same problem and I was able to fix it with the transmit power. More power helps :-)

[image]

Neruk, 2023-04-13:
> I had the same problem and I was able to fix it with the transmit power. More power helps :-)

@mag2bue Are you running 2 inverters or just one and which model/s?

Hmm.. that's weird my HM-300 (having issues) is approx. 1m away (no walls + in line of sight) from the DTU and my HM-600 (no problems) is outdoors 4-5m away including a wall and a roof.

Also I am using the "antenna" version.

[image]

mag2bue, 2023-04-15:
I'm have

> @mag2bue Are you running 2 inverters or just one and which model/s?
>
> Hmm.. that's weird my HM-300 (having issues) is approx. 1m away (no walls + in line of sight) from the DTU and my HM-600 (no [...]

It's the same with me. No wall in between and only 1 meter away from the inverter.

Did the setting not help you?  How is the wifi reception?

ederpatrick, 2023-04-16:
I have the same problem on a single hm-1500 with antenna version 1m between hm and esp32... but i was running with a release from 1 year ago. So i assume it might not be direct related to dtu release as i not changed it.
Was running since months without issues but since a few weeks i am seeing dtu live view gets red very often without any logical reason, changing transmit power to high makes it worse. Repositioning a bit of the esp might make it bettee. And it works best for me with lowest power but even then after some time it starts getting bad.

Now i upgraded to newest release of dtu but its happening with it too.

HA is allways Uptodate also the addons
Mosquito reconfigured (using 3.1.1 tcp)
Firmware HM-1500 v23.4.15
Esp32 is powered by a 2A powersup

ederpatrick, 2023-04-25:
@tbnobody  sry to bother you probably, but do you have any ideas where our issue might come from and can i do something against this "disconnects"?  

It would be cool to have a feedback as the main problem is when i run some automations via home assistant (for example if batterie pack is getting empty reduce limit of hoymiles to 1%) it might be sending the limit commands in that moments when dtu isnt in contact with the hoymiles (red color) and so hoymiles ignores it. Also i have sometimes problems in the morning when for example my batterie is charged again to 20% via PV and i decide via automation hoymiles should go on with limit 20% he might not start and just ignores all commands and stays red.

In HA the reachable entitie is going off/on all the time..
DTU setting is 5seconds as you recommendet

BerriJ, 2023-07-26:
I had the same problem. Adding a 100 μF capacitor to gnd/vcc of my NRF solved the problem.
I bet this is not software related at all.

ederpatrick, 2023-07-26:
Hi BerriJ,

i agree a capacitor solved the issue in HA permanent and its reachable all the time now via HA.
Definatly no sw issue!
Thanks and Regards

closed 2024-01-29 by tbnobody