tbnobody/OpenDTU #783: [Request] Less information when logged out and PWA

florian-senn, 2023-04-07:
### Is your feature request related to a problem? Please describe.

Hi!

Vielen Dank für  die Entwicklung von OpenDTU, es ist wirklich eine sehr tolle Lösung  für Hoymiles-WR!

Ich nutze OpenDTU mit einem Traefik Reverse Proxy und öffentlich zugänglicher Domain. Deshalb hat es mich gestört, dass gewisse Optionen und Werte öffentlich sichtbar sind.

---

Hey there,

thanks for OpenDTU, it's a very convenient and good working solution!king solution!

I'm running OpenDTU behind a Traefik SSL Reverse Proxy so it is publicly available - but I want to hide some information and the menu for non-logged in visitors, so I've made same changes to the web app (some additional v-ifs) :

[image]

I've also encountered non-logged in visitors possibility to navigate to options/security and change certain options like visibility. Maybe there's some part in the router missing, but I couldn't find which one.

What about implementing this "extra privacy mode" as another option.

Further I'd like to suggest PWA, so we can install OpenDTU on smartphones and use them like native apps.

Thanks and best regards

### Describe the solution you'd like

Entsprechend habe ich ein paar v-ifs hinzugefügt. Ich denke man könnte dies noch um eine "super privacy mode"-Einstellung ergänzen, falls Abstufungen gewünscht sind. Weiters habe ich festgestellt, dass es möglich ist, auch abgemeldet in gewisse Einstellungsbereiche zu kommen und dort Einstellungen zu ändern. Den entsprechenden Teil im Router konnte ich nicht finden. Weiters schlage ich vor, PWA zu implementieren, damit OpenDTU als App installier- und nutzbar ist. Der Aufwand dafür ist wohl überschaubar, leider habe ich mit Vite keine Erfahrung und bin bisher daran gescheitert.

tbnobody, 2023-04-09:
> Weiters habe ich festgestellt, dass es möglich ist, auch abgemeldet in gewisse Einstellungsbereiche zu kommen und dort Einstellungen zu ändern. 

Welche Bereiche sind das?

florian-senn, 2023-04-09:
Ich kam in die Sicherheitseinstellung unter /settings/security und konnte dort die Nur-Lese-Berechtigung ändern.

Allerdings kann ich es derzeit nicht reproduzieren, da mein build inzwischen die Routen schützt (siehe pull request) und ich eben sämtliche Menüpunkte im ausgeloggten Zustand ausblende.

tbnobody, 2023-04-09:
Kann normal nicht passieren, ist im Backend nicht zugelassen. Wenn nicht die korrekten Headers mit Username und Passwort kommen wird dort nichts zurück gegeben.

florian-senn, 2023-04-09:
Sobald ich zuhause bin, setz ich das offizielle Build mal auf einen ESP32 und versuche es zu rekonstruieren. Kann es sein, dass diese Einstellung in einem lokalen Storage oder durch VueJS zumindest im Frontend aktualisiert wird, ohne dass das Backend direkt beteiligt ist?

tbnobody, 2023-04-09:
> Kann es sein, dass diese Einstellung in einem lokalen Storage oder durch VueJS zumindest im Frontend aktualisiert wird, ohne dass das Backend direkt beteiligt ist?

Nein, es wird direkt vom Backend gelesen und auch direkt ans Backend geschrieben.

quangnguyen, 2023-04-15:
Beiseite, Traefik kann doch Middlewares wie z.B. BasicAuth oder Oauth2 für bestimmten Paths.

stefan123t, 2024-07-10:
@florian-senn hast Du das vermeintliche security Problem mit http://opendtu-ABCDEF/api/settings/security reproduzieren können. Ich habe die Tage mal die API Dokumentation und den Code angesehen und das sollte m.E. mit BasicAuth geschützt sein.

Abgesehen davon was meinst Du mit Progressive Web App (PWA) und willst Du dafür ein eigenes Issue aufmachen ?

Hier ein Artikel um eine Vue.JS Anwendung als Progressive Web App zu modifizieren und z.B. auf dem Smartphone zu speichern: 
https://blog.logrocket.com/building-pwa-vue/

@tbnobody geht das wirklich soo einfach ?

```bash
vue add pwa
```