tbnobody/OpenDTU #788: [Configuration] Backup WLAN

m-froehlich, 2023-04-10:
### Is your feature request related to a problem? Please describe.

The OpenDTU only allows a single WLAN to be configured. Problem is, that I have to blindly configure another WLAN, if I need to carry it physically into the reach of another AP. If I do a typo somewhere, it is difficult to get along.

Concrete situation: I configure the OpenDTU in my living room. When I'm done I carry it outside into the garage, where it is supposed to be running the next years. In the garage the inside WLAN is unreachable. Instead there's an AP just for the outside area.

### Describe the solution you'd like

From my Shellys I know the option to configure two WLANs, a primary one and a backup.

Would you mind doing the same for the OpenDTU?

tbnobody, 2023-04-10:
Duplicate of #66 

> If I do a typo somewhere, it is difficult to get along.

Why is it difficult? It will open up a AP again if it cannot find the configured AP or some minutes after power up. You can connect with a mobile phone, tablet or laptop and change the settings again?
On the other hand this is a one time use case. I mean, its not like you are moving this device around every day.

m-froehlich, 2023-04-10:
Well for one it's cold in the garage.

And additionally in the first time when I set up the DTU I had to rest and carry it around multiple times.

And I have to switch off Mobile Data every time I connect with the AP.

Yes, it works without the backup WLAN. But also there's a reason for the Shellys to have one, isn't there?

tbnobody, 2023-04-10:
I am not yet convinced. but it any case it's a duplicate of #66. therefor I am going to close this one.

closed 2023-04-10 by tbnobody

m-froehlich, 2023-04-10:
I'm fine with closing this issue. And of course I can live without a second WiFi, because it works without it somehow. But still it helps a lot in the setup time and removes some stumble stones.