tbnobody/OpenDTU #790: [Request] Add speedometer chart controlled by analog servo

hag2001, 2023-04-10:
### Describe the solution you'd like

Demo: Speedometer controlled by Analog Servo, 4.8-6.0V, 0.12/0.1 s/60° 

[image]

tbnobody, 2023-04-10:
This Projekt is meant as a  bridge between the hoymiles protocol and mqtt / web UI. Therefor I will not implement such this as servo control etc. You are free to realize this by modifying the code yourself or use a second ESP which connects to opendtu using web UI or mqtt. 
But unfortunately, in order to keep the code clean, stable and expandable, i can't include any display idea.

hag2001, 2023-04-10:
Sure, I've implemented this already ... If somebody is interested ...

tbnobody, 2023-04-10:
I am going to move this to the discussions as it is more visibile there