tbnobody/OpenDTU #797: Static IP Address on WiFI Interface don't work

Technikfan, 2023-04-11:
### What happened?

Fresh flashed ESP32 Pico Kit  Development Board v4
After set a static IP address the ESP32 isn't reachable over this but the connection to the AP address 192.168.4.1 is hold.
After cold start it isn't possible to reconnect to the SSID of ESP32 AP and also not to the static address.

### To Reproduce Bug

I can follow steps reproduce:
Fresh ESP32 flashing via VSCode works.
Then i connect via Wifi to ESP32 with hardcoded IP address 192.168.4.1
Then i set the parameter for access to my Wifi SSID.
Now then are two path are possible:
Path 1) 
I set the DHCP CLient Mode off and set a static IP Address
The static address isn't reachable.
After an coldstart of ESP32 in no access to the ESP32 possible.
The static address isn't reachable and an new connect to the SSID of ESP32-AP also don't possible.
The SSID is shown but i can't connect to them.
Path 2) 
I let DHCP Mode on and the ESP32 get's an IP address from DHCP Server.
The ESP32 is further reachable via 192.168.4.1 and the given new IP Adress.
Then i set a new static address, but the address don't goes active.
The DHCP Client IP is further binded to interface.
After cold reboot the ESP32 isn't reachable again.
Same result as Path 1)

It is also not possible to call up the system info.

### Expected Behavior

After set an static IP address should be possible to access to the address.
Also after reboot.
Connect to AP-Mode SSID should work.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

I can't show

tbnobody, 2023-04-11:
Just doublechecked it here... works perfectly... did you also enter the correct netmask and (if required) default gateway?

Technikfan, 2023-04-12:
Hi, yes i did and i am in the same flat network. The netmask and the gateway can't any show-stopper.

Technikfan, 2023-04-12:
As you ask i remember there is an additional information, maybe it is interesting.
In the info-menu is a point for look on network information.
I see information about an Ethernet-Interface and an Wifi AP Interface.
Stands the "Ethernet Interface" for the WiFi Client?
There are no Address-Informations on screen., only the netmask and the hostname is filled, the rest are ever 0.0.0.0.

Yust i tryed it again.
I recognize, that i go after fresh upload and connect to 192.168.4.1 on "openDTU-.." and first change the informations to the SSID in my network and then set the static ip-addresses, the ESP received between this steps an IP address from my DHCP Server and hold it. To this point the setted ip address is ignored.
Best regards

tbnobody, 2023-04-12:
>  Stands the "Ethernet Interface" for the WiFi Client?

Nope, it means that you are using the Ethernet Mode of the firmware. This happens if you enable the Ethernet PHY. 

* Does your board support ethernet? 
* Please provide a screenshot of Settings --> Hardware?
* What version are you using?

Technikfan, 2023-04-12:
There is no Ethernet-Interface on the ESP32. 
I use this on [https://www.elektor.de/esp32-pico-kit-v4](url).

[image]
[image]
[image]

I cloned from github in VSCode at last Monday so described in the README.md.
I can't see the Hash in the System-Info-Page, because there comes no informations.
[image]

But in the config.json is this information: "version":71680 and the configured ip-adresses are also saved inthere.

From your README.md:
### First configuration

- [ OK] After the initial flashing of the microcontroller, an Access Point called "OpenDTU-*" is opened. The default password is "openDTU42".

- [OK ] Use a web browser to open the address http://192.168.4.1/

- [OK ] Navigate to Settings --> Network Settings and enter your WiFi credentials. The username to access the config menu is "admin" and the password the same as for accessing the Access Point (default: "openDTU42").

- [ NOK] OpenDTU then simultaneously connects to your WiFi AP with these credentials. Navigate to Info --> Network and look into section "Network Interface (Station)" for the IP address received via DHCP.

- [NOK ] If your WiFi AP uses an allow-list for MAC-addresses, please be aware that the ESP32 has two different MAC addresses for its AP and client modes, they are also listed at Info --> Network.

- [ NOK] When OpenDTU is connected to a configured WiFI AP, the "OpenDTU-*" Access Point is closed after 3 minutes.

- [ OK] OpenDTU needs access to a working NTP server to get the current date & time. Both are sent to the inverter with each request. Default NTP server is pool.ntp.org. If your network has different requirements please change accordingly (Settings --> NTP Settings).

- [OK ] Add your inverter in the inverter settings (Settings --> Inverter Settings)

P.S.: In the network-settings.cpp is an entry for include ETH.h, but i don't find this file.

tbnobody, 2023-04-12:
> I cloned from github in VSCode at last Monday so described in the README.md.

Which environment did you flash to the ESP? You should use the generic environment. Or try the pre-compiled bin.

Technikfan, 2023-04-12:
I used the generic enviroment.

_[platformio]
default_envs = generic
...
[env:generic]
board = esp32dev
build_flags = ${env.build_flags}
    -DHOYMILES_PIN_MISO=19
    -DHOYMILES_PIN_MOSI=23
    -DHOYMILES_PIN_SCLK=18
    -DHOYMILES_PIN_IRQ=16
    -DHOYMILES_PIN_CE=4
    -DHOYMILES_PIN_CS=5_

Technikfan, 2023-04-12:
Now i flashed with the latest precompiled-bin files.
The same result.

tbnobody, 2023-04-12:
Your doesn't seem to use a simple ESP32 compatible MCU. Seems to be a "ESP32-PICO-D4" MCU. 
This MCU seems to have less hardware components (e.g. the ethernet PHY is missing). 

Therefor you have to create an build environment yourself. maybe `board = pico32` would be a start. But you have to test this yourself as I don't have such a board with such a MCU.

Technikfan, 2023-04-13:
With esp-wroom-32 it works.
Thanks for support.
Today the code don't work on esp32-pico

fischaxe, 2023-04-15:
Maybe this [answer](https://github.com/espressif/arduino-esp32/issues/4436#issuecomment-715902830) in expressif repository might help you.

```
[env:pico32]
board = pico32
build_flags = ${env.build_flags}
    -DHOYMILES_PIN_MISO=19
    -DHOYMILES_PIN_MOSI=25
    -DHOYMILES_PIN_SCLK=21
    -DHOYMILES_PIN_CS=22
    -DHOYMILES_PIN_IRQ=4
    -DHOYMILES_PIN_CE=5
```

Technikfan, 2023-04-15:
This is still a problem.

But step by step:
I realised that I made a mistake in my first attempts.
I looked for and used the same IO pins as in the generic environment.
But today I see that one of these pins is not connected to the breadboard.
That's how blind you can be.

Then I used the pins you suggested.
At the first moment I know, yes it works, but then....
Bad synchronisation with the HM1500.
After restarting, no access to the DTU.

I found some documentation, there are the eye opening tips in there.
The ADC2_CHx pins cannot be used when WiFi is used.
And on the other hand, some of them are INPUT pins only.
So I looked for other IO pins.
I thought at first that it would be better if the I2C pins were free for further use.

So I tried the pins ADC1_CH4 and 5 for IRQ and CE.
But then again bad sync and after reboot no access to the DTU.

Now I tried to use the I2C pins for IRO and CE.
But the same game.

I wonder what is the basic problem?

Another comment:
When DTU has no sync to DTU and I click on the manu line in Live-View to manage max power values from HM-1500, nothing comes.

And for ease of use, I think this feature should be better in Settings->Inverter.

Best regards

Technikfan, 2023-04-16:
Hi tbnobody,

with this enviroment setup it seems to work now:

[env]
monitor_port = COM10
upload_port = COM10

[platformio]
default_envs = pico32

[env:pico32]
; https://www.elektor.de/esp32-pico-kit-v4
; https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-pico-kit.html#
platform = espressif32
board = pico32
;upload_flags = -Dserial_dtr=0
;upload_flags = -Dserial_rts=0
upload_speed = 115200
monitor_speed = 115200
build_flags = ${env.build_flags}
    -DHOYMILES_PIN_MISO=12
    -DHOYMILES_PIN_MOSI=13
    -DHOYMILES_PIN_SCLK=14
    -DHOYMILES_PIN_IRQ=10
    -DHOYMILES_PIN_CE=9
    -DHOYMILES_PIN_CS=15

I use the HSPI instead of the VSPI pins on the ESP32-PICO-D4 board to connect the SPI interface to the NRF24L01+.
In the next few days I will receive OLED displays and will look into adding a display to the DTU.  We'll see if we can get it to work with SPI as well.

Yet i'm not sure, over the stabiliti of network stack. I recognized twice that the time-Sync with NTP server is broken and when i go to NTP Settings and klick on "Synchronize time" all is fine.
I'll watch this.

Best regards

Technikfan, 2023-04-16:
with this enviroment setup it seems to work now:

I use the HSPI instead of the VSPI pins on the ESP32-PICO-D4 board to connect the SPI interface to the NRF24L01+.

[platformio]
default_envs = pico32

[env:pico32]
; https://www.elektor.de/esp32-pico-kit-v4
; https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-pico-kit.html#
platform = espressif32
board = pico32
;upload_flags = -Dserial_dtr=0
;upload_flags = -Dserial_rts=0
upload_speed = 115200
monitor_speed = 115200
build_flags = ${env.build_flags}
    -DHOYMILES_PIN_MISO=12
    -DHOYMILES_PIN_MOSI=13
    -DHOYMILES_PIN_SCLK=14
    -DHOYMILES_PIN_IRQ=10
    -DHOYMILES_PIN_CE=9
    -DHOYMILES_PIN_CS=15

In the next few days I will receive OLED displays and will look into adding a display to the DTU.  We'll see if we can get it to work with SPI as well.

Yet i'm not sure, over the stabiliti of network stack. I recognized twice that the time-Sync with NTP server is broken and when i go to NTP Settings and klick on "Synchronize time" all is fine.
I'll watch this.

Technikfan, 2023-04-24:
I have a newbie question..
I received the displays now and try to add one to the Pico Board.
I had prepare an own pin-mapping.json file and did understand the prodeure so, that i go to configuration managment and upload the config file as pin_mapping.json.
And then the esp reboot and i shuld find the Config-Profile in hardware management.
But i don't find there the profile in the drop-down field.
An manual reboot and additional reload of json makes no difference.
Best regards
[pin_mapping.zip](https://github.com/tbnobody/OpenDTU/files/11311991/pin_mapping.zip)

Technikfan, 2023-04-24:
Thanks, I also realised that the json was the problem, but you are faster :-).
I tested a sample file from github and it worked fine, so only my file could be the cause.

Everything is working fine now.
The part has now been busy for several days without any dropouts and the display also runs straight away (if you do everything right :-) ).
I think the case can be closed as successfully solved.

I used for the display the I2C_SDA on PIN 21 and the I2C_SCL on PIN 22, classic use.

Thanks for your support and best greatings.

closed 2023-04-24 by Technikfan