tbnobody/OpenDTU #814: Only 5 Hoymiles mikro inverters can be checked for inverter info

baba-joe, 2023-04-17:
### What happened?

If more than 5 Hoymiles inverters are added inverter info can only be retrieved for the first 5 inverters. The 6th will give an error message in a window reading "No Information available Did not receive any valid data from the inverter till now. Still trying..." and the 7th shows a rotating open circle...

### To Reproduce Bug

Add more than 5 Hoymiles mikro inverters.

### Expected Behavior

Since the documentation claims that 10 inverters are supportet I would expect to be able to retrieve inverter info for all 10 possible inverters.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.4.15

### Anything else?

I happen to have 10 Hoymile mikro inverters available for testing in case you are interested?

tbnobody, 2023-04-17:
After adding the inverters, how long did you wait until you pressed the inverter info button?

baba-joe, 2023-04-17:
I have added them a day ago but still the same behaviour. Also I have tried to swap inverters by removing some and keeping the ones not showing info added. It always works OK with 5 or less inverters but as soon as I add a 6th it is not possible to retrieve info. How to I have to use the console to get the output you need? It seems to be not possible to keep the console open while trying to retrieve inverter info?

baba-joe, 2023-04-17:
[debug_console_OpenDTU_inverterinfomissing.txt](https://github.com/tbnobody/OpenDTU/files/11248623/debug_console_OpenDTU_inverterinfomissing.txt)

tbnobody, 2023-04-17:
The issue was automatically closed. Please doublecheck whether it's fixed now

baba-joe, 2023-04-18:
No, how would it?
[image]
[image]

tbnobody, 2023-04-18:
> No, how would it?

Which version are you running?

baba-joe, 2023-04-18:
still 23.4.15 

was not aware about an update and waiting for your response

let me check after I pushed 23.4.17

baba-joe, 2023-04-18:
now I can retrieve info for all 7, very nice, thanks!

commit 59f43a82 (2023-04-17): BREAKING CHANGE: Web API Endpoint /api/devinfo/status requires GET parameter inv=

Only the requested inverter is sent to the client. This reduces the size of the json response --> Less memory.

It should also fix #814.
closed 2023-04-18 by tbnobody