tbnobody/OpenDTU #815: Implementation of zero feed-in (Nulleinspeisung) for multiple inverters based on mqtt topics within OpenDTU

juergen-weber, 2023-04-17:
### Is your feature request related to a problem? Please describe.

no

### Describe the solution you'd like

Due to the regulations that apply to the power grid market, a zero feed-in function seems to be essential for power grids. At the very least, such a function may reduce the administrative nightmare of registration.
Power meters like the Shelly EM3, which is widely used in the DIY scene, publish the current power flows via mqtt. Together with the inverter numbers collected by OpenDTU, a zero power function can be created.
Challenging is the support of a zeo feed-in function for multiple inverters.

### Describe alternatives you've considered

Implemention of zero feed-in functionality within systems like Home Assistant.

DejanBukovec, 2023-04-17:
This can be done different ways...
One of them is that OPENDTU listen to a predefined topic(Which we can set in settings) and is power_consumption request. Then OpenDTU depend of value and inverter count and panel power settings increase "non persistent limit" for inverters to get power_consumption to as close as possible "target_value" set in OpenDTU settings... Also if possible to have some hysteresis setting will also be ok...

I will be happy if there is also included issue #681 which is about setting inverters to power set by "persistent limit" when MQTT disconnect to prevent to much power go to grid... Some country's have limit 600W, some 800W some users want 0W... So for 

For example when we have 3 inverters and set "persistent limit"  on each of them 200W max possible total output when MQTT stop working or when inverrters reboot will be 600W. When MQTT reconnect inverters will start increasing/decreasing "non persistent limit" to get "power_consumption" MQTT topic as close as  "target_value" ...

philippsandhaus, 2023-04-18:
There are already solutions available which realise this by running on an external system: https://github.com/reserve85/HoymilesZeroExport
https://github.com/ThePradox/SolarExportControl

An addition, there is a fork of OpenDTU which is adds functionality to charge a battery in favour of exporting to the grid, but it can also be used without a battery for setting the inverter limits for zero export. The downside is that currently only one inverter is supported: https://github.com/helgeerbe/OpenDTU-OnBattery

DejanBukovec, 2023-04-18:
Yes zero export can be done by using shelly script or some homeassistant automation... But it will be nicer if OpenDTU can do it itself... Without any other software/device involved...

[OpenDTU-OnBattery](https://github.com/helgeerbe/OpenDTU-OnBattery) is not option if support only one inverter...

But none of this solutions will not solve issue #681 when network/server do not work...

cthiele-mogic, 2023-05-10:
Are there any working HA automations? I found only Python scripts or node-red stuff for this issue, but it seems much easier to integrate directly in Home Assistant. If you have any links / addon / integration let me know.

DejanBukovec, 2023-05-11:
I didn't install my solar system yet but I think that in HA Automation this can be done easy... Depend of how complicated system do you want to have...
Best aproach I think will be(It is not tested):
1.) Install house power meter and integrate it to HA(ZWave, Zigbee, Wifi...). I use Qubino ZMNHXD1 which have power consumption and production for each phase. Set it to send changes on your preferred interval or power change...
2.) Integrate OpenDTU to MQTT and HA to get info of current power limit and option to set it.
3.) Make automation which is triggered by power change on your meter. If you have solar system on one phase then make one trigger if you have 3 phase system then make 3 triggers and assign them ID's to know which one has been changed.
Then if you have 3 phase solar system as action check which trigger has been triggered and depend of it continue and set power to correct inverter. If you use 1 phase system then you can skip checking trigger.
4.) Then check current power consumption on phase which has change power consumption and if is negative then decrease current power limit for value of negative power consumption if is positive then increase limit... Also be aware that inverter need some time to set limit so to often triggered automation will not work ok... If you have more than 1 inverter then you need divide positive/negative power consumptions with inverter count and increase power limit to all inverters...

There are many different setups so each configuration will be different...

jstammi, 2023-05-12:
Hi,

openDTU integration to HA is done easily after having
* configured & enable MQTT HA integration
* configured "Home Assistant MQTT-Auto-Discovery-Parameter" accordingly
* enabled "Home Assistant MQTT-Auto-Discovery" option in openDTU

Afterwards all devices and entities appear "magically" in HA :-)
Not only for the sensor, but also for controlling/commanding part, see images below.

Additionally I have an ESPHome based reader at the IR interface of the main power meter. That way I know among others the total actual incoming/outgoing current (<5€ if soldering yourself - here it is a REALLY very simple task).

With such everything is available in principle for close-to-zero-feed-in (there will alway be some delay from measuring to regulating the actual current to ~0 netto. Which I do not do, yet, as I do not yet have a battery in place where the remaining current could go to.

At the moment I just generate some notification on out mobiles in case we are not consuming our "Balkonkraftwerk" pv current - to start then e.g. the washing machine (just done again).

[image]
[image]
[image]

Ulliboy, 2023-05-13:
Hi jstammi,

super, hört sich sehr gut an und Home Assistant läuft bei mir seit einigen Tagen auf einem Fujitsu Mini-PC. 

Für Neueinsteiger, wie mich, sind aber die Begrifflichkeiten in den Masken oft unklar und verwirrend. Könntest Du bitte ausgefüllte Screenshots von der MQTT Maske in OpenDTU und die relevanten Masken in Home Assistant (Moskito?) einstellen. Das würde mir helfen, die Sache zu verstehen. Sozusagen als Anschub.

Ulliboy, 2023-05-15:
Hi jstammi,

die Verbindung von OpenDTU zu Home Assistant hab ich gestern noch bekommen, die Übergabe läuft. War nur ein Berechtigungsproblem. Wie Du schon geschrieben hast, läuft automatisch. Einfach genial gemacht, danke.

Schön wäre natürlich, wenn das Setzten des jeweiligen Limits direkt automatisch von OpenDTU gemacht würde, nachdem mqtt den Wert direkt vom z.B. "hichi ir wifi tasmota Lesekopf" per wifi angeliefert hat. Dann wäre da nicht noch ein fragiles Automatisierungssystem und weitere Hardware dazwischen. Die stabile Funktion ist einfach zu wichtig.

jstammi, 2023-05-16:
Perfect. I am on vacation, so I could not really help last days (still ongoing ;-)).

For the zero feed-in:
I am not familiar with the inverters behaviour on setting limits at some rate up to now.
Do they apply a limit (close to) immediately (and please do not judge on this by looking to openDTU values, because those may be up to 5s old at max fetching frequency)?
Does this differ by the models?
How often is it possible/allowed to set a new limit?

E.g. my power meter provides it's power value at >1Hz AFAIS. Does it make sense to set a new limit at such rate?

By today I expect that there is needed some model specific heuristic of how to update the limit. And such would make it IMHO much more complicate to implement than it may seem.
Or does someone know of a simple and always valid/proven to work algorithm on this?

cthiele-mogic, 2023-05-16:
I think its not so easy, the value can't be set too often - the limit value should be the same for X minutes and the limit should be only updated if the avg of the last X minutes (maybe other value) is too high...otherwise it would be possible that the limit is updated every 5s  and I would say this wouldn't be good for the inverter lifetime.

Ulliboy, 2023-05-16:
Das Limit wird in "non persistant" geschrieben. Dann wird nichts auf der Hardware des inverters gespeichert.

Es gibt auf GitHub und Discord einige Berichte von Leuten, die es schon mit unterschiedlichen Ansätzen realisiert haben.

[https://discord.com/channels/984173303147155506/1051985890219663430 ](https://discord.com/channels/984173303147155506/1051985890219663430)

Am einfachsten und übersichtlichsten wird es vermutlich ausschließlich mit mqtt sein. Ich habe auch noch keinen digitalen Stromzähler, den ich auslesen kann, will erst nächstes Jahr anmelden.

Schönen Resturlaub.

stefan123t, 2024-07-10:
@juergen-weber please double check the Dynamic Power Limiter (DPL) function in the daughter project[ OpenDTU-OnBattery](https://github.com/helgeerbe/OpenDTU-OnBattery/).

I have described a https://github.com/tbnobody/OpenDTU/issues/272#issuecomment-2219644352 with a Shelly Pro3EM as Power Meter (PM) here in issue #272.

DejanBukovec, 2024-07-11:
Around half year ago I switch to AhoyDTU which work better in my case with 3 HM-800 inverters. After that I never switch back to OpenDTU and try if issues are fixed... I have plan use one DTU per inverter to speed up zero-export function and then I will maybe switch back to OpenDTU.

Half year ago I develop Zero export ESP32 based device which can read data from several meters(Shelly, MQTT, HomeAssistant) and also use dedicated solar sensors and offer option to use it on 1 phase or 3 phase grid and regulate zero export per phase...

stefan123t, 2024-07-11:
Thanks for your response. Interesting to hear that you are solving the ActivePowerLimit update interval constraint using more hardware (TM) / multiple DTUs.

Let us wait for the OP @juergen-weber to reply.