tbnobody/OpenDTU #817: NRF24 not connecting - v23.4.17 (ahoy works :-( )

njordan77, 2023-04-18:
### What happened?

I was on AHOY wit an NRF24L01+ board. worked fine and did not run into any communications issue between the ESP32 and my NRF board.

Now switched to OpenDTU, its a very nice and more responsive GUI. But whatever i try the board does not seem to connect.

Info in the console on serial at boot:
```
Check for default DTU serial... done
Initialize Hoymiles interface... NRF: Connection error!!
  Setting radio PA level... 
  Setting DTU serial... 
  Setting poll interval... 
```

I did use the standard PINs like 4,516,18,.....even did change the device profile to "standard/default" and saved, but i guess it is set anyway if no other is available.

Did check all connections if a cable is broken.....

Did reflash the image via ESP-flasher tool.

I system-info in GUI it looks like this which is consistent with the situation that it does not work. Tried to find some helpful discussions here, but seems this issue is not discussed at all.

Sorry, and thanks a lot for supporting!
Norbert

### To Reproduce Bug

any time after the boot its this situation,

### Expected Behavior

Standard NRF+ Chip should connect.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.4.17

mroenne2022, 2023-04-18:
Please have a look and check this : 
[image]

mroenne2022, 2023-04-18:
Hello @njordan77 , i have a ESP32/ 38Pin , at first with Ahoy and the IRQ of GPIO 1 . Now i use OpenDTU with the same ESP32 Board and use GPIO 16 for IRQ. Perhaps you can try it too !?
[image]

njordan77, 2023-04-18:
thanks for your reply and input. i've seen this discussion before and was hoping that this could fix my issue. But i already was on GPIO16 as this seems to be the standard port. i also tried gpio1 (even though it was not very likely). but did not work.

njordan77, 2023-04-18:
I'm using these PINs, but i guess this is standard anyway. did not change any PIN configuration from the default binary file.

PIN_MISO=19
PIN_MOSI=23
PIN_SCLK=18
PIN_IRQ=16
PIN_CE=4
PIN_CS=5

njordan77, 2023-04-18:
thanks, i know the pinout rather well of NodeMCUs. also checked for broken wires, but all this is not the case..will switch to another ESP-Board of another brand, maybe this works out as it should :-)

tbnobody, 2023-04-18:
Could you provide a screenshot of settings --> hardware? Just to doublecheck the pin assignment

njordan77, 2023-04-19:
update: i changed to another brand of ESP32 board and now it works without any issue. i'm somehow irritated as the NodeMCU32 is a standard board. would be intresting to understand the reason - but maybe it helps to some others also struggling with NodeMCU Boards (ESP32 version).

closed 2023-04-19 by njordan77