tbnobody/OpenDTU #818: build flags missing "-fexceptions"

benderl, 2023-04-18:
### What happened?

I tried to build the current firmware in VSCode with PlatformIO.
The build of the default fails with this errors:
```
src/WebApi_prometheus.cpp: In member function 'void WebApiPrometheusClass::onPrometheusMetricsGet(AsyncWebServerRequest*)':
src/WebApi_prometheus.cpp:98:30: error: exception handling disabled, use -fexceptions to enable
     } catch (std::bad_alloc& bad_alloc) {
                              ^~~~~~~~~
src/WebApi_prometheus.cpp:99:115: error: 'bad_alloc' was not declared in this scope
         MessageOutput.printf("Call to /api/prometheus/metrics temporarely out of resources. Reason: \"%s\".\r\n", bad_alloc.what());
                                                                                                                   ^~~~~~~~~
```

### To Reproduce Bug

Clone the repo and openit in VSCode. Hit the build button and watch the terminal.

### Expected Behavior

The build shoud not fail.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

ebc1b3aed20ab376695a3cba08efdb0f8729f7d0

### Relevant log/trace output

```Shell
Processing generic (board: esp32dev; framework: arduino; platform: espressif32@>=6.1.0)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Firmware Revision: v23.4.17
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2023.2.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 2.0.7 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 44 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP Async WebServer @ 1.2.3+sha.1dde9cf
|   |-- AsyncTCP @ 1.1.1
|   |-- ArduinoJson @ 6.21.2
|   |-- FS @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- ArduinoJson @ 6.21.2
|-- espMqttClient @ 1.4.2+sha.06c150c
|   |-- AsyncTCP @ 1.1.1
|   |-- WiFiClientSecure @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- RF24 @ 1.4.5
|   |-- SPI @ 2.0.0
|-- U8g2 @ 2.34.17
|   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|-- sunset @ 1.1.7
[... 113 lines cut ...]
src/WebApi_ws_live.cpp:241:112: note: suggested alternative:
In file included from /home/lutz/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_construct.h:59,
                 from /home/lutz/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_tempbuf.h:60,
                 from /home/lutz/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_algo.h:62,
                 from /home/lutz/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/algorithm:62,
                 from /home/lutz/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:172,
                 from .pio/libdeps/generic/ArduinoJson/src/ArduinoJson.hpp:18,
                 from .pio/libdeps/generic/ArduinoJson/src/ArduinoJson.h:9,
                 from include/WebApi_ws_live.h:4,
                 from src/WebApi_ws_live.cpp:5:
/home/lutz/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/new:54:9: note:   'std::bad_alloc'
   class bad_alloc : public exception
         ^~~~~~~~~
*** [.pio/build/generic/src/WebApi_ws_live.cpp.o] Error 1
============================================================================================================== [FAILED] Took 18.68 seconds ==============================================================================================================

Environment    Status    Duration
-------------  --------  ------------
generic        FAILED    00:00:18.678
========================================================================================================= 1 failed, 0 succeeded in 00:00:18.678 =========================================================================================================
```

### Anything else?

I added "-fexceptions" after line 24 in `platformio.ini` and the project builds just fine. I am not shure if this is the right location for the additional flag.

tbnobody, 2023-04-18:
Just look at #719 
This issue is somehow related to your environment....

benderl, 2023-04-18:
Thanks for the hint. Removing the complete `.platformio` folder in my home has solved this. I did a diff of the old and new folder but was not able to find any important changes.

Maybe this should be mentioned in the main README.md?

closed 2023-04-18 by benderl