tbnobody/OpenDTU #820: Firmware newer v23.4.12 -> no information of chip-status and type

DcFamas, 2023-04-19:
### What happened?

Hey there,
when I update the firmware to a version higher then v23.4.12, the chip type and chip status is invalid. I tested with 23.4.12, 23.4.15 and the newest 23.4.17. I guess it is because of the implementation of the new chip CMT2300A.

v23.4.12:
[image]

v.23.4.15 and v23.4.17:
[image]

The polling of the inverter still works.

### To Reproduce Bug

.

### Expected Behavior

.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.4.17

tbnobody, 2023-04-19:
You just did not press Refresh or F5 or CTRL+F5 in your browser after the update.

DcFamas, 2023-04-19:
Yes I'm stupid. Thought a reboot of the device is enough:
[image]