tbnobody/OpenDTU #826: OpenDTU auf ESP32 Pico Kit 4.0 does not start after power on without UART Terminal

charlie000111, 2023-04-19:
### What happened?

Hello,
ESP32 Pico Kit V4 was flashed after compiling (master version c 4 7 b 7 d d 3 ) with VS Code. Communication via radio module with NRF24L01+ and Hoymiles HM-600 works. However, OpenDTU only starts if a terminal program displays the data of the virtual COM interface (via USB).

### To Reproduce Bug

After plugging the USB connection into the PC (without running terminal) or via an external 5V supply of the ESP32 module, the ESP32 does not start up. This means the OpenDTU IP address cannot be reached by browser. 
There is no activity on the UART TxD signal nor on the signals to the radio module.
When then after some seconds a terminal program (e.g. Teraterm) is set up for the virtual COM port the monitor log of the software start is displayed. After that everything runs correctly and the live data can be displayed in the browser.
The module signals EN and IO0 are high immediately after power on and do not change their value. Hence, these signals cannot be the cause.

### Expected Behavior

OpenDTU starts up without necessity of running monitor UART connection.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

c47b7dd3

tbnobody, 2023-04-20:
The issue is maybe related to https://github.com/espressif/arduino-esp32/issues/6983

I don't own such a board. Could you doublecheck what happens if you insert

```cpp
Serial.setTxTimeoutMs(0);
```

directly after 

```cpp
// Initialize serial output
Serial.begin(SERIAL_BAUDRATE);
```

in the `main.cpp` file?

charlie000111, 2023-04-20:
Thnak you for your sugestion! But, then I cannot compile it any more. PLease see the attached error message.
[image]

charlie000111, 2023-04-20:
src/main.cpp: In function 'void setup()':
src/main.cpp:29:12: error: 'class HardwareSerial' has no member named 'setTxTimeoutMs'; did you mean 'setRxTimeout'?
     Serial.setTxTimeoutMs(0);       // added GÖP
            ^~~~~~~~~~~~~~
            setRxTimeout
Archiving .pio\build\generic\lib0e3\libFS.a
Compiling .pio\build\generic\lib8b5\WiFi\WiFiClient.cpp.o
*** [.pio\build\generic\src\main.cpp.o] Error 1
=================================================== [FAILED] Took 286.07 seconds ===================================================
Environment    Status    Duration
-------------  --------  ------------
generic        FAILED    00:04:46.066
============================================== 1 failed, 0 succeeded in 00:04:46.066 ==============================================

charlie000111, 2023-04-21:
It appears to me that the boot process is blocked at all by something because the first lines in the terminal aftre a terminal start are not like those in the main.cpp in setup() ("Starting OpenDTU   ...    Initialize FS... done"), but:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (619) esp_core_dump_flash: No core dum‚…ÉÑ¥Ñ¥½¹found!
E (619) esp_core_dump_flash: No core dump partition found!

Starting OpenDTU
Initialize FS... done
Reading configuration... done
Reading PinMapping... [   167][E][vfs_api.cpp:105] open(): /littlefs/pin_mapping                                                                                  .json does not exist, no permits for creation
using default config done
Initialize Network... done
Setting Hostname... Configuring WiFi STA using new credentials... done
Initialize NTP... done
Initialize SunPosition... done
Initialize MqTT... done
Initialize WebApi... done
Initialize Display... done
Initialize LEDs... done
Check for default DTU serial... done
Initialize Hoymiles interface... NRF: Connection successful
  Setting radio PA level...
  Setting DTU serial...
  Setting poll interval...
  Adding inverter: 114183169585 - Hoymiles HM600.1 done
done
Switch to WiFi mode
Setting Hostname... done
Configuring WiFi STA using new credentials... WiFi disconnected
Try reconnecting
Network lost connection
E (1185) wifi:sta is connecting, return error
[   396][E][WiFiSTA.cpp:278] begin(): connect failed!
done
Configuring WiFi STA static IP... done
WiFi connected
WiFi got ip: 192.168.179.53
Network connected
Websocket: [/livedata][1] connect
Fetch inverter: 114183169585
Request SystemConfigPara
TX RealTimeRunData Channel: 23 --> 15 83 16 95 85 80 11 56 92 80 0B 00 64 42 C3                                                                                   F5 00 00 00 00 00 00 00 00 9B 40 85
Interrupt received
RX Channel: 75 --> 95 83 16 95 85 83 16 95 85 01 00 01 01 36 00 0C 00 27 00 0A 0                                                                                  0 00 00 00 00 00 83
Interrupt received
RX Channel: 61 --> 95 83 16 95 85 83 16 95 85 02 52 A7 00 00 00 00 06 DD 00 00 0                                                                                  9 41 13 8B 00 25 4C
Interrupt received
RX Channel: 75 --> 95 83 16 95 85 83 16 95 85 83 00 00 00 02 03 E9 00 DB 00 09 3                                                                                  4 BF A7
RX Period End
Success

Any ideas?

trs3, 2023-04-28:
I see a comparable behaviour on my Esp32 Lolin S2 mini.

Plugging in the device into a USB power outlet doesn’t let it boot up. It doesn’t receive an IP and it’s not online.

Only when I plug it into my computer and start a serial connection it would boot up, get an IP and the OpenDTU interface is available.

Not sure what to do 😌

trs3, 2023-04-28:
Well, I commented out the serial.begin part in main.cpp and built with vscode. Now it works from an external usb power source and there is no need to start a serial connection to boot the esp.
Now I don't have a serial output anymore but I don't need that anyway.
Are there any other drawbacks that I am not aware of?

```
// Initialize serial output
// Serial.begin(SERIAL_BAUDRATE);
// while (!Serial)
//    yield();
```

commit 65b728ee (2023-05-03): Fix: No OpenDTU startup of CDC (like Pico or ESP32-S3) devices if no usb host is connected

Should also fix #826
closed 2023-05-03 by tbnobody