tbnobody/OpenDTU #828: [Request] Master + Slave option in OpenDTU

mowegener, 2023-04-21:
### Describe the solution you'd like

I would like an Master + Slave feature in OpenDTU. 
Til now I do have an Hoymiles HM-1500 inverter (esp32 + NRF24L01+). Now I would like to extend my systems with an Hoymiles HMT-2250 inverter (esp32 + CMT2300A), but for that I need the CMT2300A Radio module. That means 2 seperate OpenDTU systems. Now my request is an feature, to add an second OpenDTU as a "slave operator" to my current system (master operator), so that I only have one system to look at when I want to see the data. I do work with the mqtt option, that means I would have 2 different values to work with. An comparable use-case is the slave operator option in ioBroker.

Example:
Master operator (bundles and publishes the values) <------ slave operator (sends the values to the first operator)

tbnobody, 2023-04-21:
First of all, al master/slave system will not be implemented (it just generates a lot of useless work). 

Just let both DTUs publish their data into the MQTT broker. I don't understand why this is not a possible solution?
On the other hand, just connect both RF modules to the same ESP.

glucosum, 2023-04-21:
> On the other hand, just connect both RF modules to the same ESP.

Maybe a diagram of an example connection to make both work simultaneously on one ESP ....

tbnobody, 2023-04-21:
Diagram is not (yet) available. But you can look at the sample device profiles. There are already some definitions. e.g.
https://github.com/tbnobody/OpenDTU/blob/ebc1b3aed20ab376695a3cba08efdb0f8729f7d0/docs/DeviceProfiles/nodemcu_esp32.json#L102-L117

mowegener, 2023-04-21:
Thanks for the answers! I think im going with the "both RF modules on one esp" option

closed 2023-04-21 by tbnobody