tbnobody/OpenDTU #831: Sporadisch Fehler Peaks

themanfrommoon, 2023-04-22:
### What happened?

Ich habe 2 Hoymiles HM-600 mit je zwei 400W Solarmodulen.

Seit den letzten Updates (leider kann ich nicht genau sagen ab welcher Version) treten bei meiner OpenDTU immer wieder Fehler auf. Dabei werden folgende Werte angezeigt: 2.113.429 kWh, 6553,5Volt und 655,35Ampere (65536 ist der Maximalwert einer unsigned 16bit Zahl).
Zuerst hatte ich eine ESP32 mit NRF24L01 + PA + LNA in Betrieb und ca 50m Luftlinie zwischen WR und DTU. Dann habe ich das Funkmodul testweise auf EBYTE E01 ML01DPS gewechselt und maximale Leistung eingestellt. Zwischenzeitlich immer mal wieder ein neues Softwareupdate aufgespielt. Dann gingen die Probleme los. Dann wieder Funkleistung reduziert. Dann nicht mehr 50m entfernt, sondern nur noch 2,5m. Alles bisher ohne Stützelko. Nachdem die Probleme losgingen ein 100µF Elko direkt an die Pins vom EBYTE gelötet. Das hat keine Veränderung ergeben. Dann wieder auf das NRF24L01 + PA + LNA gewechselt. Auch keine Änderung.

Da jede Änderung an der Hardware und Funkstrecke keinen Einfluss hatte gehe ich inzwischen von einem Softwareproblem aus.

ICH habe MQTT laufen. mosquitto, Node-RED und Volkszähler läuft bei mir auf einem RaspberryPi.
Node-Red holt sich die Daten von mosquitto und schreoibt sie in den Volkszähler.

Dort kann man die unnatürlichen Peaks seit einer Weile sehr gut sehen.
Hier mal einige Screenshots:

[image]

[image]
[image]

Was könnte ich noch testen?

Beste Grüße,
Chris

### To Reproduce Bug

Es tritt sporadisch ohne zutun auf

### Expected Behavior

Es sollten keine Fehler auftauchen.

Wenn es durch irgendwelche Phänomene zu solchen Fehler kommt, und diese sich nicht eleminieren lassen, dann könnte man ja wenigstens einen Plausibilitätscheck einbauen:
Spannung kann nur Werte zwischen 0 und z.B. 1500V haben, alles was drüber und drunter liegt wird rausgesiebt.
Ströme können nur zwischen 0 und z.B. 30A liegen, alles was drüber und drunter liegt wird rausgesiebt.
Leistung kann nur zwischen 0 und z.B. 45kW (1500W x 30A) liegen, alles was drüber und drunter liegt wird rausgesiebt.
Die Plausibilitätsgrenzen kann man ja auch editierbar machen, so dass sie jeder individuell auf seine Installation anpassen kann.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.4.17

### Relevant log/trace output

```Shell
18:24:55.566 > Fetch inverter: 114182979823
18:24:55.623 > TX RealTimeRunData Channel: 3 --> 15 82 97 98 23 80 14 32 04 80 0B 00 64 44 0A 57 00 00 00 00 00 00 00 00 67 6A E2 
18:24:55.682 > Interrupt received
18:24:55.786 > RX Channel: 75 --> 95 82 97 98 23 82 97 98 23 01 00 01 01 7C 00 A8 02 7E 01 75 00 AE 02 89 00 00 6D 
18:24:55.835 > Interrupt received
18:24:55.991 > RX Channel: 61 --> 95 82 97 98 23 82 97 98 23 02 31 05 00 00 32 46 07 10 07 0D 09 07 13 85 04 CE 98 
18:24:56.128 > RX Period End
18:24:56.128 > Last missing
18:24:56.128 > Request retransmit: 3
18:24:56.128 > TX RequestFrame Channel: 23 --> 15 82 97 98 23 80 14 32 04 83 9A 
18:24:56.192 > Interrupt received
18:24:56.299 > RX Channel: 61 --> 95 82 97 98 23 82 97 98 23 83 00 01 00 35 03 E8 01 4E 00 01 58 A5 7A 
18:24:56.358 > RX Period End
18:24:56.358 > Success
18:25:00.569 > Fetch inverter: 114182974547
18:25:00.702 > TX RealTimeRunData Channel: 40 --> 15 82 97 45 47 80 14 32 04 80 0B 00 64 44 0A 5C 00 00 00 00 00 00 00 00 57 19 13 
18:25:00.803 > Interrupt received
18:25:00.854 > RX Channel: 23 --> 95 82 97 45 47 82 97 45 47 01 00 01 01 72 00 E7 03 59 01 71 00 E3 03 47 00 00 8C 
18:25:01.136 > RX Period End
18:25:01.136 > Last missing
[... 734 lines cut ...]
18:47:17.199 > Interrupt received
18:47:17.300 > RX Channel: 75 --> 95 82 97 45 47 82 97 45 47 83 00 00 00 34 03 E8 01 43 00 05 60 9D 73 
18:47:17.361 > RX Period End
18:47:17.361 > Success
18:47:21.587 > Fetch inverter: 114182979823
18:47:21.640 > TX RealTimeRunData Channel: 61 --> 15 82 97 98 23 80 14 32 04 80 0B 00 64 44 0F 99 00 00 00 00 00 00 00 00 12 66 50 
18:47:21.689 > Interrupt received
18:47:21.810 > RX Channel: 40 --> 95 82 97 98 23 82 97 98 23 01 00 01 01 76 00 79 01 C5 01 71 00 81 01 DC 00 00 73 
18:47:21.862 > Interrupt received
18:47:21.920 > RX Channel: 23 --> 95 82 97 98 23 82 97 98 23 83 00 01 00 26 03 E8 01 47 00 01 91 FB F7 
18:47:22.149 > RX Period End
18:47:22.149 > Middle missing
18:47:22.149 > Request retransmit: 2
18:47:22.149 > TX RequestFrame Channel: 75 --> 15 82 97 98 23 80 14 32 04 82 9B 
18:47:22.213 > Interrupt received
18:47:22.273 > RX Channel: 40 --> 95 82 97 98 23 82 97 98 23 02 31 15 00 00 32 56 07 20 07 1D 09 05 13 86 03 77 07 
18:47:22.334 > RX Period End
18:47:22.334 > Success
18:47:22.383 > TX AlarmData Channel: 3 --> 15 82 97 98 23 80 14 32 04 80 11 00 64 44 0F 99 00 00 00 00 00 00 00 00 C8 7D 8B 
18:47:22.443 > Interrupt received
18:47:22.523 > RX Channel: 61 --> 95 82 97 98 23 82 97 98 23 81 00 01 80 01 00 01 38 FC 38 FC 00 00 00 00 83 C8 DE 
18:47:23.025 > RX Period End
18:47:23.025 > Success
18:47:26.771 > Fetch inverter: 114182974547
18:47:26.827 > TX RealTimeRunData Channel: 23 --> 15 82 97 45 47 80 14 32 04 80 0B 00 64 44 0F 9E 00 00 00 00 00 00 00 00 22 40 F8

tbnobody, 2023-04-23:
Interessant wäre ein Log-Auszug aus einem Zeitbereich in dem ein Peak auftritt. 
Bisher sind solche Effekte nur bekannt wenn 2 DTU's verwendet werden.

themanfrommoon, 2023-04-23:
Was bedeutet 2 DTUs?

Zwei OpenDTUs nebeneinander, bei denen unterschiedliche Wechselrichterseriennummern eingetragen sind? (das habe ich)
Zwei OpenDTUs nebeneinander, bei denen die gleichen Wechselrichterseriennummern eingetragen sind? (das habe ich, kann es aber ändern. Die DTU erreicht den Wechselrichter aber nicht)
Eine OpenDTU und eine AHOYdtu, bei denen die gleichen Wechselrichterseriennummern eingetragen sind? (das hatte ich eigentlich noch vor, mindestens zum testen)

Wie generiere oder wo finde ich den Log-Auszug? (Die Konsolenmeldungen sind ja oben bereits enthalten)

tbnobody, 2023-04-23:
Log Auszug = Consolen Meldung. Der Peak erscheint aber kurz vor 19:00 Uhr. Dein Log ist von 18:24 - 18:47.

Bezüglich mehreren DTUs (und dabei ist egal, ob Original, OpenDTU oder Ahoy). Das wird nicht funktionieren. Siehe #727 und #26
Mit dem Log Auszug für einen Bereich in dem wirklich ein Peak enthalten ist lässt sich dann ganz leicht prüfen ob die DTU ggf. doch einen der Inverter erreicht und daher dieser Effekt auftritt.

themanfrommoon, 2023-04-23:
Ein Screenshot ist von 18:27 und der andere von 18:47.
Zu beiden Screenshots sind die Konsolenmeldungen im ersten Post drin!

tbnobody, 2023-04-23:
Ah ok. da sah der Graph etwas seltsam aus.
Welche Software Version verwendest du? Und hast du ggf. das Logging über einen längeren Zeitraum (5min) vor und nach der Zeit zu der es aufgetreten ist?

tbnobody, 2023-04-24:
> Die Softwareversion steht auch im ersten Post drin.

Oh entschuldige. Ich war da gestern wohl etwas nachlässig. 

> dass ein Betrieb von mehreren DTUs an einem Wechselrichter technisch nicht möglich ist, 

Werde das dem Troubleshooting Kapitel der Readme hinzufügen.

commit ece488cb (2023-04-24): Doc: Added hint regarding multiple DTUs per inverter

Fix #831
closed 2023-04-24 by tbnobody